pbatard/rufus #1885: Feature request: compiling for Win64/x86, too?

zedd-0, 2022-03-12:
hi, is it possible to compile rufus as a native Windows 64bit desktop application?
so it could used on an PE system, witch do not have a 32bit subsystem.

thx.

pbatard, 2022-03-12:
Use the AppStore version if you want a 64-bit version: https://www.microsoft.com/en-us/p/rufus/9pc3h3v7q9ch.
Or pick the 64-build from the VS2022 Artifact at https://github.com/pbatard/rufus/actions/runs/1969976438 (noting that this is unsigned and therefore not considered as an official build).

closed 2022-03-12 by pbatard

zedd-0, 2022-03-12:
a simple portable version would be nice.
there is already a x64 version... why is it not on github?

pbatard, 2022-03-12:
> why is it not on github?

Maintenance = time and very few people don't have 32-bit subsystem. It only looks painless, easy and quick to add something like a 64-bit win32 download until you do it yourself, and realize the implications (such as people, who have no clue about 32 vs 64-bit downloading the 64-bit version and trying to run it on a 32-bit environment).

I'm already getting enough questions about portable vs regular version, so, as long as the __actual need__ for 64-bit is minimal, I am not planning to publish a pure 64-bit version outside of the Windows Store (since I don't have to care about the 32 vs 64 bit issue there, as the store automatically picks the right version for the user).

Again, if you want a pure 64-bit version, use the Windows Store.

zedd-0, 2022-03-12:
but the Windows Store do not have a download link, so i could extract the files...

my tool-set is now bigger than the actual file.
and just to be prepared for all possible cases and complications...

you could make life a tiny bit easier, to put this file on the release page...

p.s.: "portable vs regular version"
i see no difference, the 'regular version' works fine as an portable version, in my use cases.

pbatard, 2022-03-12:
Again, everything looks quick and easy as long as you don't have to do it yourself.

You are trying to exchange making __YOUR__ life easier with regards to your specific need for my time. That's not how it works. Especially when it comes to an application that you get entirely for free.

By the way, there are ways to access the Windows Apps folder and pick the x64 version. And I already provided you with a link to a 64-bit build (which is actually the same binary as the one you'll get from the store except it's unsigned and therefore will pop up a message about being unofficial).

As far as I am concerned, if you are still left unsatisfied by this, it's your problem, not mine, as you can get the 64-bit version if you want... except it requires to invest your own time doing so rather than mine.

zedd-0, 2022-03-12:
"You are trying to exchange making YOUR life easier"
yes, I'm the only guy on this world, witch has to use PE systems... sorry.

yes i see your Artifact builds, and it is no problem there are unsigned.
but the i cannot disable the warning.
I have understood, that it is an unofficial version.
("stop talking, app!" :D )

could you put a disable option in the ini file?

pbatard, 2022-03-12:
> yes, I'm the only guy on this world, witch has to use PE systems... sorry.

That's not what I'm saying. I'm saying there isn't a critical mass enough of people without 32-bit subsystem to be worth it.

> could you put a disable option in the ini file?

Not gonna happen. This warning exists for a reason.