pbatard/rufus #1889: USB書き込み後のUSB破損

BarrelThyme8615, 2022-03-21:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
USBに書き込むと複数のパーティションが作成され、USBが読み込めなくなりました。
USBがもとから壊れてるのかと思い、別のUSBも試しましたが同じように破損しました。
Chrome OSを書き込もうとしていました。
Log
---
```
Found USB 2.0 device 'ELECOM MF-MSU2A USB Device' (056E:6027)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 943, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 12
Disk GUID: {4A9F77D7-4F46-4544-BDED-6791E1ABB37B}
Max parts: 128, Start Offset: 17408, Usable = 7759428096 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'STATE'
  Detected File System: ext4
  ID: {EABB9C52-1EE5-0B48-A3A3-EB6EA68706B3}
  Size: 1.2 GB (1275068416 bytes)
  Start Sector: 4481024, Attributes: 0x0000000000000000
Partition 2:
  Type: Chrome OS Kernel Partition
  Name: 'KERN-A'
  Detected File System: (Unrecognized)
  ID: {25BE6A9C-52B0-354D-9181-F02ECD0D326B}
  Size: 16 MB (16777216 bytes)
  Start Sector: 20480, Attributes: 0x01FF000000000000
Partition 3:
  Type: Chrome OS Root Partition
  Name: 'ROOT-A'
  Detected File System: ext2
  ID: {61E65740-6FE4-9E44-ADB8-EE757853E2BF}
  Size: 2 GB (2147483648 bytes)
  Start Sector: 286720, Attributes: 0x0000000000000000
Partition 4:
  Type: Chrome OS Kernel Partition
  Name: 'KERN-B'
  Detected File System: (Unrecognized)
  ID: {1E864FF0-C16B-DD42-8D40-76BB21C08FC3}
  Size: 16 MB (16777216 bytes)
  Start Sector: 53248, Attributes: 0x0000000000000000
Partition 5:
  Type: Chrome OS Root Partition
  Name: 'ROOT-B'
  Detected File System: (Unrecognized)
  ID: {E206B140-856A-AB49-AD89-92FFF1D86C11}
  Size: 2 MB (2097152 bytes)
  Start Sector: 282624, Attributes: 0x0000000000000000
Partition 6:
[... 165 lines cut ...]
  Detected File System: (Unrecognized)
  ID: {88FF804B-8CFE-C54D-BC9E-5AF31A7628FD}
  Size: 512 bytes (512 bytes)
  Start Sector: 16450, Attributes: 0x0000000000000000
Partition 10:
  Type: Chrome OS Reserved Partition
  Name: 'reserved'
  Detected File System: (Unrecognized)
  ID: {848D91C3-4B0A-584C-A5C7-83D29848758F}
  Size: 512 bytes (512 bytes)
  Start Sector: 16451, Attributes: 0x0000000000000000
Partition 11:
  Type: {CAB6E88E-ABF3-4102-A07A-D4BB9BE3C1D3}
  Name: 'RWFW'
  Detected File System: (Unrecognized)
  ID: {63A8ECCD-C64D-154A-92A9-190A52F55E67}
  Size: 8 MB (8388608 bytes)
  Start Sector: 64, Attributes: 0x0000000000000000
Partition 12:
  Type: EFI System Partition
  Name: 'EFI-SYSTEM'
  Detected File System: FAT16
  ID: {F0A736E7-1605-B740-9F5C-76FB35C49532}
  Size: 16 MB (16777216 bytes)
  Start Sector: 249856, Attributes: 0x0000000000000004

JonnyTech, 2022-03-21:
I found your image at https://github.com/crosbuilder/CustomBuilds and I think that you may need the ISO file at https://github.com/crosbuilder/CustomBuilds/releases/download/R45-7262.52.2015_09_06_1921/bootable.iso to boot.

The repo there is now unmaintained:

> This project is over. It will not be updated in the future.

So if they need to fix their img files then you may be out of luck.

closed 2022-03-21 by BarrelThyme8615

pbatard, 2022-03-21:
Because it creates a lot of GPT partitions, ChromeOS is known to trigger __A WINDOWS BUG__ that prevents a USB drive from being properly recognized. This is a pure Windows issue, that Microsoft needs to fix.

If you reformat your USB drive on a different OS (Linux, Mac) it will work fine.
You may want to report that problem to Microsoft, because it's really a problem with Windows and not Rufus.

Google Translation:
> ChromeOSは多くのGPTパーティションを作成するため、USBドライブが正しく認識されないようにするWINDOWSソフトウェアバグをトリガーすることが知られています。 これは純粋なWindowsの問題であり、Microsoftが修正する必要があります。
>
> 別のOS（Linux、Mac）でUSBドライブを再フォーマットすると、正常に動作します。
これは実際にはWindowsの問題であり、Rufusの問題ではないため、この問題をMicrosoftに報告することをお勧めします。