pbatard/rufus #1891: A question about the recording algorithm.

tfastgame, 2022-03-23:
Hello. I have recently contacted here, and asked about the possibility of porting the utility to Linux.
As it turned out, I gathered my team, which developed Rufus from scratch, and tried to develop its algorithm, which would allow to achieve the same results as the original program. But we did not succeed, apparently because we do not have much experience in this area.
Could Rufus developers help and write a step by step algorithm? I mean the one that your program works on Windows, and we would rework it for Linux.

Why don't I read the source code and make an algorithm based on it? I don't know C that this program is written in, I only know Python so I am asking for your help. Thank you in advance!

JonnyTech, 2022-03-23:
https://github.com/WoeUSB/WoeUSB - Linux shell
https://github.com/WoeUSB/WoeUSB-ng - Python

tfastgame, 2022-03-23:
> https://github.com/WoeUSB/WoeUSB - Linux shell https://github.com/WoeUSB/WoeUSB-ng - Python

Well, I would still like to look at the algorithm. But thanks for that too!

pbatard, 2022-03-23:
> Could Rufus developers help and write a step by step algorithm?

Sorry but no.

If you want to port Rufus to a non Windows OS, it is your job to understand what the Windows version does, from reading its source, and then translating that to your version.

I will also posit that you are under the impression that producing the algorithm is something that is easy. It is not, and would probably take us two weeks full time work to produce something that comes close to covering 80% of what Rufus does, and where you would find that a lot of the steps would boil down to: _"Call this __specific Windows API, that is not found anywhere else but Windows__ and perform action X"_, 

Please bear in mind that booting is more complicated than people realize, that Rufus enables more than single booting path (there's about a dozen of them, if not more), and that if your idea is that if you have the "algorithm" for one platform, then it should be easy to port the application to a different platform, you are going to sorely disappointed when you find that at least 35-50% of what Rufus does is tied to using Windows APIs.

If you want to port Rufus (or some limited functionality of Rufus, like the WoeUSB people did) to a different platform, then it is entirely __your__ job to:
1. Do your research.
2. Read the code of the existing application to understand what it does.
3. Figure out how to convert each functionality to your language and platform.

And that's because, if you don't have the motivation to pursue the above, and instead depend on external help to prop you forward, then it's very unlikely unlikely that you will have the motivation to see your port through, which you absolutely need to engage in a project like the one you have in mind.

That's not to say that we are unwilling to help you. But we can only help you once you actually have something to show for it since, then, we can be sure that you have some understanding of what you are actually getting yourself into, rather than a pipe dream of _"with an 'algorithm', I should be able to port Rufus to whatever language and whatever platform I want"_, that might sound great on paper, but might turn to something very different once you actually start to properly look into it.

closed 2022-03-23 by pbatard