pbatard/rufus #1893: Rufus takes over 12 hours to make a Zorin OS (iso) flash drive

Markizzzzz, 2022-03-25:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
[image]
Yesterday i started working on my flash drive (128 GB, 96 GB allocated to persistent storage), the progress was going great up until the point it hit 50% on formatting, then it started going abnormally slow (12,5h with 97% on formatting). I am using Windows 11 and Rufus 3.18.1877 (can be seen in the screenshot). The flash drive is brand new, Silicon Power Blaze B02, it was labelled as USB 3.2 Gen 1. Can't use another flash drive because of a lack of one and the same goes for the other ports: I only have 1 port that supports USB 3.0. Zorin OS support page lists only its SHA256 checksum, which is identical to what i have in the logs.

Log
---
```
Rufus x86 v3.18.1877
Windows version: Windows 11 Home, 64-bit (Build 22000.556)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0419 (ru-RU)
Will use default UI locale 0x0419
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ru-RU'
NOTE: The Russian (Русский) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 3.0 device 'USB DISK 3.2 USB Device' (13FE:6700)
1 device found
Disk type: Removable, Disk size: 128 ГБ, Sector size: 512 bytes
Cylinders: 15068, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x839868FB
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 115.4 GB (123924185088 bytes)
  Start Sector: 36224, Boot: No
Found USB 3.0 device 'USB DISK 3.2 USB Device' (13FE:6700)
1 device found
Disk type: Removable, Disk size: 128 ГБ, Sector size: 512 bytes
Cylinders: 15068, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x839868FB
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 115.4 GB (123924185088 bytes)
  Start Sector: 36224, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20191223 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Zorin OS 16.1 Core 64bit'
  Size: 2.8 ГБ (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: Zorin-OS-16.1-Core-64-bit.iso (2.8 ГБ)

Computing checksum for 'C:\Users\mcmar\Downloads\Zorin-OS-16.1-Core-64-bit.iso'...
  MD5:    e9d1906483978c073a28c663718d402d
  SHA1:   371c5bbacaf931a6cbcb732ee94ec4c3233b2e4c
  SHA256: 2bc01d2553758c0637d786442c654a47ecba4c6368c91f503e09ab09f46e4dd4
Downloading https://rufus.ie/files/syslinux-6.04/20191223/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
File length: 66.5 КБ
Successfully downloaded 'ldlinux.sys'
Download signature is valid ✓
Downloading https://rufus.ie/files/syslinux-6.04/20191223/ldlinux.bss
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.bss
File length: 512 байт
Successfully downloaded 'ldlinux.bss'
Download signature is valid ✓

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive2 for shared write access
Analyzing existing boot records...
Drive has a Windows 2000/XP/2003 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 19.4 GB)
● Creating Linux Persistence Partition (offset: 20863519744, size: 96.0 GB)
Waiting for logical drive to reappear...
Using Ubuntu-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (ext3)...
1572863 possible inodes out of 25165822 blocks (block size = 4096)
1258291 blocks (5.0%) reserved for the super user
Creating 768 inode sets: [1 marker = 9.6 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Creating 65536 journal blocks: [1 marker = 819.2 block(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
```

Markizzzzz, 2022-03-25:
The process has now finished, it took 14:40:28 in total. Is there anything I can do on my side to improve the time for future operations?

willzyx-hub, 2022-03-25:
> The process has now finished, it took 14:40:28 in total. Is there anything I can do on my side to improve the time for future operations?

Hello... either your USB drive speed is slow or your computer had a problem so i suggest you to change the USB Drive first before looking on the computer because Rufus speed depend on both factor

Markizzzzz, 2022-03-25:
Should i try doing that with the other ports, i was a bit wrong about my pc, other slots are just Gen1 instead of Gen2, all of them support 3.0. Also, is it worth trying to change the file system from LargeFat32 to NTFS?

willzyx-hub, 2022-03-25:
> Should i try doing that with the other ports, i was a bit wrong about my pc, other slots are just Gen1 instead of Gen2, all of them support 3.0. Also, is it worth trying to change the file system from LargeFat32 to NTFS?

No do not format as NTFS it won't work

pbatard, 2022-03-25:
> 96 GB allocated to persistent storage

That, along with a USB flash drive that is very slow for random write I/O (please be very mindful that it's not because you have a brand new flash drive that is 3.2 Gen 1 that it's random I/O write speed might not be __excruciatingly slow__ &mdash; For instance, I recently purchased a 3.2 Gen 1 SanDisk drive that has a slower random write speed than any other flash drive I've purchased over the last 15 years) is the source of your problem.

When creating the journal, the `ext3` formatting algorithm needs to write a lot of random blocks, and if your drive has a slow random I/O speed, it's going to be slow. Note that we are using the same code as the one used by Linux to allocate these blocks (our source is from the ext2fs project), so there isn't much we can do to speed things up if your flash drive is slow.

So your options are:
1. Drastically reduce the size of your persistent partition (since it's not a good idea to keep persistent data on a flash drive for an extended duration anyway, as flash memory is not very reliable and you should always expect that drive to fail at any moment).
2. Purchase a drive that is SSD based and therefore will have a much better random write I/O (and be very mindful that if a manufacturer advertises that their drive is 3.2 Gen 1, but don't publish detailed write speeds, it's probably because they are trying to fool consumers into believing that the drive is fast, when it isn't).
3. Create a `UseExtVersion` DWORD registry key under `HKEY_CURRENT_USER\SOFTWARE\Akeo Consulting\Rufus\` (or create a similar entry in your `rufus.ini` if you use the portable version) and set it to `2`, as this will force the persistent partition to use `ext2` rather than `ext3` and therefore drop the creation of the journal altogether. We basically put this setting in after we indeed realized that formatting to `ext3` would be exceedingly slow on drives that have slow write I/O, and that some people might therefore want a way to force `ext2` which is faster since it doesn't need to perform a lengthy block allocation for the journal.
4.

Markizzzzz, 2022-03-25:
Ok, i will definitely try doing that. The creation of journals was the main thing that was taking 11-12 hours, so hopefully the overall process will not take that much time. Also, after i booted into the Zorin OS Core (after 15 hours :)), it seemed that the persistent storage didn't work. I tried booting with and without safe boot, but my test.txt always disappeared (i did try saving it). I have not seen any signs of Zorin in incompatible software section of the FAQ. So should I switch to something else (eg. Mint) or is there anything I can do about it?

pbatard, 2022-03-25:
Not surprised that persistence does not work with Zorin. The problem is that Linux distro maintainers have not harmonized any common way to handle persistence (they can't even agree if the kernel option should be called `persistent` or `persistence`), which makes it very tricky for applications like Rufus to try to make persistence work with every single distro out there.

Therefore, so as not to have to spend our life adding support for whatever [NIH](https://en.wikipedia.org/wiki/Not_invented_here) persistence implementation this or that distro decides to use, [Rufus only supports Debian-like or Ubuntu-like methods of persistence](https://github.com/pbatard/rufus/blob/14847a1a89e9f6deaff7c15007a53665b7050e7b/src/iso.c#L331-L357).

It therefore looks like Zorin doesn't use a method that is compatible with Debian or Ubuntu and thus persistence will not work. I also don't think Mint follows the Debian or Ubuntu method of adding persistence, so it probably won't work either.

If you __really__ want to fix this, you should report to the Zorin or Mint distro maintainers that the fragmentation of the methods of enabling persistence between Linux distributions is impacting you, and that they should really strive to make an effort not to reinvent the wheel, but instead discuss with other distro maintainers to try to come up with a common standard of adding persistence to a live distro.

Markizzzzz, 2022-03-26:
Yeah, changing to ext2 made wonders, i've prepared a flash drive with same persistent storage amount, but with lubuntu in just a bit less than 2 hours instead of 14. And the persistent storage works too, albeit my flash drive is still very slow XD. Thanks for the help!
[image]

closed 2022-03-26 by Markizzzzz