pbatard/rufus #1894: Defender block credential theft

richbreton, 2022-03-25:
﻿

Checklist
---------
- [ x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x ] I also tried one or more of the following:
  - [x ] Using a different USB drive.
  - [x ] Plugging the USB into a different port.
  - x[ ] Running Rufus on a different computer.
- [ x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Just letting you know that Rufus is now tripping up defender again, due to some new code pushed out 2/22.

The error "ASR (Attack Surface Reduction) blocked rufus-3.18.exe triggering the rule "Block Credential Stealing from the Windows Local Security Authority Subsystem (lsass.exe)

It appears that anything accessing lsass.exe from a privileged account and tries to dump the memory will flag this. Here is an article explaining the change:

https://petri.com/microsoft-defender-block-windows-credential-theft/

Here is the microsoft documentation on the defender feature:

https://docs.microsoft.com/en-us/microsoft-365/security/defender-endpoint/attack-surface-reduction-rules-reference?view=o365-worldwide

I'm not attempting to say this is a bug in Rufus just a new reality that has to be dealt with on the windows platform. I hope this is helpful and thank you

Log
---
```
<FULL LOG>
```

JonnyTech, 2022-03-25:
Which version of Windows: 10/11,  Home/Pro,  x86/AMD64? Which build of Rufus: Github release / Windows Store / CI artifact? Can you share a screenshot please?

pbatard, 2022-03-25:
I'm not seeing Defender being tripped, and VirusTotal (who should use the latest version of Defender as well) [doesn't appear to be tripped by Rufus 3.18 either](https://www.virustotal.com/gui/file/22820692cb7295cd13bf62ab984a8f5b37e3cb09999b6aa2ad27a704e3380c48/detection).

It's a shame you did not produce a log, because I don't know what version of Windows you are running as well as other details that the log would have produced information about. So I have to ask you for more details:

1. What version of Windows are you seeing this on? What edition? What arch?
2. Was it with the release 32-bit version of Rufus 3.18 or something else?
3. Are you sure you are using the official release of Rufus and not some malicious version downloaded from a shaddy site? Especially, we're certainly not accessing lsass in any way shape or form, so I'm puzzled as to why Defender would be triggered.

pbatard, 2022-03-28:
Considering that we're still not seeing verified reports of the official version of Rufus being flagged by Defender and that OP has not followed up on the questions, I will close this issue.

closed 2022-03-28 by pbatard

ShadowXVII, 2022-07-25:
Happening for me -- Event Viewer log:

Microsoft Defender Exploit Guard has blocked an operation that is not allowed by your IT administrator.
 For more information please contact your IT administrator.
 	Path: rufus-3.19.exe
 	Process Name: C:\Windows\System32\lsass.exe
 	Target Commandline: "rufus-3.19.exe" 
 	Parent Commandline: 
 	Involved File: 
 	Inheritance Flags: 0x00000000
 	Security intelligence Version: 1.371.780.0
 	Engine Version: 1.1.19400.3
 	Product Version: 4.18.2205.7

There is no NTFS MOTW on this file.

pbatard, 2022-07-25:
> Unsure if this is causing the "Access to the device is Denied" error

It mots likely is.

Considering that many many people (including myself) run Rufus on Windows 11 without any issues, and `[0x00000005] Access is denied` is __ALWAYS__ caused by external software actively preventing Rufus operations, your Windows 11 installation has clearly been altered from its default settings to prevent operations like low level access to USB drives.

ShadowXVII, 2022-07-26:
Tried adding to ASR exclusions but no luck. I suspect some of the other attack surface reduction options cause this issue, i.e. access to executables running on USBs, etc.

I've used the Windows 11 media creator (MediaCreationToolW11.exe) in the interim with command line parameters mentioned online to achieve the USB installation required.

pbatard, 2022-07-26:
> I suspect some of the other attack surface reduction options cause this issue, i.e. access to executables running on USBs, etc.

Logically, I don't believe that to be the case because:
1. The `Access is denied` error happens when Rufus is trying to write empty blocks (not files or anything), and with proper administrative access, this shouldn't be prevented by the OS, even if the USB drive is riddled with files that contain malware (because why would you want to prevent malware from being overwritten? That is, unless you do have malware actively executing from the USB, but then you have a much larger problem...).
2. You are trying to write a Windows ISO anyway, which means that, if it is legit, it's content comes from Microsoft, and Microsoft tends to make sure that their binaries aren't going to be flagged as malware.

Again, the most likely explanation is that you have security software or policies actively preventing access to the drive, so I suggest you have a look at [this relevant entry from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied), along with the list of software Rufus is not compatible with (because quite a few entries there do result in `Access is denied`).