pbatard/rufus #1896: [request] Add support for Linux and MacOS

kazuhiroshigeru, 2022-03-31:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
As the title says, would you ever consider porting rufus to Linux distros and MacOS? because sometimes I need to make windows boot USBs and those require rufus but I run Arch Linux. I can use ventoy, but sometimes it doesn't work with some specific scenarios.
Log
---
```
<FULL LOG>
```

pbatard, 2022-03-31:
I quote you:

> - [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.

[Really?](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_port_Rufus_to_LinuxMac_OSSome_other_OS)

> - [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.

[Oh really???](https://github.com/pbatard/rufus/issues?q=port+Linux+)

Excuse me for being direct, but next time, don't just pretend to go through a check list when it's going to be pretty obvious that you didn't.

closed 2022-03-31 by pbatard

Hrxn, 2022-04-01:
I admire your patience, @pbatard

willzyx-hub, 2022-04-22:
> I quote you:
>
>
> [Really?](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_port_Rufus_to_LinuxMac_OSSome_other_OS)
>
>
> [Oh really???](https://github.com/pbatard/rufus/issues?q=port+Linux+) [...]

Can it be run under compatibility layer such as wine?

pbatard, 2022-04-22:
> Can it be run under compatibility layer such as wine?

No. Wine is too limited and does not provide the hardware translation layer that is required to list and access USB devices.

Wine is not a complete hardware virtualization system, and as such, it will always have its limits

willzyx-hub, 2022-04-23:
> > Can it be run under compatibility layer such as wine?
> 
> No. Wine is too limited and does not provide the hardware translation layer that is required to list and access USB devices.
> 
> Wine is not a complete hardware virtualization system, and as such, it will always have its limits

Sad :( so there is no way to get it work on non windows system

monique994, 2022-04-23:
I need lots of help my automation and my domain are somehow messed up. I can't access my bluehost provider and they keep telling me my email is Comcast.  But Google has my registrations. I dont know how to get it from the log in gooeworkspace. #Please help!!

rustyx, 2022-04-25:
I saw Rufus UI working fine under Wine, it is only USB access that is missing.
So we should be able to create a [Winelib proxy DLL](https://wiki.winehq.org/Winelib_User%27s_Guide#Building_Winelib_DLLs) to expose the raw `/dev/sdX` device.
Do you think that would work @pbatard?
(Sorry I'm a Linux dev and don't know much about raw USB access under Win32)

willzyx-hub, 2022-04-25:
> I saw Rufus UI working fine under Wine, it is only USB access that is missing. So we should be able to create a [Winelib proxy [...]

You need to add some Windows Formatting Library too i think

pbatard, 2022-04-25:
> Do you think that would work @pbatard?

That would work to a certain extent (but it __will__ require A LOT of work to write the relevant winelibs, as you're basically going to have the rewrite entire parts of the Windows OS, which won't be easy since it's closed source). However, you will still be missing PowerShell access (required for the ISO download feature) or `bootmgr.exe` access (required for Windows To Go), and probably more Windows-centric elements that Wine will have a hard time providing.

At any rate, I don't want to be involved in such a project on account that, it's pretty obvious that you're always going to end up with a half-working version of Rufus when running on Wine, and that, if all you're interested is Windows bootable media creation support, your time will be better spent helping __native__ Linux projects like [WoeUSB](https://github.com/WoeUSB/WoeUSB-ng), if you believe they fall short of what you want in an app.

willzyx-hub, 2022-04-25:
> That would work to a certain extent (but it **will** require A LOT of work to write the relevant winelibs, as you're basically going to have the rewrite entire parts of the Windows OS, which won't [...]

What about Porting the source code to Linux ?

pbatard, 2022-04-25:
> What about Porting the source code to Linux ?

Read the first link I provided in my first reply. Thank you.

rustyx, 2022-04-27:
> ... if all you're interested is Windows bootable media creation support, your time will be better spent helping **native** Linux projects like [WoeUSB](https://github.com/WoeUSB/WoeUSB-ng), if you believe they fall short of what you want in an app.

I don't care about Windows. And I really didn't want to care about Rufus, either, if only there was another way for me to make a bootable Linux USB for some laptops (e.g. ASUS UX550). And no, WoeUSB-ng didn't work, either.

Basically Rufus's "ISO mode" flashing is like magic, it **always works** and is the **only way** I found to make a USB from an Ubuntu ISO that would boot correctly. I think this has to do with UEFI BIOS bugs with ISO9660 handling, and formatting in FAT32 fixes it.

So I'll need to see what Rufus does to make a bootable FAT32 UEFI partition, perhaps there's a way to reproduce that with other tools.

JonnyTech, 2022-04-27:
@rustyx have you tried [unetbootin](https://unetbootin.github.io/)? I have never had an issue creating Ubuntu based bootable drives with it.

rustyx, 2022-04-27:
Yes of course tried UNetbootin, mkusb, bootable-iso, `dd` (of course), manual `parted` bootable fat32 (mbr/gpt) + copying ISO contents over, installing a loopback EFI GRUB + ISO file, some other tools I can't remember, etc. - **nothing** worked on that machine - GRUB starts, can load the kernel, but the kernel won't boot - "out of memory" followed by "unable to mount root fs" ([pic](https://user-images.githubusercontent.com/1296899/165626862-4d259e88-192b-43a3-8fa3-8a5ae35c6257.jpg)).
But with Rufus it just works, with both MBR and GPT (UEFI non-CSM).