pbatard/rufus #1897: Restarting on Completion and verify against image used

TyTytheFurKit, 2022-04-01:
An option to restart the computer when the USB is done being written to would be great.

(Also a verification option for when the DD method write is used)

pbatard, 2022-04-01:
Verification after writing a DD image is vaguely planned (#1880 &rarr; _Add dd image verification?_).

Restarting the computer after writing is way too specific for it to be worth it as you need to be using a very slow (read: __ill-suited__) drive to need to want this as well as:
- boot the computer you are currently running Rufus on with the new drive, which is a not so common scenario
- have made sure that your BIOS/UEFI settings are set to boot from that USB, which is not is not a given and not something you have always sorted beforehand.

So my advice for you is: If you think there is a need for this reboot feature, you should invest into using an SSD based USB drive because I'm going to assume that you are doing something like creating a Windows To Go or Linux persistent drive that take more than 15 minutes to complete, and if that is the case, then you should know that you can most certainly achieve less then 15 minutes creation speeds (yes, even for Windows to Go) by using a drive that is actually suited for the job in terms of random write I/O speed.

But then again, since you don't elaborate about the kind of scenario that makes you request this specific feature, and in all my years of creating USB drives, I have never ever thought _"Wouldn't it be great if the computer could just reboot when this operation is done?"_, I have no choice but take a __guess__ as to why you might think you'd like something like this, and come to the conclusion that this will not help that many people besides you based on that __educated guess__ alone.

closed 2022-04-01 by pbatard

TyTytheFurKit, 2022-04-03:
Nowhere did you have to attack my suggestion like that. Making me feel like an idiot for just wanting a simple option so that way if I was waiting on a machine to write to a disk and have it reboot afterwards so I wouldn't have to babysit it the entire time and manually restart. The suggestion was to fucking save time so that way people who wanted to get things done quickly like reinstalling a corrupted operating system from flash drive, or to install a different OS that didn't support upgrading (Like from Windows XP to Windows 10). Why do you feel the need to attack me for being interested in the development of this wonderful software?! I just thought it'd be a good idea to add a small feature such as "Reboot upon completion" for the few that may have fucking needed it, so if they were away and they had their BIOS set to boot USB as first priority, when they got back, whatever they chose to boot would be booted, and it would be ready for user interaction.

TyTytheFurKit, 2022-04-03:
I admit my suggestion was vague, but I thought since you were smart enough to know, you would have probably theorized as to what this suggestion could be applicable to. But no, you decided to attack my suggestion without mercy, and make me feel horrible for wanting to improve Rufus in anyway. I thought maybe you would have thought it was a good idea in theory, and use constructive criticism to maybe ask me what use cases this would be appropriate in. Just because you are a software developer, and that YOU own Rufus as a whole doesn't mean you need to shun away ideas you think are not worth your ideas. If you didn't want suggestions in the first place, MAYBE YOU SHOULD HAVE MADE THE SOFTWARE COMPLETELY CLOSED TO IDEAS AND SUGGESTIONS, AND CLOSED OFF FROM GITHUB ENTIRELY.

pbatard, 2022-04-03:
I've simply wanted you to spend time considering your suggestion, from more than your specific perspective, before creating a proposal in the tracker.

If you did, then you might have considered that in the vast majority of cases, people are not trying to boot the same computer with the media they just created, so that already drastically limits the playfield of how "actually" useful such a feature would be. Oh and considering that I have [officially ceased to support anything XP (or even Windows 7)](https://github.com/pbatard/rufus/wiki/FAQ#Windows_XP_Windows_Vista_Windows_7_NO_LONGER_SUPPORTED), that _"Windows XYZ to Windows 10"_ example is not a good incentive, even more so as Windows 10 will, in the vast majority of cases, run very poorly on a computer that was designed to run Windows XP, if at all..

Then, once again, I have to point out that if you do feel like you need to babysit the computer for a long time, you are probably not using a drive that is good enough for the job.

> I just thought it'd be a good idea

And therein lies the problem.

You did not try to extend your perspective beyond the easy _"if I have a need for it, then everybody else will"_, and that is the part where I would really have liked to have you apply the kind of _constructive criticism_ that you are requesting from others.

It's actually not that difficult to detach yourself from _"I need this, therefore everybody else will"_ and start to look at the feature from the perspective of:
- How many people are actually using Rufus to create media to boot the computer they are running?
- What's the amount of media creation time that would warrant such a feature? For instance, does such a feature really come useful if the media is created in less than a few minutes, which, again, will be the case if you create Windows installation media and use a decent modern flash drive. Even on relatively old USB 3.0 media, and on a not so young (read more than 10 years old) PC, I've never experienced a Windows installation media creation time of more than 5 minutes.
- If that amount of time is quite extensive (e.g. more than 5 minutes), could it be because the media or image mode (e.g. Windows To Go, which is __very different__ from the XP &rarr; 10 installation path you are putting forward) are actually ill suited, and therefore, shouldn't the problem rather be addressed at this level rather than with a new software feature?

> you would have probably theorized as to what this suggestion could be applicable to

I did. And that is precisely why I rejected it. I tried to look from a global perspective, and from that viewpoint, it makes very little sense to have it, when I am still seeing about every underlying cause from requesting such a feature to boil down to using media that is ill suited for that specific task, and I actually see it __more useful__ to make it prohibitive enough to have people reflect on this, and alter their environment, as it should actually save them time and be beneficial to them in the long run.

> I thought maybe you would have thought it was a good idea in theory

And  thought that maybe you would have thought that there was a chance somebody else may not see it with the same keen eye as you did, because it originated from you, and applied some critical thinking as to whether this idea was actually as good as you felt it was.

>  doesn't mean you need to shun away ideas you think are not worth your time.

If it can prevent people from nilly willy thinking they have a great idea for a feature, and prevent other developers waste their time trying to explain, perhaps in a nice manner than I did, why it's not actually that great an idea and why an issue tracker is not a repository for _"Here's an idea for your software. You sort it out!"_ but a place where we do expect people to have applied some critical thinking before placing a proposal (which does not require to have any knowledge of how to develop software), then, yes, I will continue to not so nicely shun ideas that I don't think should have been put in the open in the first place (that is unless the person doing so does have a very compelling usage scenario to present it with), so that people whose ideas were reject do act more cautiously next time around, as I believe it will benefit everybody in the long run.