pbatard/rufus #1898: img shrink feature request

infr-automation, 2022-04-02:
Feature request

shrink IMG and it's partitions, to the size of the target storage device

Tried with 128MB sdcard and supergrub2-2.04s2-beta2-multiarch-USB.img (as well as openwrt x86_64 which as ext2 partition)
started and errored with Write error at sector 196608: [0x0000001B] The drive cannot find the sector requested.

Thank you.

pbatard, 2022-04-02:
Not going to happen.

If you believe that the `supergrub2-2.04s2-beta2-multiarch-USB.img` could be shrunk to 128MB, then ask the people who created that image to do it. After all, they are the ones who are best suited to know if their content can actually be shrunk, and how to accomplish just that.

I'm not a magician here &mdash; I can't suddenly figure out how to automate something that the people who created a __VERY CUSTOM__ image have not done themselves, and also my development time is limited, and even if this was achievable, the cost/benefit ratio is way too high.

As a matter of fact, be prepared to hear that people won't want to invest __WEEKS__ of their limited time on this planet just to save you (and a very small number of people) the less than €10 that a [4 GB+ SD Card costs](https://www.aliexpress.com/wholesale?catId=0&SearchText=SD+Card).

So do yourself a favour, and go buy yourself a new SD Card.

&rarr; Rejected.

closed 2022-04-02 by pbatard

infr-automation, 2022-04-02:
Relax, you will make time for what you want and need out of this planet, regardless of my feature requests.

Some industrial machines don't take bigger or newer storage.

The tested img was just an example.

Thanks for your work.

pbatard, 2022-04-02:
I still don't understand why you ask me rather than the people who create the supergrub image.

Or, if you actually want to see the other side of the equation, and get an idea of why people may choose not to devote their time solving somebody else's very limited problem, and think that there will be that many people using industrial machines that can't take more than 128 MB cards, you __could__ write the image to a virtual drive, then create a second 128 MB virtual drive, create the right partitions there and copy the GRUB content from the first virtual drive to that other smaller drive. And then you could upload that smaller supergrub image to a file sharing site somewhere and bask in the knowledge that you'll have done something useful for people just like you, and that it will have costed you a lot less time than what it would take me to even try to accomplish what you request. And of course, in the process of helping others, you might gain some knowledge on how anything that you don't have to do yourself always seems super easy and a lot less time consuming than it actually is.

So, my question to you is: Do you wanna help other users who are running into the very specific scenario you just described (and for which adding a dedicated feature in Rufus would be a complete overkill) or not? Because I've just told you how you could accomplish that without the need to invoke a busy external developer's resources.