pbatard/rufus #1901: Thanks for the update, and the link to gifrecorder.

incognito10278, 2022-04-04:
Thanks for the update, and the link to gifrecorder.

I tried some more with various drives, including drives that are larger than 500 GB as well as a 500 GB drive, but I never got your error.

I am therefore going to close this issue and wait to see if other people report something similar. If not, then it means it's probably something specific to your environment.

_Originally posted by @pbatard in https://github.com/pbatard/rufus/issues/1544#issuecomment-632687668_

incognito10278, 2022-04-04:
i also have the same error and cant seem to solve it

pbatard, 2022-04-05:
I need to see your log to determine if it's the same error.

I'm afraid I simply can't do anything from people reporting "me too!" on an issue I have never been able to replicate. You must provide data that might be able to help me reproduce the issue, but you haven't done that.

Without such data, and some hints from users like you as to what specific environmental parameters you have, compared to a standard Windows installation (since this issue only seems to occur to a very limited number of people), there can be no progress on this issue. 

At the very least, besides a log, which is __ALWAYS REQUIRED, EVEN IF YOU ARE FOLLOWING UP ON AN EXISTING ISSUE__, you should provide some clear information regarding the type of USB drive that was used including manufacturer, capacity and speed. Please remember that developers don't use crystal balls, so, even if it may sound harsh, the shortest way to get your issue remain unresolved, is to provide as little information as possible, such as discarding a __mandatory__ check list and just stating _"i also have the same error and cant seem to solve it"_.

pbatard, 2022-04-09:
With no new data to add the the previous issue, I have no choice but to close this duplicate.

closed 2022-04-09 by pbatard