pbatard/rufus #1902: GRUB stalls on a system that has only BIOS support with no UEFI

fenixleon, 2022-04-06:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I have several old laptops that only support BIOS or Legacy mode. They do not have support for EFI/UEFI. The laptops all have a 64-bit CPU.  I've tried several desktop flavors of Ubuntu - Ubuntu Budgie Kubuntu and Lubuntu The hashes of the downloaded ISOs match. To rule out USB flash drive failures or issue, I have used different USB flash drives. Similarly, I have plugged the USB flash drives into different USB ports.

The laptop successfully boots, gets past the vendor splash image/screen e.g. HP and stops at Welcome to GRUB!. There are no other messages and it doesn't progress any further. Following the exactly the same steps with Ubuntu, it loads successfully.

The default settings in Rufus are not changed e.g. the partition scheme is MBR, the file system is FAT, etcetera

Log
---
```
Rufus x86 v3.18.1877 (Portable)
Windows version: Windows 10 Pro, 64-bit (Build 19044.1586)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
0 devices found
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Detected Grub version: 2.04
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu-Budgie 21.10 amd64'
  Size: 3.0 GB (Projected)
  Has a >64 chars filename
  Uses: GRUB2
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: ubuntu-budgie-21.10-desktop-amd64.iso (3.0 GB)
0 devices found
Found USB 2.0 device 'USB FLASH DRIVE USB Device' (1005:B113)
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 3.9GB, Sector size: 512 bytes
Cylinders: 471, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x574A3042
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Detected File System: FAT12
  Size: 4 MB (4194304 bytes)
  Start Sector: 484, Boot: No
Partition 3:
  Type: GNU/Linux (0x83)
  Detected File System: ext4
  Size: 1.1 GB (1185611776 bytes)
  Start Sector: 5263360, Boot: No
Computing checksum for 'C:\Temp\operating-systems\ubuntu-budgie-21.10-desktop-amd64.iso'...
  MD5:    1dd976ca321958dee838fea8ab6bed76
  SHA1:   9830ab9e8aa49cf55baacf8f8f9a63e25253bf85
  SHA256: 080bb15f3192463c8fc197ed2dfcbf7a26f0168fc38ee4aca54ed4ba15245233
```

pbatard, 2022-04-06:
Thanks for the report. I have just tested `ubuntu-budgie-21.10-desktop-amd64.iso` (same SHA256 as the one you used) written in ISO mode and had no issue booting in BIOS/Legacy mode.

Now, your log is missing the actual media creation part, which is problematic, and the partition layout from the log appears to indicate that you wrote the image in DD mode (since there exist multiple partitions with an ESP). Yet, when I write `ubuntu-budgie-21.10-desktop-amd64.iso` in DD mode, I don't get the same layout at all!:
```
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Detected Grub version: 2.04
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu-Budgie 21.10 amd64'
  Size: 3.0 GB (Projected)
  Has a >64 chars filename
  Uses: GRUB2
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: ubuntu-budgie-21.10-desktop-amd64.iso (3.0 GB)

Computing checksum for 'D:\ISOs\ubuntu-budgie-21.10-desktop-amd64.iso'...
  MD5:    1dd976ca321958dee838fea8ab6bed76
  SHA1:   9830ab9e8aa49cf55baacf8f8f9a63e25253bf85
  SHA256: 080bb15f3192463c8fc197ed2dfcbf7a26f0168fc38ee4aca54ed4ba15245233

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Opened \\.\PhysicalDrive8 for exclusive write access
Requesting logical volume handle...
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

Found USB device 'SanDisk Extreme Pro USB Device' (0781:5588) [ID]
1 device found
No volume information for drive 0x88
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15567, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {218953EE-E6C1-4E37-BC64-3B4C941EA515}
Max parts: 248, Start Offset: 32768, Usable = 128043647488 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'ISO9660'
  Detected File System: (Unrecognized)
  ID: {218953EE-E6C1-4E37-BC65-3B4C941EA515}
  Size: 3.0 GB (3190812672 bytes)
  Start Sector: 64, Attributes: 0x1000000000000001
Partition 2:
  Type: EFI System Partition
  Name: 'Appended2'
  Detected File System: FAT12
  ID: {218953EE-E6C1-4E37-BC66-3B4C941EA515}
  Size: 4.1 MB (4333568 bytes)
  Start Sector: 6232120, Attributes: 0x0000000000000000
Partition 3:
  Type: Microsoft Basic Data Partition
  Name: 'Gap1'
  Detected File System: (Unrecognized)
  ID: {218953EE-E6C1-4E37-BC67-3B4C941EA515}
  Size: 300 KB (307200 bytes)
  Start Sector: 6240584, Attributes: 0x1000000000000001
```

Still, I'm seeing no issue with the ISO being written in ISO or DD mode on the systems that I have so I would advise you to:

1. Test DD mode in Rufus if you haven't done so (I get a feeling that you did), and if you get the same issue, get in touch with the Ubuntu maintainers, because Rufus makes absolutely __no__ alterations to the data being written when using DD mode, so, if that doesn't work, it's  a problem between your hardware and Ubuntu, not Rufus.
2. If you find that DD mode works, then I'll need to see a full log of how you created the problematic image (and not just a log of Rufus with a random USB plugged). You may also want to used <kbd>Alt</kbd>-<kbd>D</kbd> beforehand, to delete all the files that Rufus has previously downloaded, including GRUB 2.04's `core.img` as, if that file got corrupted on your computer, it would of course impact GRUB boot.

But at any rate, whatever issue you are having seems to be linked to your specific environment, as I am not seeing any issue booting in BIOS mode.

fenixleon, 2022-04-07:
Thanks Pete. I've included additional information and logs below. 

- Deleted the Rufus generated files e.g. `core.img`.
- Regenerated the portable USB.

When using `dd` mode, no new files are downloaded by Rufus.

The logs below are from creating a new Ubuntu Budgie portable USB. Using `dd ` makes no difference. It stops with the message `GRUB`. It doesn't progress any further.

```
Found USB 2.0 device 'USB FLASH DRIVE USB Device' (1005:B113)
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 3.9GB, Sector size: 512 bytes
Cylinders: 471, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {218953EE-E6C1-4E37-BC64-3B4C941EA515}
Max parts: 248, Start Offset: 32768, Usable = 3880387072 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'ISO9660'
  Detected File System: (Unrecognized)
  ID: {218953EE-E6C1-4E37-BC65-3B4C941EA515}
  Size: 3.0 GB (3190812672 bytes)
  Start Sector: 64, Attributes: 0x1000000000000001
Partition 2:
  Type: EFI System Partition
  Name: 'Appended2'
  Detected File System: FAT12
  ID: {218953EE-E6C1-4E37-BC66-3B4C941EA515}
  Size: 4.1 MB (4333568 bytes)
  Start Sector: 6232120, Attributes: 0x0000000000000000
Partition 3:
  Type: Microsoft Basic Data Partition
  Name: 'Gap1'
  Detected File System: (Unrecognized)
  ID: {218953EE-E6C1-4E37-BC67-3B4C941EA515}
  Size: 300 KB (307200 bytes)
  Start Sector: 6240584, Attributes: 0x1000000000000001

Computing checksum for 'C:\Temp\operating-systems\ubuntu-budgie-21.10-desktop-amd64.iso'...
  MD5:    1dd976ca321958dee838fea8ab6bed76
  SHA1:   9830ab9e8aa49cf55baacf8f8f9a63e25253bf85
  SHA256: 080bb15f3192463c8fc197ed2dfcbf7a26f0168fc38ee4aca54ed4ba15245233

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive3 for exclusive write access
Requesting logical volume handle...
```

The logs below are from creating a new Ubuntu only portable USB. The laptop boots successfully.

```
Found USB 2.0 device 'USB FLASH DRIVE USB Device' (1005:B113)
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 3.9GB, Sector size: 512 bytes
Cylinders: 471, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {218953EE-E6C1-4E37-BC64-3B4C941EA515}
Max parts: 248, Start Offset: 32768, Usable = 3880387072 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'ISO9660'
  Detected File System: (Unrecognized)
  ID: {218953EE-E6C1-4E37-BC65-3B4C941EA515}
  Size: 3.0 GB (3190812672 bytes)
[... 107 lines cut ...]
  Detected Syslinux version: 6.04/20191223 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 20.04.4 LTS amd64'
  Size: 3.1 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: ubuntu-20.04.4-desktop-amd64.iso (3.1 GB)

Computing checksum for 'C:\Temp\operating-systems\ubuntu-20.04.4-desktop-amd64.iso'...
  MD5:    ca0c7f1cf362eef25f54f6cd61c5fdba
  SHA1:   cd90ca5e95a4479cb1e31402988d49357e4ac261
  SHA256: f92f7dca5bb6690e1af0052687ead49376281c7b64fbe4179cc44025965b7d1c

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive3 for exclusive write access
Requesting logical volume handle...
```

fenixleon, 2022-04-07:
Thanks Pete. I tested the ISO image with `dd` using a Linux system as well as the startup disk creator tool. It results in the same issue. I've raised a bug report with Canonical - https://bugs.launchpad.net/ubuntu/+source/grub2/+bug/1968147

pbatard, 2022-04-07:
Thanks for doing this.

One thing you may also want to test, since this looks like a GRUB issue, is to test whether this may have been fixed in GRUB 2.06, as Ubuntu are still using GRUB 2.04, so it's possible this is a know problem that has been addressed in recent GRUB versions.

You can create a __bare__ GRUB 2.06 media with Rufus, by making sure that, first, you are showing the _advanced drive properties_ and then selecting _Grub 2.06_ in the _Boot selection_ dropdown (note that the GRUB option will only appear when _advanced drive properties_ is showing).

Then you can try to boot that media and see if it goes further than a frozen `Welcome to GRUB!` prompt.

Note that, because it's a bare GRUB media, you will still get an error about `/boot/grub/i386-pc/normal.mod` not being found, but if you see that error, it means that GRUB 2.06 should work, and that your issue might have to do with something that was fixed between 2.04 and 2.06.

fenixleon, 2022-04-07:
The bare GRUB media returns "GRUB loading. Welcome to GRUB! error: file '/boot/grub/i386-pc/normal.mod' not found. Entering rescue mode...' followed by `grub rescue>` command

pbatard, 2022-04-07:
Then it looks like GRUB 2.06 might work, as `error: file '/boot/grub/i386-pc/normal.mod' not found.` is expected (since this is a __bare__ GRUB installation, for which you are supposed to provide the various GRUB modules, including `normal.mod`).

Now, the other possibility is that the freezout is with `normal.mod` (which, again is a GRUB binary, as GRUB is not monolithic but is split across multiple binaries), in which case another test you may want to carry out is:

1. Create the bootable USB media in ISO mode using Rufus
2. Delete or rename the `/boot/grub/i386-pc/normal.mod` file you will find on the USB once Rufus is done
3. See if you get the same outcome as with the GRUB 2.06 test above.

If you do get to `grub rescue>` with the test above, then it means that the problem is likely to still affect GRUB 2.06 and is something that has to do with one of the GRUB modules being loaded (which come from the ISO, so, depending on the distro, they may have been customized by the Linux distro maintainers).

If you don't get to `grub rescue>` with the test above, then it means that the problem has likely been fixed between the GRUB 2.04 and GRUB 2.06 releases, in which case you may want to pass that information up to the Ubuntu folks.

fenixleon, 2022-04-07:
Thanks Pete. Deleting the ` /boot/grub/i386-pc/normal.mod` returns `grub rescue>`

pbatard, 2022-04-07:
OK, so it looks like one of the GRUB modules might be incompatible with your hardware. Hopefully the Ubuntu/GRUB people will have some ideas on how to help you troubleshoot further.

fenixleon, 2022-04-07:
Thanks for the help so far.