pbatard/rufus #1907: Fake rufus copy, with a replicate look

UltraToon, 2022-04-15:
When searching up "rufus" in the google search engine, the first result is a replicate website of rufus, that is a scam. 

I verified that the downloads from the website are viruses, and should not be used.

[image]

[image]

pbatard, 2022-04-15:
Please report it to https://safebrowsing.google.com/safebrowsing/report_badware/?hl=en. 

Also, your first "result" is actually an Ad (since it is prefixed by ___Ad___), and therefore this is not actually a result.
Some people are trying to profit off Rufus, and appear to be paying good money to place ads that unscrupulous users like yourself will believe are an actual search result, when they certainly aren't. So please be mindful that not everything displayed in a Google search might be an actual search result.

And if you want to help, report that site to the link I gave you, as there is only so much I can do to try to weed out these fake sites.

pbatard, 2022-04-15:
Addon: You may also want to report this deceptive ad at https://support.google.com/ads/troubleshooter/4578507?hl=en.

closed 2022-04-15 by pbatard

kiencuongpo, 2022-04-16:
https://chrome.google.com/webstore/detail/ublacklist/pncfbmialoiaghdehhbnbhkkgmjanfhe