pbatard/rufus #1909: 1TB Expansion Card (flash drive) categorized as USB Hard Drive

eclecticc, 2022-04-21:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
This is possibly an edge case or semantic ambiguity on the definition of USB Hard Drive.  We have created 1TB and 250GB Expansion Cards, which are small USB flash drives that insert into the Framework Laptop (https://frame.work/blog/storage-expansion-cards).  Aside from Flash capacity, these drives are the same (same USB flash controller).  The 250GB Expansion Card appears with the default Rufus settings, while the 1TB Expansion Card only appears when "List USB Hard Drives" is checked, failing with "Device eliminated because it was detected as a Hard Drive (score 8 > 0)".

Log
---
```
Rufus x86 v3.18.1877 (Portable)
Windows version: Windows 11 Pro, 64-bit (Build 22000.613)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found UAS (USB 3.1) device ' USB DISK 3.2 SCSI Disk Device' (13FE:6500)
Could not get Disk Extents: [0x00000001] Incorrect function.
No logical drive found (unpartitioned?)
Device eliminated because it was detected as a Hard Drive (score 8 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
0 devices found
Found UAS (USB 3.1) device ' USB DISK 3.2 SCSI Disk Device' (13FE:6500)
Device eliminated because it was detected as a Hard Drive (score 8 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found UAS (USB 3.1) device ' USB DISK 3.2 SCSI Disk Device' (13FE:6500)
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
Disk type: FIXED, Disk size: 1TB, Sector size: 512 bytes
Cylinders: 121601, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x321D2693
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Detected File System: FAT12
  Size: 4 MB (4194304 bytes)
  Start Sector: 6969764, Boot: No
Found UAS (USB 3.1) device ' USB DISK 3.2 SCSI Disk Device' (13FE:6500)
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
Disk type: FIXED, Disk size: 1TB, Sector size: 512 bytes
Cylinders: 121601, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x321D2693
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Detected File System: FAT12
  Size: 4 MB (4194304 bytes)
  Start Sector: 6969764, Boot: No
```

eclecticc, 2022-04-21:
Note that the VID/PID for both are 13fe:6500 if it makes sense to add them to vidpid_score.

pbatard, 2022-04-23:
Thanks for the report.

I have to say that I'm leaning into leaving the current Rufus behaviour as is, as pushing for a 1TB drive to be recognized as a removable flash drive media is "pushing it" IMO.

For one thing, your Kingston controller (`13FE` is the VID for Kingston) is presenting the media as `FIXED`, which means that it is intended for precisely the type of media that we usually don't want to list by default in Rufus (because `FIXED` is the direct opposite of `REMOVABLE`, so, clearly, you are using a controller for a media that is not intended to be removed... with removable media). And then there's the matter of the size.

Even if it is possible that one can find a 1 TB "flash drive", I am designing Rufus for the common user, and I don't see them spending a large amount of money on a 1 TB drive just to user Rufus to install an OS. So I will continue to make sure that, when Rufus detects a drive with a large capacity, it does __not__ list it as default, as I don't want to have people lose data from a backup drive just so that I could help your __custom__ scenario.

Therefore, I'm afraid you will have to contend with enabling the listing of additional drives when using Rufus, as I am not planning to add an exception for your model.

closed 2022-04-23 by pbatard