pbatard/rufus #1911: Error:[0xC0030090] The directory is not a subdirectory of the root directory

ApexPhoenix, 2022-04-23:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
For some reason, I tried to install kali/ubuntu into my 32GB flash drive and this error appeared. I investigated other issues, but the answer was unclear to me. Is there any way I can at least install this ``.iso`` file in it?

Error : [0xC0030090] The directory is not a subdirectory of the root directory 
---
```
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from 'C:\Users\Programmer\AppData\Local\Rufus\syslinux-6.04\20200816\' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'H:' as volume mountpoint
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 9.5 GB)
● Creating Linux Persistence Partition (offset: 10192170496, size: 20.0 GB)
Waiting for logical drive to reappear...
Using Debian-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (ext3)...
327679 possible inodes out of 5242875 blocks (block size = 4096)
262143 blocks (5.0%) reserved for the super user
Creating 160 inode sets: [1 marker = 2.0 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Failed to create ext3 'lost+found' dir: Corrupted entry
Re-mounted volume as H: after error

Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3849, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x01E155CD
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 9.5 GB (10191121920 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 20.0 GB (21474819072 bytes)
  Start Sector: 19906583, Boot: No
Closed Windows format prompt
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3849, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x01E155CD
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 9.5 GB (10191121920 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 20.0 GB (21474819072 bytes)
  Start Sector: 19906583, Boot: No
```

pbatard, 2022-04-23:
> - [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Considering that you did __not__ in fact copy the full log (your log does __not__ start by `Rufus version: x.y.z`), I'm going to assume that your subsequent checkmarks are also not truthful, such as using a different drive and testing on a different computer, or even running a bad blocks check.

Please perform the actual steps that were requested above and post a complete log. Especially, I will not expect to see the report of a full bad blocks check for your drive in the log, thank you.

ApexPhoenix, 2022-04-23:
Sorry my bad, I didn't copy it correctly.
```
Rufus x86 v3.18.1877
Windows version: Windows 10 Education, 64-bit (Build 19042.1645)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3849, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0208D4E2
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 29.5 GB (31665946624 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20200816 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Kali Live'
  Size: 3.6 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: kali-linux-2022.1-live-amd64.iso (3.5 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from 'C:\Users\Programmer\AppData\Local\Rufus\syslinux-6.04\20200816\' for Syslinux installation

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'G:' as volume mountpoint
Opened \\.\PhysicalDrive2 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 4.5 GB)
● Creating Linux Persistence Partition (offset: 4823449600, size: 25.0 GB)
Waiting for logical drive to reappear...
Using Debian-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (ext3)...
409599 possible inodes out of 6553598 blocks (block size = 4096)
327679 blocks (5.0%) reserved for the super user
Creating 200 inode sets: [1 marker = 2.5 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Failed to create ext3 'lost+found' dir: Corrupted entry
Re-mounted volume as G: after error

Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3849, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x01943E15
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 4.5 GB (4822401024 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 25.0 GB (26843539968 bytes)
  Start Sector: 9420800, Boot: No
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3849, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x01943E15
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 4.5 GB (4822401024 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 25.0 GB (26843539968 bytes)
  Start Sector: 9420800, Boot: No
Closed Windows format prompt
```

pbatard, 2022-04-26:
Thanks.

You are not the first person to report this issue, but I'm afraid that it's a problem I have never been able to replicate on my side, so I will need your help to try to troubleshoot it.

Basically, the issue is that the ext library we use appear to be able to create a directory, even though it successfully created the root directory beforehand, so I'm going to need a little bit more insight as to where exactly the directory creation process fails.

Can I therefore ask you to please run [this TEST version of Rufus](https://rufus-web.akeo.ie/testing/rufus-3.19_TEST1.exe), where I have enabled more debug output for the directory creation, and post your log?

Note that, for the record, I don't think that persistence will work with Kali anyway, due to every other Linux distro using a completely different method for setting up persistence, and (because it would be way too time consuming to try to add support for all) Rufus only supports persistence for direct Debian or Ubuntu derivatives, which I don't believe Kali is.

For reference this is the log of a test I have carried out, where I tried to replicate your conditions
<details>
<summary>Show log</summary>

```
Rufus x86 v3.18.1877
Windows version: Windows 10 Pro for Workstations, 64-bit (Build 19044.1645)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB device 'Kingmax Kingmax USB Device' (1687:0903) [ID]
1 device found
No volume information for drive 0x88
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3951, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0615CD6C
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 6.3 GB (6734988288 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 24.0 GB (25769802240 bytes)
  Start Sector: 13156322, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20200816 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Kali Live'
  Size: 3.7 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: kali-linux-2021.2-live-amd64.iso (3.6 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from 'C:\Users\nil\AppData\Local\Rufus\syslinux-6.04\20200816\' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Opened \\.\PhysicalDrive8 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 5.3 GB)
[... 77 lines cut ...]
Updating G:\md5sum.txt:
● /boot/grub/grub.cfg
Finalizing, please wait...
G:autorun.inf already exists - keeping it

Found USB device 'Kingmax Kingmax USB Device' (1687:0903) [ID]
Using 'autorun.inf' label for drive G: 'Run Debian GNU/Linux'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3951, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x06171AFB
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 5.3 GB (5661250560 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: ext3
  Size: 25.0 GB (26843539968 bytes)
  Start Sector: 11059178, Boot: No
```
</details>

ApexPhoenix, 2022-04-26:
Here is the new log @pbatard 
```
Rufus x86 v3.19.1886 (Test1)
Windows version: Windows 10 Education, 64-bit (Build 19042.1645)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Unable to send request: The request has timed out.
0 devices found
Found USB 2.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:A200)
NOTE: This device is a USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found USB 2.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:A200)
NOTE: This device is a USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found USB 2.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:A200)
NOTE: This device is a USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3849, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x15701A3A
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 4.5 GB (4822401024 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 25.0 GB (26843539968 bytes)
  Start Sector: 9420800, Boot: No
Found USB 2.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:A200)
NOTE: This device is a USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3849, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x15701A3A
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 4.5 GB (4822401024 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 25.0 GB (26843539968 bytes)
[... 166 lines cut ...]
  Detected File System: (Unrecognized)
  Size: 25.0 GB (26843539968 bytes)
  Start Sector: 9420800, Boot: No
Found USB 2.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:A200)
NOTE: This device is a USB 3.0 device operating at lower speed...
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
2 devices found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3849, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x05D53C6E
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 4.5 GB (4822401024 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 25.0 GB (26843539968 bytes)
  Start Sector: 9420800, Boot: No
Closed Windows format prompt
```

pbatard, 2022-04-27:
Much appreciated!

We're definitely getting somewhere as that `retval != EXT2_ET_FILE_NOT_FOUND` line from the log is telling us that the issue is with [the `ext2fs_lookup()` call failing for some reason](https://github.com/pbatard/rufus/blob/48574f10d6fda0d2f4ee01458b2543fc29165af1/src/ext2fs/mkdir.c#L159-L167).

This is quite strange, since none of the previous calls that should have created the directory returned an error, which means that the directory should exist and therefore be locatable.

I guess I'm going to need a bit more data from that call, if that's okay with you, so can I please ask you to try again, with [this TEST 2 build](https://rufus-web.akeo.ie/testing/rufus-3.19_TEST2.exe), and paste your log again?

ApexPhoenix, 2022-04-27:
Here is the log for test 2 @pbatard 
````
Rufus x86 v3.19.1886 (Test2)
Windows version: Windows 10 Education, 64-bit (Build 19042.1645)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:A200)
NOTE: This device is a USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3849, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x05D53C6E
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 4.5 GB (4822401024 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 25.0 GB (26843539968 bytes)
  Start Sector: 9420800, Boot: No
Unable to send request: The request has timed out.
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20200816 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Kali Live'
  Size: 3.6 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: kali-linux-2022.1-live-amd64.iso (3.5 GB)
Found USB 2.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:A200)
NOTE: This device is a USB 3.0 device operating at lower speed...
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
2 devices found
Disk type: FIXED, Disk size: 1TB, Sector size: 512 bytes
Cylinders: 121601, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x018B03F6
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 676.5 GB (726399668736 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 255.0 GB (273804152832 bytes)
  Start Sector: 1418751401, Boot: No
Found USB 2.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:A200)
[... 114 lines cut ...]
Found USB 2.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:A200)
NOTE: This device is a USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3849, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0AB3453C
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 9.5 GB (10191121920 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 20.0 GB (21474819072 bytes)
  Start Sector: 19906583, Boot: No
Closed Windows format prompt
````

pbatard, 2022-04-28:
Thanks again. So this confirms that `ext2fs_lookup` fails with code `EXT2_ET_DIR_CORRUPTED`.

This might be real tricky to troubleshoot without being able to replicate the error locally, especially as I am not familiar at all with how the ext2/3 file system handles directory creation (I am mostly reusing this existing ext2/3 file system library).

I guess that, as long as you're willing too test, we can try to dive deeper into the rabbit hole to try to locate how the error comes to manifest itself, so can I ask you again to try [TEST build 3](https://rufus-web.akeo.ie/testing/rufus-3.19_TEST3.exe), where I added a few more messages, and report? Thanks.

ApexPhoenix, 2022-04-29:
Rufus x86 v3.19.1886 (Test2)
Windows version: Windows 10 Education, 64-bit (Build 19042.1645)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:A200)
NOTE: This device is a USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3849, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x05D53C6E
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 4.5 GB (4822401024 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 25.0 GB (26843539968 bytes)
  Start Sector: 9420800, Boot: No
Unable to send request: The request has timed out.
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20200816 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Kali Live'
  Size: 3.6 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: kali-linux-2022.1-live-amd64.iso (3.5 GB)
Found USB 2.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:A200)
NOTE: This device is a USB 3.0 device operating at lower speed...
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
2 devices found
Disk type: FIXED, Disk size: 1TB, Sector size: 512 bytes
Cylinders: 121601, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x018B03F6
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 676.5 GB (726399668736 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 255.0 GB (273804152832 bytes)
  Start Sector: 1418751401, Boot: No
Found USB 2.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:A200)
NOTE: This device is a USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
[... 109 lines cut ...]
  Size: 20.0 GB (21474819072 bytes)
  Start Sector: 19906583, Boot: No
Found USB 2.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:A200)
NOTE: This device is a USB 3.0 device operating at lower speed...
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
min_drive_size = 31666995200
2 devices found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3849, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x14EBAA4B
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 9.5 GB (10191121920 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 20.0 GB (21474819072 bytes)
  Start Sector: 19906583, Boot: No
Closed Windows format prompt
````

ApexPhoenix, 2022-04-29:
@pbatard why do you think the error is happening in the first place?

pbatard, 2022-05-01:
> why do you think the error is happening in the first place?

I have absolutely no idea.

Again, I did not write the code that deals with ext2/ext3 formatting and directory creation. I am merely reusing it (it's the official code used by Linux, and the changes needed to port it to Windows were minimal, so it would be expected to work as well as on Linux). And only a very limited number of people seem to be running into these issues (99.9% of people who create persistent partitions never see this error).

All I can say from your latest log is that after the code created the `lost+found` directory and is finalizing the directory creation operation by linking it to the parent directory (so that the newly created directory can actually be listed) it finds an ext2fs record length of zero on the newly created directory for some reason. And that's about all I can say about it for now...

Unfortunately, this means that we still need to keep digging, so, again, can I ask you to please run [TEST build 4](https://rufus-web.akeo.ie/testing/rufus-3.19_TEST4.exe) that adds yet a bit more instrumentation on the error trail we are following?

ApexPhoenix, 2022-05-02:
@pbatard Here is the updated log for (Test 4)
```
Rufus x86 v3.19.1888 (Test4)
Windows version: Windows 10 Education, 64-bit (Build 19042.1645)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:A200)
NOTE: This device is a USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
min_drive_size = 31666995200
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3849, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0292E03D
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 9.5 GB (10191121920 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 20.0 GB (21474819072 bytes)
  Start Sector: 19906583, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20200816 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Kali Live'
  Size: 3.6 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: kali-linux-2022.1-live-amd64.iso (3.5 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from 'C:\Users\Programmer\AppData\Local\Rufus\syslinux-6.04\20200816\' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'H:' as volume mountpoint
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 9.5 GB)
● Creating Linux Persistence Partition (offset: 10192170496, size: 20.0 GB)
Waiting for logical drive to reappear...
Using Debian-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (ext3)...
327679 possible inodes out of 5242875 blocks (block size = 4096)
262143 blocks (5.0%) reserved for the super user
Creating 160 inode+ sets: [1 marker = 2.0 set(s)]
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
dirent->rec_len = 0, fs->blocksize = 4096
ext2fs_process_dir_block: 0 < 8
retval != EXT2_ET_FILE_NOT_FOUND
ext2fs_lookup(fs, 00000002, lost+found, 10, 0, &scratch_ino) = 0x7F2BB723
Failed to create ext3 'lost+found' dir: Corrupted entry
Re-mounted volume as H: after error

Found USB 2.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:A200)
NOTE: This device is a USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
min_drive_size = 31666995200
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3849, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x02938B91
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 9.5 GB (10191121920 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 20.0 GB (21474819072 bytes)
  Start Sector: 19906583, Boot: No
Closed Windows format prompt
Found USB 2.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:A200)
NOTE: This device is a USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
min_drive_size = 31666995200
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3849, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x02938B91
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 9.5 GB (10191121920 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 20.0 GB (21474819072 bytes)
  Start Sector: 19906583, Boot: No
```

ventoy, 2022-05-08:
@pbatard 
Just for referrence! 
The USB stick manufacture information is `General UDisk USB Device` not as normal as `Kingston .../Sandisk...`
So maybe we should consider the possibility that it is fake or faulty or contains some bad sectors.
With some fake or faulty USB stick, when you write data to some position and read it back, the data does not match, especially when the position is at the end of the USB stick.
https://www.ventoy.net/en/doc_fake_usb.html

pbatard, 2022-05-08:
Hi LongPanda!

Somehow I don't think this is a fake drive issue, though this is of course always a possibility, considering that one of the thing that was explicitly asked of OP to do (and I asked for that again after they opened the issue) was to run a bad blocks check using Rufus, and the bad blocks check from Rufus is specially designed to detect and report fake drives.

So, if OP did follow what they have been instructed to do from the very beginning, we should be able to rule out that the issue is cause by a fake drive.

Plus, the error is way too consistent to look like something that could be cause by a fake drive, and I've had quite a few reports for this error to make me think that this isn't related to using fake drives.

Still, to be completely sure, and since it never hurts, I'd like to ask OP to please run a bad blocks check, as detailed in step 5 of the check list at the top of this issue (which, I'll remind them, they __did__ check and should therefore already have carried out at least once), and post the log with the result of the bad blocks check, so that we can be 100% certain that we're not dealing with a fake drive.

For the record, I'm still looking into this, but of course, since I still can't replicate it, progress is going to be very slow.

All I can say at this stage is that it appears that when creating the directory entry for the `lost+found` directory, the record length (`dirent->rec_len`) is apparently being set to `0` on write, in OP's environment, whereas it should be set to `12`...

ventoy, 2022-05-08:
@pbatard 
That's OK.

@ApexPhoenix 
I think some comparative test may be helpful.
1. Test with different persistence partition size (e.g.  2GB/4GB/8GB ...)
2. Test on another USB stick if you have one.

ApexPhoenix, 2022-05-09:
@ventoy 
I did that already, it only works with my hard drive of 10 terabytes, it is probably because of my operating system.

pbatard, 2022-05-09:
Can you please still post a log with the results of the bad blocks check for the device that fails? At this stage, I'd __really__ like to see the output from a bad blocks check log. Thanks.

ApexPhoenix, 2022-05-14:
@pbatard Here is TEST 5, with bad blocks check. Sorry, for taking to much I was sick during this week!

```
Rufus bad blocks check started on: 2022.05.12 18:21:31Block 969: corruption error
Block 970: corruption error
Block 971: corruption error
Block 972: corruption error
Block 973: corruption error
Block 974: corruption error
Block 975: corruption error
Block 976: corruption error
Block 977: corruption error
Block 978: corruption error
Block 979: corruption error
Block 980: corruption error
Block 981: corruption error
Block 982: corruption error
Block 983: corruption error
Block 984: corruption error
Block 985: corruption error
Block 986: corruption error
Block 987: corruption error
Block 988: corruption error
Block 989: corruption error
Block 990: corruption error
Block 991: corruption error
Block 992: corruption error
Block 993: corruption error
Block 994: corruption error
Block 995: corruption error
Block 996: corruption error
Block 997: corruption error
Block 998: corruption error
Block 999: corruption error
Block 1000: corruption error
Block 1001: corruption error
Block 1002: corruption error
Block 1003: corruption error
Block 1004: corruption error
Block 1005: corruption error
Block 1006: corruption error
Block 1007: corruption error
Block 1008: corruption error
Block 1009: corruption error
Block 1010: corruption error
Block 1011: corruption error
Block 1012: corruption error
Block 1013: corruption error
Block 1014: corruption error
Block 1015: corruption error
Block 1016: corruption error
Block 1017: corruption error
Block 1018: corruption error
Block 1019: corruption error
Block 1020: corruption error
Block 1021: corruption error
Block 1022: corruption error
Block 1023: corruption error
Block 1024: corruption error
Block 1025: corruption error
Block 1026: corruption error
Block 1027: corruption error
Block 1028: corruption error
Block 1029: corruption error
Block 1030: corruption error
Block 1031: corruption error
Block 1032: corruption error
Block 1033: corruption error
Block 1034: corruption error
Block 1035: corruption error
[... 228 lines cut ...]
Block 1264: corruption error
Block 1265: corruption error
Block 1266: corruption error
Block 1267: corruption error
Block 1268: corruption error
Block 1269: corruption error
Block 1270: corruption error
Block 1271: corruption error
Block 1272: corruption error
Block 1273: corruption error
Block 1274: corruption error
Block 1275: corruption error
Block 1276: corruption error
Block 1277: corruption error
Block 1278: corruption error
Block 1279: corruption error
Too many bad blocks, aborting test
Check completed: 311 bad block(s) found

  0 read error(s)

  0 write error(s)

  311 corruption error(s)Rufus bad blocks check ended on: 2022.05.13 15:48:15
```

ApexPhoenix, 2022-05-14:
Also there is a new message ```Error: 0xC0030570 The file or directory is corrupted and unreadable

pbatard, 2022-05-14:
I'm sorry to hear that you were sick. But dude, seriously?!?

I quote that you __DID__ check the following when you opened the issue:
> -  [x] I ran a bad blocks check, by clicking Show advanced format options then Check device for bad blocks, and confirmed that my USB is not defective.

Then, because I had doubts that you actually did it, I asked you __very explicitly__ to make sure you ran a bad blocks check:

> Please perform the actual steps that were requested above and post a complete log. Especially, I will expect to see the report of a full bad blocks check for your drive in the log, thank you.

It is clear however that you did not perform any of these when asked, and as a result, wasted everybody's time, including yours.

Oh, and I'm going to go on a limb and assume that you also _lied_ when you say you did the following:

> - [x] I also tried one or more of the following:
>  - [x] Using a different USB drive.

Your drive __IS__ defective, and that is the reason for your issue. And please, next time you want to receive help, please follow the advice that is being given to you instead of making everybody go on a wild goose chase because you can't be bothered to run a simple bad blocks check.

closed 2022-05-14 by pbatard

ventoy, 2022-05-14:
Again it proves that if an USB stick's information doesn't contains the name of the manufacturer like `Kingston /Sandisk...`
and only contains a common name like `General UDisk xxx/Generic USB xxx/...` it has high posibility to be defective.

ApexPhoenix, 2022-05-15:
I'm really sorry, I might have just misclicked. Therefore, can someone please tell me how to fix it? @ventoy @pbatard