pbatard/rufus #1915: 错误 0xc00305b4

wolaopoali, 2022-04-29:
Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 28.6 GB)
Could not access first GUID volume: [0x00000012] There are no more files.
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 28.6 GB)
Waiting for logical drive to reappear...
Timeout while waiting for logical drive
Logical drive was not found - aborting

Found USB 3.0 device 'SanDisk Ultra USB 3.0 USB Device' (0781:5591)
Could not access first GUID volume: [0x00000012] There are no more files.
No logical drive found (unpartitioned?)
1 device found
Could not access first GUID volume: [0x00000012] There are no more files.
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3738, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {00000000-0000-0000-0000-000000000000}
Max parts: 128, Start Offset: 17408, Usable = 30751965696 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {CAC802F4-9EBA-4ABD-981C-9771288647A4}
  Size: 28.6 GB (30750934528 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
rufus错误 0xc00305b4

pbatard, 2022-04-29:
Please follow the check list and don't truncate the log.

You truncated the log, so I don't have enough data, which means that I simply cannot help you.

Unless you can provide a complete log, I will have no choice but to close this issue.

__Following the check list is NOT optional!!__

pbatard, 2022-04-29:
Also, your error looks like a pure hardware issue with your flash drive, so please post a log with the result of the bad blocks check (the check list, that you ignored, told you how to use Rufus to check for bad blocks).

pbatard, 2022-05-02:
Closing on account that OP did not properly follow the check list.

closed 2022-05-02 by pbatard