pbatard/rufus #1916: "Disk or ISO image (Please choose)" Incorrect translation of the Polish string.

temporaryFuckMicrosoft, 2022-04-29:
﻿

Checklist
---------
- [ no ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ no, but there's just 13 issues nothing resembled this ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ no ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ no ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ no ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ no ] I also tried one or more of the following:
  - [ USB drive shows up, no ] Using a different USB drive.
  - [ no ] Plugging the USB into a different port.
  - [ no I literally can only run a virtual machine rn ] Running Rufus on a different computer.
- [ no because it won't let me select "Disk" as a source ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Unable to select a DVD drive as the source for rufus or it's an incorrect translation.

Missleading translation.

"Dysk lub obraz ISO" is equal in meaning to "Select DVD/BluRay/CD Disc or Diskette or Hard Drive or Partition or ISO image";

"Obraz dysku lub ISO" == "The image of the drive or ISO"

In Polish the word "Image"("Obraz") needs to come before the thing that is being enumerated, if it is after the first enumeration then it counts as the thing added to the individual Nth item rather than all N items.

Log
---
```
<FULL LOG>
CRITICAL ERROR: Human read a piece of string in his native tongue, he is going to make the wrong assumption, better watch out.
```

pbatard, 2022-04-30:
Obviously I don't speak Polish, and I also can't blame translators for making mistakes, so I will ask you to please tell me exactly what you'd like to see knowing that what you are able to select is a DVD image or a BluRay image or a CD image or a Diskette image or a Hard Drive image or an ISO image (which is really the same as the first 3), and not a physical optical disc media inserted into a physical DVD drive.

Basically, what you want to do (select a physical optical media that is inserted into a physical DVD drive) will not work, because it is not a feature that Rufus provides.

So, knowing this, what would you suggest the translation should be?

temporaryFuckMicrosoft, 2022-04-30:
Any of the following:

Wybierz obraz dysku lub płyty (Choose an image of a disk[raw image] or disc[iso])
Obraz płyty lub dysku (proszę wybrać) (Image of a disc or disk (please choose))

temporaryFuckMicrosoft, 2022-04-30:
#1917

commit 4cdbad28 (2022-05-02): [loc] update Polish translation

* Closes #1916
* Also removes wininet.dll from MSVC vcxproj.
closed 2022-05-02 by pbatard