pbatard/rufus #1918: Rufus persistent partition too slow

eddisch, 2022-05-01:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When i use persistent partition size (no matter how much) rufus is soooo slow that it throws a error that explorer.exe (r) wants to use the drive. I tried alt+, and tried to terminate explorer but in 1 hour it moved to (formatting) 14.0%!

Log
---
```
Rufus x86 v3.18.1877
Windows version: Windows 11 Pro, 64-bit (Build 22000.613)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
Detected installed Windows Language Pack for 0x0419 (Russian (Русский))
Localization set to 'ru-RU'
NOTE: The Russian (Русский) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 2.0 device 'Generic Mass Storage USB Device' (31C0:1234)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 64 ГБ, Sector size: 512 bytes
Cylinders: 7699, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from 'C:\Users\eddis\AppData\Local\Rufus\syslinux-6.04\20191223\' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 3.0 GB)
● Creating Linux Persistence Partition (offset: 3204424192, size: 56.0 GB)
Waiting for logical drive to reappear...
Using Ubuntu-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (ext3)...
917503 possible inodes out of 14680063 blocks (block size = 4096)
734003 blocks (5.0%) reserved for the super user
Creating 448 inode sets: [1 marker = 5.6 set(s)]
+++
```

eddisch, 2022-05-01:
Btw i can help translate russian.

pbatard, 2022-05-01:
> `Found USB 2.0 device 'Generic Mass Storage USB Device' (31C0:1234)`

There's you're problem: You are using a USB 2.0 drive, so the persistent partition speeds are always going to be slow, because there is a lot of writing involved.

Please use a USB 3.0 drive, with decent write speed, connected to a USB 3.0 port. Creating `ext3` partitions from a Windows environment is a slow operation by nature, and you're also likely to experience slow speeds when running Ubuntu unless you use a USB 3.0 drive.

marcosfrm, 2022-05-01:
EXT3 is slow to create even on Linux with native tools. Pete, does this code have licensing problems?

https://github.com/maharmstone/btrfs

There is a Windows `mkfs.btrfs` implementation there somewhere... it is a much faster FS to create.

eddisch, 2022-05-01:
And btw can i help translate?
```
NOTE: The Russian (Русский) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
```

closed 2022-05-01 by eddisch