pbatard/rufus #1919: Rufus builds Windows To Go USB install but it blue screens and Windows won't start on target.

g-reno, 2022-05-01:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.

- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
---------------
HDD died in my HP dv7t-7000 laptop taking with it my Windows 7 Home Premium install and my Fedora install.  BIOS is set for BIOS boot.
This next install of OS's I want to have Windows 7 Home Premium on a USB separate from the HDD.  So I looked into how to do this and Rufus looked like a good option.
I used a Windows 10 tablet I have to run Rufus and build the Window To Go USB3 flash drive.  I checksummed the Windows 7 Home Premium ISO I had downloaded and found it in my internet search as being official ISO.
I ran Rufus using MBR and BIOS / UEFI, NTFS 4096.  It successfully installed onto the USB3 flash drive.  But when I run it on my HP laptop it gives a very brief blue screen flash then shows the Windows Error Recovery screen with Safe Mode boot options.
So I reran Rufus this time with a USB3 HDD plugged in and it again successfully installed onto the USB3 HDD.  But again when I r
un it on my HP laptop it blue screens and shows the Windows Error Recovery screen.
What can I do to get it to run?  If it's drivers is there a special place I can put drivers?

Log
---
Log is massive.  Attached it instead of pasting it.
```
<FULL LOG>
```
[rufus.log](https://github.com/pbatard/rufus/files/8599448/rufus.log)

g-reno, 2022-05-01:
Screenshot of blue screen.
[image]

JonnyTech, 2022-05-01:
https://github.com/pbatard/rufus/issues/1323#issuecomment-498614706

g-reno, 2022-05-01:
This is something other than USB3 issue.  My laptop has USB2 ports and USB3 ports.
The blue screen occurs no matter what USB version port you try the flash drive in.
Yes, USB3 would be nice on boot but it's not a requirement on my laptop.

pbatard, 2022-05-02:
1. [Per the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Windows_XP_Windows_Vista_Windows_7_NO_LONGER_SUPPORTED), I no longer support anything that has to do with Windows 7 (because my time is limited and it's just not worth using it on obsoleted OSes, as any work done here is bound to help less and less people).
2. Rufus creates Windows To Go drive in the same manner as the one Microsoft officially advises, therefore, unless you encounter an issue during creation of the drive, whatever problem you are facing is a Microsoft problem, not a Rufus problem.

As a result I will close this issue. I suggest you post on forums that are dedicated to running Windows 7 (if you can still find any).

closed 2022-05-02 by pbatard