pbatard/rufus #1920: Install Windows To Go On an External Hard Disk

OverloadedTech, 2022-05-04:
Hi, i want to install windows to go using Rufus on an external hard drive. But i can't view it. Where is the key combination to show external hard drives?

JonnyTech, 2022-05-04:
https://github.com/pbatard/rufus/wiki/FAQ#why-doesnt-rufus-list-external-hdds-by-default

> Of course, you can still list those external drives if you know what you are doing, by checking the relevant option under Advanced drive properties. But this feature is not something that is enabled by default.

OverloadedTech, 2022-05-04:
Ok thanks where is the option to activate

pbatard, 2022-05-04:
https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected

And next time follow the check list. It is aimed __precisely__ at people like you, who think it's only there as an inconvenience rather than something __that would have helped them solve their issue in less than 5 minutes__...

closed 2022-05-04 by pbatard