pbatard/rufus #1921: Format Linux persistence partition as Btrfs

marcosfrm, 2022-05-04:
From https://github.com/pbatard/rufus/issues/1918#issuecomment-1114305805.

https://github.com/maharmstone/btrfs

WinBtrfs is released under the LGPLv3, which will not be an issue for Rufus.

Filesystem creation routines appear to be compartmented in `src/ubtrfs/ubtrfs.c` (plus four headers: `src/btrfs.h`, `src/btrfsioctl.h`, `src/crc32c.h`, `src/xxhash.h`). It would require finding out the sequence called from `src/mkbtrfs/mkbtrfs.c` and replicate it.

With exception of RHEL, Btrfs is supported on most major Linux distributions. And it is *much* faster to create compared to EXT3.

400 GB persistence partition (external HDD case, USB 2.0):

```
Rufus x86 v3.18.1877
Windows version: Windows 10 Pro, 64-bit (Build 19044.1645)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0416 (pt-BR)
Will use default UI locale 0x0416
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pt-BR'
NOTE: The Portuguese Brazilian (Português do Brasil) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 2.0 device 'ST500LM0 12 HN-M500MBB USB Device' (174C:5106)
NOTE: This device is a USB 3.0 device operating at lower speed...
No logical drive found (unpartitioned?)
Device eliminated because it was detected as a Hard Drive (score 16 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found USB 2.0 device 'ST500LM0 12 HN-M500MBB USB Device' (174C:5106)
NOTE: This device is a USB 3.0 device operating at lower speed...
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x81
Disk type: FIXED, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60801, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
Drive does not have a Boot Marker
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Fedora-S-dvd-x86_64-36'
  Size: 773.1 MB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: Fedora-Server-netinst-x86_64-36_Beta-1.4.iso (668 MB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from 'C:\Users\Sabugo\AppData\Local\Rufus\syslinux-6.04\' for Syslinux installation

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'E:' as volume mountpoint
Warning: Could not delete partition(s): Windows error code 0x000003F0 (FormatMessage error code 0x00003AFC)
Opened \\.\PhysicalDrive1 for shared write access
Requesting logical volume handle...
No logical drive found (unpartitioned?)
Could not unmount drive: Windows error code 0x00000006
Trying to continue regardless...
Analyzing existing boot records...
Drive does not have a Boot Marker
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 65.8 GB)
● Creating Linux Persistence Partition (offset: 70611142144, size: 400.0 GB)
Waiting for logical drive to reappear...
Using Debian-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (ext3)...
6553599 possible inodes out of 104857593 blocks (block size = 4096)
5242879 blocks (5.0%) reserved for the super user
Creating 3200 inode sets: [1 marker = 40.0 set(s)]
++++Closed Windows format prompt
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Creating 131072 journal blocks: [1 marker = 1638.4 block(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Created 'persistence.conf' file
Done
Formatting (Large FAT32)...
Opened \\?\Volume{032b2a62-0000-0000-0000-100000000000} for exclusive write access
Size : 65.8 GB 137910339 sectors
Cluster size 32768 bytes, 512 bytes per sector
Volume ID is f1a:1152
32 Reserved sectors, 16831 sectors per FAT, 2 FATs
2154322 Total clusters
2154321 Free clusters
Clearing out 33758 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 16863
Writing Partition Boot Record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting label...
Format completed.
Writing Master Boot Record...
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume \\?\Volume{032b2a62-0000-0000-0000-100000000000}\
Installing Syslinux 6.04...
Opened \\?\Volume{032b2a62-0000-0000-0000-100000000000} for shared write access
Using existing 'C:\Users\Sabugo\AppData\Local\Rufus\syslinux-6.04\ldlinux.sys' ✓
Using existing 'C:\Users\Sabugo\AppData\Local\Rufus\syslinux-6.04\ldlinux.bss' ✓
Successfully wrote 'ldlinux.sys'
Successfully wrote Syslinux boot record
Successfully remounted \\?\Volume{032b2a62-0000-0000-0000-100000000000}\ as E:
Extracting files...
Image is an ISO9660 image
This image will be extracted using Joliet extensions (if present)
Extracting: E:\.discinfo (28 bytes)
Extracting: E:\EFI\BOOT\BOOT.conf (1.3 KB)
Extracting: E:\EFI\BOOT\BOOTIA32.EFI (723.0 KB)
Extracting: E:\EFI\BOOT\BOOTX64.EFI (906.8 KB)
Extracting: E:\EFI\BOOT\fonts\unicode.pf2 (2.3 MB)
Extracting: E:\EFI\BOOT\grub.cfg (1.3 KB)
  Patched E:\EFI\BOOT\grub.cfg: 'Fedora-S-dvd-x86_64-36' ➔ 'TEST'
  Patched E:\EFI\BOOT\grub.cfg: 'inst.stage2' ➔ 'inst.repo'
Extracting: E:\EFI\BOOT\grubia32.efi (1.6 MB)
Extracting: E:\EFI\BOOT\grubx64.efi (2.5 MB)
Extracting: E:\EFI\BOOT\mmia32.efi (660.2 KB)
Extracting: E:\EFI\BOOT\mmx64.efi (830.1 KB)
Extracting: E:\Fedora-Legal-README.txt (2.5 KB)
Extracting: E:\LICENSE (1 KB)
Extracting: E:\images\efiboot.img (9.9 MB)
Extracting: E:\images\install.img (520.0 MB)
Extracting: E:\images\macboot.img (20.8 MB)
Extracting: E:\images\pxeboot\initrd.img (95 MB)
Extracting: E:\images\pxeboot\vmlinuz (11.3 MB)
Extracting: E:\isolinux\boot.cat (2 KB)
Extracting: E:\isolinux\boot.msg (84 bytes)
Extracting: E:\isolinux\grub.conf (283 bytes)
Extracting: E:\isolinux\initrd.img (95 MB)
Extracting: E:\isolinux\isolinux.bin (38 KB)
Extracting: E:\isolinux\isolinux.cfg (2.7 KB)
  Patched E:\isolinux\isolinux.cfg: 'Fedora-S-dvd-x86_64-36' ➔ 'TEST'
  Patched E:\isolinux\isolinux.cfg: 'inst.stage2' ➔ 'inst.repo'
Extracting: E:\isolinux\ldlinux.c32 (113.4 KB)
Extracting: E:\isolinux\libcom32.c32 (175.7 KB)
Extracting: E:\isolinux\libutil.c32 (23.6 KB)
Extracting: E:\isolinux\splash.png (118 bytes)
Extracting: E:\isolinux\vesamenu.c32 (26.1 KB)
Extracting: E:\isolinux\vmlinuz (11.3 MB)
Created: E:\syslinux.cfg → /isolinux/isolinux.cfg
Finalizing, please wait...

Found USB 2.0 device 'ST500LM0 12 HN-M500MBB USB Device' (174C:5106)
NOTE: This device is a USB 3.0 device operating at lower speed...
1 device found
Disk type: FIXED, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60801, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x032B2A62
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 65.8 GB (70610093568 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: ext3
  Size: 400.0 GB (429496704000 bytes)
  Start Sector: 137912387, Boot: No
```

~ 13 minutes creating EXT3 filesystem.

Btrfs on the same partition:

```
PS C:\Windows\system32> Measure-Command { mkbtrfs.exe \Device\Harddisk1\Partition2 NEW | Out-Default }
Completed successfully.

Days              : 0
Hours             : 0
Minutes           : 0
Seconds           : 0
Milliseconds      : 444
Ticks             : 4449323
TotalDays         : 5,14967939814815E-06
TotalHours        : 0,000123592305555556
TotalMinutes      : 0,00741553833333333
TotalSeconds      : 0,4449323
TotalMilliseconds : 444,9323
```

One problem is how `persistence.conf` would be created.

Rufus TODO is already heavy, yeah, but one more entry does not hurt...

pbatard, 2022-05-04:
> With exception of RHEL

That's a major deal breaker right here (yes, even if we don't officially support persistence for RH derivatives).

> One problem is how `persistence.conf` would be created.

Another deal breaker.

I'm also not that confident that, whereas `btrfs` might be faster for file system creation, it is better in the long run for persistence partition. That's actually part of the reason why we're not bothering exploring `ext4` formatting, because with all its bells and whistles, `ext4` (and most likely `btrfs`) doesn't seem to be that well designed for a file system that runs on a device with low random I/O speeds, like consumer USB Flash Drives. Instead those appear to be designed for SSD media, which makes sense, but may result in additional wear and tear for already unreliable UFDs. I've actually debated whether we needed the journalling features of `ext3` when introducing the feature, so, if anything, we might as well switch to using `ext2` over `ext3`, as it'd solve the creation speed issue without having to integrate new untested code.

IMO, what would be more productive would be to investigate switching to using [the new Windows backend code that has recently been integrated into ext2fs](https://github.com/tytso/e2fsprogs/commits?author=alvarez86) (since I never got a chance to submit my update as a patch, and someone else also did a Windows implementation on their own)... and that's already been on my TODO list for some time.

No, I'm not saying that I'm never going to explore `btrfs` support, but I'm afraid it's way too premature for the time being, especially if part of the reason is to avoid an issue, without understanding what is causing the issue in the first place...

As such, I will close this request for the time being.

closed 2022-05-04 by pbatard

marcosfrm, 2022-05-04:
Understandable. 👍