pbatard/rufus #1922: Non bootable + ext2/3 fails

marcosfrm, 2022-05-04:
Selecting "Boot selection -> Non bootable" then ext2 or ext3 as "File system" does not work:

(drive zeroed before running Rufus)
```
Rufus x86 v3.18.1877
Windows version: Windows 10 Pro, 64-bit (Build 19044.1645)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0416 (pt-BR)
Will use default UI locale 0x0416
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pt-BR'
NOTE: The Portuguese Brazilian (Português do Brasil) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 2.0 device 'ST500LM0 12 HN-M500MBB USB Device' (174C:5106)
NOTE: This device is a USB 3.0 device operating at lower speed...
No logical drive found (unpartitioned?)
Device eliminated because it was detected as a Hard Drive (score 16 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found USB 2.0 device 'ST500LM0 12 HN-M500MBB USB Device' (174C:5106)
NOTE: This device is a USB 3.0 device operating at lower speed...
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x81
Disk type: FIXED, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60801, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
Drive does not have a Boot Marker

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'E:' as volume mountpoint
Warning: Could not delete partition(s): Windows error code 0x000003F0 (FormatMessage error code 0x00003AFC)
Opened \\.\PhysicalDrive1 for shared write access
Requesting logical volume handle...
No logical drive found (unpartitioned?)
Could not unmount drive: Windows error code 0x00000006
Trying to continue regardless...
Analyzing existing boot records...
Drive does not have a Boot Marker
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 465.8 GB)
Waiting for logical drive to reappear...
Timeout while waiting for logical drive
Logical drive was not found - aborting

Found USB 2.0 device 'ST500LM0 12 HN-M500MBB USB Device' (174C:5106)
NOTE: This device is a USB 3.0 device operating at lower speed...
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x81
Disk type: FIXED, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60801, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0001E7BB
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 465.8 GB (500106813440 bytes)
  Start Sector: 2048, Boot: No
```

Should this be allowed?

pbatard, 2022-05-04:
Not seeing that issue on my side.

The fact that you are getting a:
```
Warning: Could not delete partition(s): Windows error code 0x000003F0 (FormatMessage error code 0x00003AFC)
```
long before before the formatting operation starts would seem to indicate that this is an issue with your environment.

Can you please try a different drive, different port and different machine if possible.

Be very mindful of assuming that because you are seeing an error, everybody will see the same...

For reference, here's my log on Windows 10 19044.1645 for formatting to `ext2` (and I'm seeing no issue with `ext3` either):

```
Rufus x86 v3.18.1877
Windows version: Windows 10 Pro for Workstations, 64-bit (Build 19044.1645)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB device 'SanDisk Extreme Pro USB Device' (0781:5588) [ID]
1 device found
No volume information for drive 0x88
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15567, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0F7F2F8C
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 109.2 GB (117305233920 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 10.0 GB (10737409536 bytes)
  Start Sector: 229113833, Boot: No

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive8 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 119.2 GB)
Waiting for logical drive to reappear...
Notice: Using physical device to access partition data
Formatting (ext2)...
1953776 possible inodes out of 31260416 blocks (block size = 4096)
1563020 blocks (5.0%) reserved for the super user
Creating 954 inode sets: [1 marker = 11.9 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Done
Writing Master Boot Record...
Using Zeroed MBR
Found volume \\?\Volume{4d9c5acf-c9fa-11ec-8d63-88d7f6e01c0c}\
Successfully remounted \\?\Volume{4d9c5acf-c9fa-11ec-8d63-88d7f6e01c0c}\ as F:

Found USB device 'SanDisk Extreme Pro USB Device' (0781:5588) [ID]
1 device found
No volume information for drive 0x88
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15567, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: GNU/Linux (0x83)
  Detected File System: ext2
  Size: 119.2 GB (128042663936 bytes)
  Start Sector: 2048, Boot: No
Closed Windows format prompt
```

marcosfrm, 2022-05-04:
> Not seeing that issue on my side.
>
> The fact that you are getting a:
>
> ```
> Warning: Could not delete partition(s): Windows error code 0x000003F0 (FormatMessage error code 0x00003AFC)
> ```
>
> long before [...]

Windows is clean, fairly recent (and up-to-date) installation.

> 
> Can you please try a different drive, different port and different machine if possible.
> 

Same thing on all 3 USB ports. But the issue indeed does not happen with USB flash drives (tested two, 4 GB and 16 GB). I was using the external HDD to test EXT/Btrfs since it is much faster...

> Be very mindful of assuming that because you are seeing an error, everybody will see the same...

Right.

> Also, why is your USB 3.0 operating at USB 2.0 speed? If it's plugged on a USB 3.0 port, it should be running at USB 3.0 speed, so the fact that it doesn't would tend to indicate a hardware issue...

It is an old Acer laptop with only USB 2.0 ports.

I will try it on another machine and report back later.

marcosfrm, 2022-05-04:
https://github.com/pbatard/rufus/blob/4cdbad283a6ad00a2f200a1246bc164fa1e4cbb4/src/format.c#L1994-L1998

So `GetLogicalHandle()` returns `NULL` for unpartitioned drives. Should not the check in the `else if` branch be `(hLogicalVolume != NULL)`?

pbatard, 2022-05-04:
I'm afraid it's not that simple. Whatever state your drive was in before we started the operation is irrelevant, since it will have been partitioned by Rufus before trying to format it to `ext2`/`ext3`.

So, if you get into that call, it means something already went wrong, because your drive failed.

In other words, you are getting into fallback code, that is not supposed to run in normal operations, but which we allow to do regardless (instead of throwing an error when we could delete/create the partitions), just in case it happens to work.

Unsurprisingly however, it doesn't, because the drive was not partitioned successfully before we attempted to format it.

The one problem we need to care about is that:
``` 
Warning: Could not delete partition(s): Windows error code 0x000003F0 (FormatMessage error code 0x00003AFC)
```
error, because all subsequent issues derive from it.

Looking into Windows error codes, `0x3F0` is `ERROR_NO_TOKEN` (_An attempt was made to reference a token that does not exist._) and `0x3AFC` is `ERROR_MUI_FILE_NOT_FOUND`, which basically means that we couldn't find a translation for that message.

As to what `ERROR_NO_TOKEN` means in the context of deleting partitions, I'm afraid I have no idea.

For the record, I tried with a zeroed drive, and still saw no issue:
```
Rufus x86 v3.18.1877
Windows version: Windows 10 Pro for Workstations, 64-bit (Build 19044.1645)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB device 'SanDisk Extreme Pro USB Device' (0781:5588) [ID]
1 device found
No volume information for drive 0x88
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15567, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive8 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 119.2 GB)
Waiting for logical drive to reappear...
Notice: Using physical device to access partition data
Formatting (ext2)...
1953776 possible inodes out of 31260416 blocks (block size = 4096)
1563020 blocks (5.0%) reserved for the super user
Creating 954 inode sets: [1 marker = 11.9 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Done
Writing Master Boot Record...
Using Zeroed MBR
Found volume \\?\Volume{203abf7c-cb18-11ec-8d63-88d7f6e01c0c}\
Successfully remounted \\?\Volume{203abf7c-cb18-11ec-8d63-88d7f6e01c0c}\ as F:

Found USB device 'SanDisk Extreme Pro USB Device' (0781:5588) [ID]
1 device found
No volume information for drive 0x88
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15567, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: GNU/Linux (0x83)
  Detected File System: ext2
  Size: 119.2 GB (128042663936 bytes)
  Start Sector: 2048, Boot: No
Closed Windows format prompt
```

marcosfrm, 2022-05-05:
After a failed try, if I select GPT it works...

```
Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive does not have a Boot Marker
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 465.8 GB)
Waiting for logical drive to reappear...
Timeout while waiting for logical drive
Logical drive was not found - aborting

Found USB 2.0 device 'ST500LM0 12 HN-M500MBB USB Device' (174C:5106)
NOTE: This device is a USB 3.0 device operating at lower speed...
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x81
Disk type: FIXED, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60801, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00195579
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 465.8 GB (500106813440 bytes)
  Start Sector: 2048, Boot: No
```

```
Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 465.8 GB)
Waiting for logical drive to reappear...
Notice: Using physical device to access partition data
Formatting (ext3)...
7631024 possible inodes out of 122096385 blocks (block size = 4096)
6104819 blocks (5.0%) reserved for the super user
Creating 3727 inode sets: [1 marker = 46.6 set(s)]
++Closed Windows format prompt
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Creating 131072 journal blocks: [1 marker = 1638.4 block(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Done
Found volume \\?\Volume{7bf97cba-cf82-419b-becc-3a07689b9bf8}\
Successfully remounted \\?\Volume{7bf97cba-cf82-419b-becc-3a07689b9bf8}\ as E:

Found USB 2.0 device 'ST500LM0 12 HN-M500MBB USB Device' (174C:5106)
NOTE: This device is a USB 3.0 device operating at lower speed...
1 device found
No volume information for drive 0x81
Disk type: FIXED, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60801, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {8C0443FA-81A1-4E95-B500-1D834F6420FD}
Max parts: 128, Start Offset: 17408, Usable = 500107827712 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: ext3
  ID: {7BF97CBA-CF82-419B-BECC-3A07689B9BF8}
  Size: 465.8 GB (500106796544 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

Weirdly no `ERROR_NO_TOKEN` this time.

Feel free to close this bug, since it may be hardware related.

marcosfrm, 2022-05-06:
I recompiled 3.18 with this change:

```diff
diff --git a/src/drive.c b/src/drive.c
index 4e01f0f6..76fea7b2 100644
--- a/src/drive.c
+++ b/src/drive.c
@@ -2334,12 +2334,10 @@ BOOL CreatePartition(HANDLE hDrive, int partition_style, int file_system, BOOL m
                case FS_EXFAT:
                case FS_UDF:
                case FS_REFS:
-                       DriveLayoutEx.PartitionEntry[pn].Mbr.PartitionType = 0x07;
-                       break;
                case FS_EXT2:
                case FS_EXT3:
                case FS_EXT4:
-                       DriveLayoutEx.PartitionEntry[pn].Mbr.PartitionType = 0x83;
+                       DriveLayoutEx.PartitionEntry[pn].Mbr.PartitionType = 0x07;
                        break;
                case FS_FAT32:
                        DriveLayoutEx.PartitionEntry[pn].Mbr.PartitionType = 0x0c;      // FAT32 LBA
@@ -2349,7 +2347,7 @@ BOOL CreatePartition(HANDLE hDrive, int partition_style, int file_system, BOOL m
                        return FALSE;
                }
        } else {
-               DriveLayoutEx.PartitionEntry[pn].Gpt.PartitionType = write_as_esp ? PARTITION_GENERIC_ESP : PARTITION_MICROSOFT_DATA;
+               DriveLayoutEx.PartitionEntry[pn].Gpt.PartitionType = write_as_esp ? PARTITION_GENERIC_ESP : PARTITION_LINUX_DATA;
                IGNORE_RETVAL(CoCreateGuid(&DriveLayoutEx.PartitionEntry[pn].Gpt.PartitionId));
                wcsncpy(DriveLayoutEx.PartitionEntry[pn].Gpt.Name, main_part_name, ARRAYSIZE(DriveLayoutEx.PartitionEntry[pn].Gpt.Name));
        }
```

Now MBR partitioning works, but GPT does not. So it may be Windows craziness with non-native partition types **and** fixed disks.

MBR:

```
Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive does not have a Boot Marker
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 465.8 GB)
Waiting for logical drive to reappear...
Notice: Using physical device to access partition data
Formatting (ext3)...
7631024 possible inodes out of 122096390 blocks (block size = 4096)
6104819 blocks (5.0%) reserved for the super user
Creating 3727 inode sets: [1 marker = 46.6 set(s)]
++Closed Windows format prompt
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Creating 131072 journal blocks: [1 marker = 1638.4 block(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Done
Writing Master Boot Record...
Using Zeroed MBR
Found volume \\?\Volume{0043b4f2-0000-0000-0000-100000000000}\
Successfully remounted \\?\Volume{0043b4f2-0000-0000-0000-100000000000}\ as E:

Found USB 2.0 device 'ST500LM0 12 HN-M500MBB USB Device' (174C:5106)
NOTE: This device is a USB 3.0 device operating at lower speed...
1 device found
No volume information for drive 0x81
Disk type: FIXED, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60801, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x7404F3C9
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Detected File System: ext3
  Size: 465.8 GB (500106813440 bytes)
  Start Sector: 2048, Boot: No
```

GPT:

```
Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive does not have a Boot Marker
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 465.8 GB)
Waiting for logical drive to reappear...
Timeout while waiting for logical drive
Logical drive was not found - aborting

Found USB 2.0 device 'ST500LM0 12 HN-M500MBB USB Device' (174C:5106)
NOTE: This device is a USB 3.0 device operating at lower speed...
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x81
Disk type: FIXED, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60801, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {7D28E375-4C08-4212-857C-3F545DC24C85}
Max parts: 128, Start Offset: 17408, Usable = 500107827712 bytes
Partition 1:
  Type: Linux Data Partition
  Name: 'Main Data Partition'
  Detected File System: (Unrecognized)
  ID: {B658281D-C3BA-4D81-9510-650F5C011CF2}
  Size: 465.8 GB (500106796544 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

pbatard, 2022-05-06:
I can see part of this issue with fixed drives indeed, and now that you mention it, I think this is something I saw some time ago, and decided not to bother about, on account that most people would use a removable drive, and also, from past experience, because fighting against Windows nonsensical behaviour is always a major waste of time.

Even going with changing the partition type after we formatted the drive, or adding an extra partition before the ext one is not something I'm gonna bother with, because I fully expect adding a corner case behaviour is going to be creating more issues down the line, for a feature that is hidden behind advanced mode for a good reason (mostly because it is provided _AS IS_).

In short, I don't really see trying to work around this behaviour as something that's worth the investment for the time being. So, while I do appreciate the research, I'm just going to close this as _wontfix_.

closed 2022-05-06 by pbatard

marcosfrm, 2022-05-28:
Hi, https://github.com/pbatard/rufus/issues/1922#issuecomment-1117647377 fallback code does need fixing, no? (it probably does not matter much, since things already gone south at that point...)

marcosfrm, 2022-05-31:
Nice. Now it works with MBR. :)

There is a discrepancy: MBR correctly uses 0x83 for partition type, while GPT uses _Microsoft basic data_ (`EBD0A0A2-B9E5-4433-87C0-68B6B72699C7`). With the workaround in place, can it switch to _Linux filesystem_ (`0FC63DAF-8483-4772-8E79-3D69D8477DE4`)?

GPT using _Microsoft basic data_ for EXT2/3 also happens for persistence partitions.

pbatard, 2022-05-31:
Linux filesystem is a relatively new thing and "Microsoft" basic data is something that used to apply to any data partition, regardless of whether they were created from a Microsoft OS or contained a Windows file system.

See [note i. for the Wikipedia relevant entry](https://en.wikipedia.org/wiki/GUID_Partition_Table#cite_note-linwin-49):

> Previously, Linux used the same GUID for the data partitions as Windows (Basic data partition: `EBD0A0A2-B9E5-4433-87C0-68B6B72699C7`). 

I think I started using the MS Basic Data GUID before that new GUID became widespread and of course it makes the code simpler to use MS Basic Data for all data partition but you are making a good point, so I'll try to switch to the more recent GUID before the 3.19 release.

pbatard, 2022-06-08:
>  can it switch to Linux filesystem (`0FC63DAF-8483-4772-8E79-3D69D8477DE4`)?

Now done in [d8af83dd14f10126ad417731b3262f72a3ba051f].