pbatard/rufus #1924: write errors when trying to create ubuntu boot drive

SkeletalHamster, 2022-05-09:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
been trying to format a stick drive with 64 gigabytes with 50 gb of persistent partition, but i keep getting write errors while trying to do so. Ive done bad block checks and didnt get any problems, and ive been trying to fix this all weekend, but nothing seems to work.
Log
---
```
<Will reuse 'core.img' from './rufus_files/grub-2.06/' for Grub 2.x installation

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 7.6 GB)
● Partition 2 (offset: 8178880000, size: 50.0 GB)
Opened \\.\PhysicalDrive1 for shared write access
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Grub 2.0 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 7.6 GB)
● Creating Linux Persistence Partition (offset: 8178880000, size: 50.0 GB)
Waiting for logical drive to reappear...
Using Ubuntu-like method to enable persistence
Formatting (ext3)...
819199 possible inodes out of 13107197 blocks (block size = 4096)
655359 blocks (5.0%) reserved for the super user
Creating 400 inode sets: [1 marker = 5.0 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Creating 32768 journal blocks: [1 marker = 409.6 block(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Done
Formatting (Large FAT32)...
Opened \\?\Volume{8d5fc845-cfaa-11ec-a43c-7085c2c92272} for exclusive write access
Size : 7.6 GB 15972327 sectors
Cluster size 32768 bytes, 512 Bytes Per Sector
Volume ID is 19fb:3c66
32 Reserved Sectors, 1950 Sectors per FAT, 2 FATs
249506 Total clusters
249505 Free Clusters
Clearing out 3996 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 1982
Writing Partition Boot Record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take a while)...
Format completed.
Writing Master Boot Record...
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Grub 2.0 MBR
Writing Grub 2.0 SBR (from download) ✗
write_data: Please increase MAX_DATA_LEN in file.h
Re-mounted volume as E: after error

Found USB 2.0 device 'General USB Flash Disk USB Device' (090C:1000)
NOTE: This device is a USB 3.0 device operating at lower speed...
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7521, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x14B288FC
Drive has a Grub 2.0 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.6 GB (8177831424 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 50.0 GB (53687079936 bytes)
  Start Sector: 15974375, Boot: No
Found USB 2.0 device 'General USB Flash Disk USB Device' (090C:1000)
NOTE: This device is a USB 3.0 device operating at lower speed...
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7521, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x14B288FC
Drive has a Grub 2.0 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.6 GB (8177831424 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 50.0 GB (53687079936 bytes)
  Start Sector: 15974375, Boot: No>
```

pbatard, 2022-05-10:
<sigh>

> - [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

__No you didn't!__

Your log is truncated, and because of this I am missing __CRITICAL__ data that I need to be able to help you.

Please try again and, this time, copy your full log.

There's a reason why I ask for the full log, and I absolutely cannot help you if you withhold data as you did, when you __TRUNCATED__ the log instead of copying it all.

pbatard, 2022-05-11:
Closed due to OP having withheld crucial data that is required to troubleshoot the issue.

More specifically, if OP is using latest Ubuntu with latest version of Rufus, they should not get `Writing Grub 2.0 SBR (from download) ✗`, but of course, since OP truncated their log, it is missing the part about the version of Rufus they are using and the image they picked.

Let me be absolutely clear on this: It is simply not possible to help a person who does not provide the requested data, and does not follow up on a request to fill the missing data. As such I have no choice but to close this issue.

closed 2022-05-11 by pbatard