pbatard/rufus #1925: MysticLight  blocks access to usb.

thunderpants, 2022-05-10:
Win 10 
AMD Ryzen 7 3700X 8-Core Processor   3.60 GHz

Program stopping Rufus access when burning win10 iso to usb.

File from MysticLight (Standalone version}) ,c:\programfiles(x86) LEDKeeper.exe

LEDKeeper.exe flagged by Rufus has blocking access to usb.

Problem solved by quitting MysticLight2_Service in services and also killing process for LEDKeeper.exe
--------------------------------------------------------------------------------
Not sure why this rgb lights program is blocking a usb drive.
But it's easy to fix.

pbatard, 2022-05-10:
Thanks for the report.

It seems you're not the only person to report that [MSI Mystic Light is on the heavy handed side when it comes to interfering with legitimate applications](https://old.reddit.com/r/MSI_Gaming/comments/ffwgf6/msi_mystic_light_ledkeeperexe_triggering_easy/), so I would strongly suggest that you report this to MSI. There is really no reason why, outside of poor programming, MysticLight should interfere with disk access, so I'm going to go on a limb and assume that whoever developed MysticLight decided it would be a good idea to keep a handle open to all disk drives, including USB ones, which is very bad practice...

pbatard, 2022-05-14:
I'm going to close this issue, since there isn't much Rufus can do about third party applications that go well beyond their mandate...

closed 2022-05-14 by pbatard