pbatard/rufus #1926: Atypical 0x00000005

DS26RUS, 2022-05-10:
```
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Fedora-WS-Live-36-1-5'
  Size: 2.0 ГБ (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: Fedora-Workstation-Live-x86_64-36-1.5.iso (1.9 ГБ)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from 'D:\Downloads\Rufus\syslinux-6.04\' for Syslinux installation

Format operation started
Requesting disk access...
Could not lock access to \\.\PhysicalDrive2: [0x00000005] Access is denied.
NOTE: Could not identify the process(es) or service(s) accessing \Device\Harddisk2\DR3

Found USB 2.0 device 'Samsung Flash Drive USB Device' (090C:1000)
NOTE: This device is a USB 3.0 device operating at lower speed...
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 ГБ, Sector size: 512 bytes
Cylinders: 3900, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x7A218B9A
Drive has an unknown Master Boot Record
Partition 1:
  Type: EFI FAT (0xef)
  Detected File System: FAT16
  Size: 9.9 MB (10385408 bytes)
  Start Sector: 172, Boot: No
Partition 2:
  Type: FAT16 (0x06)
  Detected File System: (Unrecognized)
  Size: 29.9 GB (32068599808 bytes)
  Start Sector: 20480, Boot: No
```

**Rufus refuses to write any image to any disc. But the rest of the time, similar programs (balenaEtcher, UNetbootin, Universal USB Installer) work without errors. Windows antivirus disabled**. Win 11 x64 + Lenovo Legion 7 2021 (AMD)

pbatard, 2022-05-10:
https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied

The check list is __NOT__ optional.
The log should __NOT__ be truncated.

closed 2022-05-10 by pbatard

DS26RUS, 2022-05-10:
> https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied
> 
> The check list is **NOT** optional. The log should **NOT** be truncated.

1.  I can’t turn off all programs so that Rufus will work exactly? It's impossible. There are no antiviruses
2. 
```
Rufus x86 v3.18.1877 (Portable)
Windows version: Windows 11 Pro, 64-bit (Build 22616.1)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0419 (ru-RU)
Will use default UI locale 0x0419
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ru-RU'
NOTE: The Russian (Русский) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 2.0 device 'Samsung Flash Drive USB Device' (090C:1000)
NOTE: This device is a USB 3.0 device operating at lower speed...
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 ГБ, Sector size: 512 bytes
Cylinders: 3900, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x7A218B9A
Drive has an unknown Master Boot Record
Partition 1:
  Type: EFI FAT (0xef)
  Detected File System: FAT16
  Size: 9.9 MB (10385408 bytes)
  Start Sector: 172, Boot: No
Partition 2:
  Type: FAT16 (0x06)
  Detected File System: (Unrecognized)
  Size: 29.9 GB (32068599808 bytes)
  Start Sector: 20480, Boot: No

Format operation started
Requesting disk access...
Could not lock access to \\.\PhysicalDrive2: [0x00000005] Access is denied.
NOTE: Could not identify the process(es) or service(s) accessing \Device\Harddisk2\DR3

Found USB 2.0 device 'Samsung Flash Drive USB Device' (090C:1000)
NOTE: This device is a USB 3.0 device operating at lower speed...
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 ГБ, Sector size: 512 bytes
Cylinders: 3900, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x7A218B9A
Drive has an unknown Master Boot Record
Partition 1:
  Type: EFI FAT (0xef)
  Detected File System: FAT16
  Size: 9.9 MB (10385408 bytes)
  Start Sector: 172, Boot: No
Partition 2:
  Type: FAT16 (0x06)
  Detected File System: (Unrecognized)
  Size: 29.9 GB (32068599808 bytes)
  Start Sector: 20480, Boot: No

```

pbatard, 2022-05-10:
> `Build 22616.1`

Insider builds of Windows 11 are __not__ supported. If you use a Windows Insider build, you are 100% on your own.
Microsoft is testing "security" features there that interfere with legitimate applications, and we've already seen issues because of that.

> I can’t turn off all programs so that Rufus will work exactly? It's impossible. There are no antiviruses

[The list provided at the end of the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Software) is not exhaustive, which means it is up to you to identify what application (or Windows insider feature) is interfering. For instance, #1925 shows how someone managed to identify software that they were running, that interferes with Rufus.

As you can imagine, I cannot "guess" what software you might be running that may prevent Rufus from working, so it is your job to figure that out.

DS26RUS, 2022-05-10:
> Insider builds of Windows 11 are **not** supported. If you use a Windows Insider build, you are 100% on your own. Microsoft is testing "security" features there that interfere with legitimate [...]

Insider builds are not the source of the problem, this is verified by a rollback. Before I write here, I tried to guess for half a day, what interferes with your beloved Rufus but does not interfere with others.

pbatard, 2022-05-10:
> Insider builds are not the source of the problem, this is verified by a rollback.

Your log is not from a rollback, and, again, I have seen Insider builds introduce "security" features that prevented legitimate applications like Rufus from running properly, with said features being removed from the actual releases. Oh, and I've wasted a lot more than half a day trying to figure out what these Insider builds were doing that could interfere, only to find out that all that time was utterly wasted when actual release did not have the issues.

And that is the precise reason why, again, if you do choose to run Rufus on an Insider build, you are 100% on your own, and I will __NOT__ take anything for granted there.

> what interferes with your beloved Rufus but does not interfere with others.

Unlike other software, Rufus tries to acquire an exclusive lock to the target disk, so that, while Rufus is writing data, there is no risk that some other software that kept a handle to the disk could also write data and thereby create an issue that will leave users with a media that does not boot properly.

Also, considering that on a vanilla installation of Windows 11 release, I am not seeing your error occur, I can only conclude that your issue is due to either:
1. Your running an Insider build of Windows
2. Your having installed additional software or enabled features that are not present by default on vanilla Windows, and therefore, that you are best placed to figure out, as you are the __only__ person that can tell us how your environment differs from vanilla Windows.

Thankfully, not all software that creates bootable media is designed to work in the same manner, so it really doesn't matter that X or Y works whereas Rufus doesn't, __you__ still need to identify how exactly your environment different from vanilla Windows 11 to prevent Rufus from working.

DS26RUS, 2022-05-10:
> Your log is not from a rollback, and, again, I have seen Insider builds introduce "security" features that prevented legitimate applications like Rufus from running properly, with said features [...]

Thanks anyway for the replies)

pbatard, 2022-05-10:
If you can identify what's specific to your environment that is causing the issue, I'll still be interested to know.

DS26RUS, 2022-05-26:
> If you can identify what's specific to your environment that is causing the issue, I'll still be interested to know.

Hello, I found the problem. I did not see a logical dependency, but this [program](https://www.legionfancontrol.com/) interferes.

pbatard, 2022-05-26:
Thanks for following up. I have added _Legion Fan Control_ to the [list of incompatible software](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Software).

DS26RUS, 2022-05-26:
To write a flash drive through Rufus, just close the LegionFanControl
process for a while