pbatard/rufus #1927: CD-DVD-drive not present in the device list

PhilDuValais, 2022-05-13:
﻿

Checklist
---------
- [ x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
HP ProDesk 400 G6 - Windows 11 with HP USB CD-DVD-R/RW-drive

The D:Drive cannot be selected from Rufus.
The CD-DVD drive is present and working.

Log
---
```
Rufus x86 v3.18.1877
Windows version: Windows 11 Pro, 64-bit (Build 22000.675)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x040C (fr-FR)
Will use default UI locale 0x040C
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'fr-FR'
0 devices found
```
[image]
[image]

pbatard, 2022-05-13:
I think you are misunderstanding what Rufus is about.

It is __not__ an application that is designed to select data from an optical disc. Instead it expects that you have an ISO image (in other words, a single file with a `.iso` extension) that is a 1:1 copy of an optical disc image, and that resides on your internal hard drive or SSD, rather than on a disc drive.

Typically, an ISO is what you use to create ("burn") the optical disc media that you can then use with a disc drive.
So that is what you need to start with, as, if you only have an optical disc in a disc drive, you won't be able to use that as a source in Rufus, it __must__ be an ISO.

So you will first need to either:
- Download the `.iso` file that you used to create the optical disc that you currently have in your optical disc drive. Note that, if this is a Windows ISO, then Rufus can help you download such a file: https://github.com/pbatard/rufus/wiki/FAQ#Help_I_dont_see_the_ISO_download_button
- Create a `.iso` file from the optical disc you have using either a disc &rarr; ISO application (or Rufus through the [<kbd>Alt</kbd>-<kbd>O</kbd> shortcut](https://github.com/pbatard/rufus/wiki/FAQ#Power_keysCheat_modes) though this feature is provided __as is__, with no guarantee that it will work).

Then, once you have the `.iso` file on your hard drive, you can use the `SELECT` button in Rufus to select it.

Finally, judging from your screenshot, it looks to me like you may already have the ISO you need, through that `Win10_21H2_Fr...` file on your desktop. That's probably what you used to burn your optical disc, whereas that step was not needed and, instead, this is the file you should select in Rufus.

As I believe I have answered your question, I will close this issue.

closed 2022-05-13 by pbatard

PhilDuValais, 2022-05-13:
> Create a .iso file from the optical disc you have using either a disc → ISO application (or Rufus through the Alt-O shortcut <https://github.com/pbatard/rufus/wiki/FAQ#Power_keysCheat_modes> though this feature is provided as is, with no guarantee that it will work).

Hello,

Thanks and regards

Phil