pbatard/rufus #1928: Windows 11 OOBE bypass request

hooddy, 2022-05-14:
Is it possible to add to Rufus in addition to currently existed options in Extended Windows 11 installations (no TPM, no Secure Boot) and no OOBE trick?

pbatard, 2022-05-14:
I'm not aware of a no OOBE trick.

There's using a `unattend.xml` file to fill in the answers to OOBE and there's disabling network if you want to force the creation of a local account.

I am not planning to add any of those to Rufus, because my understanding is that it wouldn't be useful to that many people, and Rufus is not designed as a sysadmin/poweruser utility in the first place. Instead it is designed for conventional regular installation, where users are expected to fill the OOBE.

closed 2022-05-14 by pbatard

hooddy, 2022-05-14:
I mean trick yo install with no MS account. Not to diasble whole OOBE. 
Like this one https://www.windowscentral.com/how-set-windows-11-without-microsoft-account
I suppose its okay to have it in option Extended installation. But for sure I can did any trick with no Rufus at all.

pbatard, 2022-05-14:
That's just disabling the network (Solution 2) that I was mentioning above. Also note that TPM/Secure Boot bypass is relatively easy to achieve (though __not__ with the Windows Store version, due to Microsoft restrictions, where the option has to be disabled altogether) because we just need to mount an offline registry hive. On the other hand, even if I saw some merit to adding the feature (which, again, I don't), issuing an `oobe /bypassnro` on an offline image looks like a major pain in the ass.

So, once again, I am not planning to add the feature you want. If you want a local account, just make sure your network is disconnected, like you did for Windows 10.

hooddy, 2022-05-14:
@pbatard but how be with action like this one? Windows will required online in future release with no option buy oobe trick
https://www.windowscentral.com/windows-11-pro-will-soon-require-internet-connection-when-setting-first-time

pbatard, 2022-05-14:
We'll deal with it when/if we get there... which won't be before October this year at the very least (22H2... or later).

Experience has taught me that trying to pre-emptively deal with what Microsoft say they are _planning_ to do but haven't actually done is frequently a complete waste of time.

hooddy, 2022-05-14:
can't agree but i take your opinion. well its a pity.

pbatard, 2022-05-14:
Please bear in mind that you are asking somebody else to invest their time, entirely for free, in order to preempt something that is far from having been finalized, and for a feature that will most likely not benefit as many people as you believe it will.

See also [this relevant FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want). I guess I'm going to have to repeat one of my usual sayings:

> Anything that you don't have to accomplish yourself always looks like a 5 minutes job.

Finally, I think that trying to group this matter with the TPM and SB bypasses is not the proper thing to do, because, the TPM and SB requirements are pure hardware limitations that actually prevent you to proceed with Windows installation, whereas the same cannot be said about the account creation, where someone isn't actually going to end up stuck unless they can figure out a workaround. Which is why do see this matter as a very low priority item.

hooddy, 2022-06-25:
@pbatard wow it's here at least!
> Add a new selection dialog for Windows 11 setup customization
but I can't find it in current beta
[image]

pbatard, 2022-06-25:
Well, _officially_, [Microsoft has released Windows 11 22H2](https://blogs.windows.com/windows-insider/2022/06/07/releasing-windows-11-version-22h2-to-the-release-preview-channel/) (note how [as opposed to the other blog posts](https://blogs.windows.com/windows-insider/posts/) this is not being announced as a release of Windows 11 Insider but as a release of Windows 11 22H2), so, if Microsoft are more or less declaring that it is stable, __of course__ we are going to take a look at it, in the same manner as we'd have done it if, per their previous stance, Microsoft had simply sat on it until October.

Per Microsoft official designation, we are not looking at an Insider version here, which I still don't want to bother with, but at a bona fide release.