pbatard/rufus #1931: new feature

eddilara, 2022-05-18:
make it so rufus can easily write a macos installesd dmg file to a usb and be bootable on a mac

eddilara, 2022-05-19:
this can be done without having to have access to hfs file system
it just has to convert the dmg to img....then write it

pbatard, 2022-05-20:
Unfortunately, if [this](https://github.com/Lekensteyn/dmg2img/blob/develop/dmg2img.c) is what you are thinking about, it is GPLv2 __only__ so I can't use it in Rufus (has to be GPLv2 __or later__), and, at any rate, porting it to Windows, along with sorting out its dependencies, looks like a lot of work anyway.

In short the cost/benefit ratio is simply too high at this stage, and, considering how Apple makes obtaining `.dmg` files rather difficult, I'm not really seeing much incentive to invest resources into this.