pbatard/rufus #1932: Rufus stops before Setupin Win 7 and FreeDos

dazecutr, 2022-05-18:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus stops after PXE drivers are loaded and before Setup starts. I a couple rectangles and "H"s in a dot matrix format. Pressing any key will reboot to the same thing. I tried on a sd card, I tried using FreeDos  but got the same thing and no prompt. There are no error messages. Im on an Acer Laptop with 20GB ram and 4 terabyte in external drives, Nvidia  Geforce MX150

Log
---
```
<Rufus x86 v3.18.1877
Windows version: Windows 10 Home, 64-bit (Build 19043.1706)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'General UDisk USB Device' (048D:1234)
Found USB 3.0 device 'WD My Book 25ED USB Device' (1058:25ED)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found UAS (USB 2.0) device 'Seagate Expansion SCSI Disk Device' (0BC2:231A)
NOTE: This device is a USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7648, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {D9605649-9673-427E-A418-16C1A5520BB6}
Max parts: 128, Start Offset: 17408, Usable = 62914525696 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {E3E65820-F8A5-4F86-9003-EB41D749C613}
  Size: 58.6 GB (62912424960 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {6008FA87-2473-4E0A-8815-13D0FE9B6B83}
  Size: 1 MB (1048576 bytes)
  Start Sector: 122877878, Attributes: 0x9000000000000000
>
```

pbatard, 2022-05-18:
> Rufus stops after PXE drivers are loaded and before Setup starts. 

Then it's an issue with your computer's boot process, not Rufus.

If you made absolutely sure that your USB drive has no bad blocks, _as you indicate you did in the check list_, then, if your drive still doesn't boot, the problem is with your BIOS/UEFI or its settings, and there's nothing Rufus can do to somehow toggle your computer to boot legacy media.

I suggest you post in Acer user forums and ask how one is supposed to enable legacy boot for your computer model, because there is no reason that, if your USB isn't defective, your Acer computer should not be able to boot a legacy FreeDOS media.

pbatard, 2022-05-20:
I'm going to close this issue, since it is a BIOS/UEFI settings problem rather than a Rufus problem.

closed 2022-05-20 by pbatard