pbatard/rufus #1937: There is a serious problem about Win 11 Business Edition from Select menu of MediaCreationTool.bat

GoldenLighter, 2022-05-23:
I downloaded windows 11 ISO image with this bat file.([https://github.com/AveYo/MediaCreationTool.bat](https://github.com/AveYo/MediaCreationTool.bat))
Business Edition - Enterprise
But it seems there is a problem about this iso file.
First of all the command below has error on these files to get windows os versions :
**dism /Get-WimInfo /WimFile: M :\ sources\install.esd**
As you see install extension is esd not wim
Secondly rufus does not show any drop down list related to remove TPM & Secure Boot during create a bootable usb media with this iso.
Means there is a problem about this iso file & rufus can not read it.
After create a bootable usb with MBR format for old pc & reboot from usb it does not work and sticks on press any key to boot from usb page.
What is wrong about "Media Creation Tool" or "rufus"?
Where can i download pure and latest VL windows 11 enterprise iso with no issue?
I want to install that os on an old pc > No UEFI -> No GPT -> So rufus can help me on that.

I downloaded win 10 iso from Select menu of that bat file (Enterprise) and it was ok with rufus.
I installed it on my machine.

People say you can not install win 11 without GPT & UEFI and this is not true.
I downloaded win 11 from another place (wim extention) and create a bootable media with MBR format using with rufus.
Every thing was ok and i had n't any issue during boot with it.

So tell me what is wrong about win 11 iso output of that bat file & why rufus can not recognize it to bypass TPM and SecureBoot?
Also why created bootable media of that iso file with rufus (MBR format) does not work!

pbatard, 2022-05-23:
> So tell me what is wrong about win 11 iso output of that bat file & why rufus can not recognize it to bypass TPM and SecureBoot?

So post a log from Rufus when you try to open that ISO. I genuinely cannot help you if you won't post a log.

Please understand that it's not because the application is free that my time is, and I'm afraid I have better things to do than try to replicate the steps you ran. So, unless you provide a FULL log from Rufus, I'm simply going to close this issue, as you are not using an official Microsoft ISO.

GoldenLighter, 2022-05-23:
That bat file is downloading all iso files from microsoft web site.
So it is legit.
This is why i am using it.
Just tell me where are rufus log files?
I tried rufus with both setup & portable versions.
Both of them have same action without install any files on Program Files folder.
It seems both versions are portable.
So where is rufus log file?

JonnyTech, 2022-05-23:
>  Just tell me where are rufus log files?

From the issue checklist:

---

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

GoldenLighter, 2022-05-23:
```
Rufus x86 v3.18.1877 (Portable)
Windows version: Windows 10 Enterprise, 64-bit (Build 19044.1706)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'UFD 2.0 Silicon-Power8G USB Device' (13FE:3E00)
1 device found
Skipping GUID volume for '\Device\Floppy0'
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 973, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0004FAC7
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 7.5 GB (8010072064 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: '11_21H2_VOL_x64_en-us'
  Size: 4.1 GB (Projected)
  Note: File on disk is larger than reported ISO size by 550 KB...
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.0.0)
Using image: Windows.iso (4.1 GB)

```
Dear owner of rufus,
I Put the log in the upper.
Please check and tell what is wrong about Business Edition of Win 11 from that bat file.

pbatard, 2022-05-23:
> That bat file is downloading all iso files from microsoft web site.

That bat file is generating an ISO by aggregating files in a manner that is not endorsed by Microsoft. It is __not__ downloading an ISO from Microsoft. As such, there are plenty of things that can produce an improper ISO.

> So it is legit.

Only ISOs or applications made by Microsoft to create an ISO are legit. This bat file is not produced by Microsoft, so, whereas what it does is _legal_, it is not a legit (i.e. _genuine_) Microsoft application, and therefore the ISOs it produces, while _legal_, are not __genuine__ Microsoft Windows ISOs.

> Just tell me where are rufus log files?

@JonnyTech, told you. Next time, please don't ignore the check list. Ignoring the check list is the best way __NOT__ to have your question answered.

 > I will send a donation because of this software.

I'm afraid [I don't accept donations](https://rufus.ie/en/#donate), but thanks anyway.

By the way, I produced a Windows 11 Enterprise ISO using your `.bat` file and saw no issue. The option to disable TPM and Secure Boot requirement was there:
```
Rufus x86 v3.18.1877
Windows version: Windows 10 Pro for Workstations, 64-bit (Build 19044.1706)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB device 'SanDisk Extreme Pro USB Device' (0781:5588) [ID]
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15567, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00C9EC23
Drive has a Grub 2.0 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 119.2 GB (128042663936 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: '11_21H2_VOL_x64_en-gb'
  Detected: Windows 11 ISO (Build 22000.194)
  Size: 4.1 GB (Projected)
  Note: File on disk is larger than reported ISO size by 530 KB...
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: 11 21H2 Enterprise x64 en-GB.iso (4.1 GB)
```

Now, considering that your log indicates that Rufus was not able to detect the `Install.esd` version (`Uses: Install.esd (version 0.0.0)`) I'm going to assume that your ISO has a problem, and since it was not produced by a Microsoft utility or downloaded from Microsoft, I'm just not going to explore this any further.

I would therefore __strongly__ encourage you to obtain Windows Enterprise through __legal__ means, because, if you do have a _legitimate_ Windows Enterprise volume license, then you should be able to download Enterprise retail ISOs from your [Volume Licensing Service Center(https://www.microsoft.com/licensing/servicecenter/default.aspx) account (which is how "legit" customers are supposed to install Windows). If you have a legal Volume License, then you should not have to use a non-Microsoft-endorsed bat file to create an ISO.

I will therefore close this issues.

closed 2022-05-23 by pbatard

GoldenLighter, 2022-05-23:
Thanks for your attention by the way.
Good Luck my friend.

GoldenLighter, 2022-05-23:
Check this please :
[https://github.com/AveYo/MediaCreationTool.bat/issues/145](https://github.com/AveYo/MediaCreationTool.bat/issues/145)

pbatard, 2022-05-23:
Please bear in mind that I did create an ISO with the script __and had no issue__.

So, as far as I can tell, the problem is neither with Rufus or the script, but with your environment.
This means that, before you try to rope project maintainers into an issue that, so far, you seem to be the only one experiencing, you may want to check very thoroughly that the issue isn't with your environment.

GoldenLighter, 2022-05-23:
I downloaded that iso file with bat script in a vps with server 2008 os from netherlands.
Twice
So what is wrong about my environment?
I also downloaded iso from another web site and it worked very well.
See the link that i shared with you.(Owner of MediaCreationTool)

pbatard, 2022-05-23:
> So what is wrong about my environment?

That's your problem, not mine.

Again, I went as far as I'm going to go to look into your issue, and found that, when running the script in my environment, there was no issue. I can only reiterate that, instead of going through VPS, third party scripts and whatnot, your solution lies into purchasing a VL license directly from Microsoft so that you get a direct download to an ISO.

Or, to make it abundantly clear, you should understand that busy people are unlikely to be interested into spending their valuable time, __for free__, to help figure out why your _"I'm going to sidestep Microsoft's official channels to install Windows 11 Enterprise"_ solution doesn't work in your environment.

AveYo, 2022-05-23:
@GoldenLighter
You're using old Server OS (on it's last ESU legs, win7 based), not consumer one, any scripts or applications are bound to behave differently.
To make your environment "work" is a futile exercise. Rather just fire up a VM with a proper consumer OS, and everything will work just fine. Tho I believe you only had to use the Default MCT script choice and it would have worked out..

The sane choice is what Pete Batard recommended: download official iso.
You don't even need business media specifically, modern Windows has virtual editions so you can get consumer media and then just change product key during / after setup.

Now, stop bothering Pete Batard (and me) with this YOU problem..

Not that it is my fault, but sorry nonetheless @pbatard.

GoldenLighter, 2022-05-23:
Just tell it does n't work.
Just tell i downloaded that iso and you are right.
There is a problem with it.
Peoples issue will help you to fix any bugs for better performance.
Buying a license from microsoft web site is my business, Not you.
So why do you bother yourself about it?

GoldenLighter, 2022-05-23:
Accept my apology finally.
Problem was for windows server 2008 during download of iso file.
Just do n't bother yourself about license and this part has nothing to do with this issue.
Thanks for doing this free.
Just let us donate you to tell you we are appreciate on what you are doing.

GoldenLighter, 2022-05-25:
Problem solved.
Rufus is the best Reliable USB Formatting Utility that i ever seen.
Problem was from windows 2008 as host to download iso file using with bat file.
I downloaded it again with win 10 and now it is ok.
Sorry guys.
rufus has no problem at all
bat file has no problem at all
Mirosoft should n't act like that and make a connection between host dl & iso file.
iso file should be independent.
By the way thanks a ton to owner of rufus and owner of bat file.