pbatard/rufus #1943: Apply Anaconda workaround to RHEL 9

marcosfrm, 2022-06-02:
Anaconda is still buggy in RHEL 9, requiring the `inst.stage2 ➔ inst.repo` workaround from https://github.com/pbatard/rufus/issues/1777.

Tested with `AlmaLinux-9.0-x86_64-minimal.iso` (`ISO label: 'AlmaLinux-9-0-x86_64-dvd'`) and it does solve the issue.

Replacing `8` with `[8-9]` in `redhat8_derivative` array will suffice. 🙏

https://github.com/pbatard/rufus/blob/746f91acc769b2b5b4ad44a31a1aa70e5619ae57/src/rufus.c#L1265-L1273

Let's hope Red Hat fixes this nasty bug for RHEL 10 (proposed fix: https://github.com/rhinstaller/anaconda/pull/3786)!

marcosfrm, 2022-06-02:
Btw, @m10k , to maintain consistency, can we drop the last dash from Miracle Linux entry (`^MIRACLE-LINUX-8.*`)?

pbatard, 2022-06-07:
Applied, thanks!