pbatard/rufus #1945: unable to locate ISO file

DPS304, 2022-06-05:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
Rufus x86 v3.18.1877 (Portable)
Windows version: Windows 10 Home, 64-bit (Build 19044.1706)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:007A)
Using 'autorun.inf' label for drive E: 'PNY-16GB'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1988, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x070AAABA
Drive has a Rufus Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 15.2 GB (16357720064 bytes)
  Start Sector: 2048, Boot: Yes
Checking for Rufus updates...
Checking release channel...
No new release version found.

<FULL LOG>
```

DPS304, 2022-06-05:
My laptop has USB 3.0 connections, so I am using an adapter to enable use with a USB 2.0 drive.  The ISO was downloaded using M$ download tool, and placed in both the 'downloads' and 'documents' directories (neither worked); Rufus would not enable a download. Rufus can't locate the ISO, so I am unable to start it

pbatard, 2022-06-05:
Please bear in mind that Rufus runs elevated, therefore as the `Administrator` user, and **not** as your regular user.
I suspect that your _Downloads_ or _Documents_ directories are specific to your current user and will therefore be different for elevated app.

Please copy the ISO to a regular directory (e.g. `C:\ISOs\`) that is common to all users, and you should be able to see them.
If you still can't, then it's a permission problem with elevated users not being able to access other users data, which you will have to fix on your own, as there isn't much Rufus canm do then.

DPS304, 2022-06-05:
No luck with creating the directory name and moving the ISO to that directory (see attached screen capture).  I will try your other suggestion
[image]
.

closed 2022-06-14 by pbatard

DPS304, 2022-06-14:
> I'm going to close this issue since there has been no update, and Rufus
> does not apply any filter to the ISO selection, so any problem with
> locating ISO files has to boil down to permission or account issues that
> Rufus can not do anything about.

Go ahead and close the request. I opened the select/download button. If i
only press the button, I see the ISO as well