pbatard/rufus #1946: Extended Windows 11 Installation (no TPM/no Secure Boot) not there for my isos.

artivision, 2022-06-06:


pbatard, 2022-06-06:
The checklist __does__ apply to you. If you choose to ignore the check list, then be prepared not to have a relevant answer.

As such, I will provide you with only __one__ of many possibilities why you may not have the extended option:
- You are running the Windows Store version of Rufus. You need to use the non-Store version from the https://rufus.ie if you want these options to work.

Of course, since you didn't bother to provide your log despite being asked to, __I have to take a complete wild guess, which may be wrong and not solve your problem at all__. But that's entirely on you.

closed 2022-06-06 by pbatard

artivision, 2022-06-06:
I have the right edition. Doesn't make logs that i can see. Wild guess from my side: Doesn't understand Esd format and cannot delete Dlls inside the Iso.

pbatard, 2022-06-06:
> Doesn't make logs that i can see. 

This was part of the _checklist_ that you blatantly __chose__ to ignore. It tells you exactly how to produce a log. Do that.

- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.