pbatard/rufus #1947: Please re-support/keep FAT32 supports for large drive?

trippleflux, 2022-06-06:
I have 64 GB USB drive, and when trying either rufus 3.18 portable or latest snapshot build, I cannot format it with MBR & FAT32 4096 clusters size, the only option available are NTFS,ex-fat, etc.

Those options above is required for my motherboard bios flashback through usb drive. Please re-add such options for larger drive.

When using most of partition manager software, I managed to format it as such above options. But I am more choosing rufus as it's low size app and low profile great app!.

marcosfrm, 2022-06-06:
_Large FAT32_ **is** FAT32, no? Any decent implementation must read it...

pbatard, 2022-06-06:
> the only option available are NTFS,ex-fat, Large FAT32.

Large FAT32 is FAT32.

> When using most of partition manager software, I managed to format it as such above options.

Not with Windows native format tool you didn't.

And therein lies your issue. FAT32 with large drives (i.e. drive that are larger than 32 GB) requires special handling, hence "Large FAT32". But there's nothing that fancy about Large FAT32, which any device that has a __proper__ FAT32 implementation should be able to support, regardless of the cluster size being used. Therefore, if your device cannot support Large FAT32 with the cluster sizes that Rufus allows, then that's a problem with your device software, not Rufus, and I am not going to go out of my way to add cluster sizes that don't work with large flash drives to solve a specific issue that has to do with what I will assert is a __very limited__ requirement to format a drive that can be used for BIOS flashback on a __very limited__ set of motherboards that have a crippled support for FAT32, i.e. a scenario that very few other people are likely to encounter.

You fall squarely on point 2 [of this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want), and the cost/benefit ratio of exploring alternate cluster sizes from the current ones Rufus allows for Large FAT32 is simply way too high.

As such I will close this request

closed 2022-06-06 by pbatard