pbatard/rufus #1949: Stuck at NTFS Fixup 100%

xeinebiu, 2022-06-13:
[image]

Antivirus (Windows Defender) is disabled.

pbatard, 2022-06-13:
Log is accessible even when Rufus seems stuck.

Please provide your log as I genuinely cannot help you without it.

Also, this could be an indication that your drive is actually defective, so I would __strongly__ advise you to run a bad blocks check, like the checklist that you ignored advised you to do.

dion-pro, 2022-06-17:
This happened in my Kingston 32 GB USB 3.2 Disk. When I eject it, it responds.

pbatard, 2022-06-17:
> Please provide your log as I genuinely cannot help you without it.

xeinebiu, 2022-06-18:
In my case, my USB does not respond anymore. No media type found an error.

I do not think Rufus broke it though as I have been using Rufus since 2015-16. I used the device on a Xiaomi MiBox S which formatted my drive, then I wanted again to make it bootable with Windows and I was facing an NTFS Stuck error.

I tried the AID with Mackbook and it took quite some time and said the USB is healthy, but RUFUS still was failing. After 4-5 of RUFUS Failing, the USB stopped completely responding.

So, I can't help guys more than this.

pbatard, 2022-06-18:
@xeinebiu,

Thanks for the report. So the reason for the "Stuck at NTFS fixup" is that your USB was failing, which makes sense, but is obviously not a problem that Rufus can fix.

Because this is a frequent problem (Flash drives do fail a lot more frequently than people expect), I'm going to assume that the same thing is true for @dion-pro, so I'm going to close this call, especially as one has to realize that the "NTFS Fixup" step is one where we simply call Windows checkdisk and wait for it to complete (Rufus does absolutely nothing but wait on a Windows native utility call to complete). So you would encounter the exact same issue if you called Windows' checkdisk manually and, obviously, because I am not Microsoft, I can't alter the behaviour of checkdisk if it appears stuck or uses 100% CPU.

closed 2022-06-18 by pbatard

xeinebiu, 2022-06-18:
@pbatard feel free to close the issue as per my side it is how I described above.

At the time I opened this issue the USB still was working for copying files but since now it does not work at all, I assume it was a USB Failure.

JonnyTech, 2022-06-18:
> After 4-5 of RUFUS Failing, the USB stopped completely responding.

Cheaper USB drives, especially plastic body ones, do get rather warm and are unable to dissipate heat well. Programs like Rufus have a very long and consistent write cycle which causes the drives to get very warm indeed. If you are doing 4-5 cycles of Rufus operations without giving the drive a suitable period to cool down then it will generally result in a drive failure (the most common situation being the firmware locking it permanently in read-only mode). I tend to alternate drives or have a small usb-powered fan pointing at the unit to avoid such failures.

These observances are my own, with multiple tests performed on many, many cheap plastic USB drives; the types given away free at conferences and exhibitions. Sometimes just two runs is enough to destroy a drive. And as each version of Windows gets bigger, then the Rufus write cycle gets longer, therefore the drives get warmer. Hence, I believe, that explains why we are getting more and more reports of drive failures.

Also, I have personally observed people running Rufus twice in quick succession in order to "make sure that the data is written correctly". I strongly advise against this procedure as it can kill the drive during the second write period.

xeinebiu, 2022-06-18:
@JonnyTech 

If this is true, then I expect the USB Drive got broken after the MacBook Pro did a full AID Scan and the device got very hot. As the process took almost one hour and the USB was all hot during that time, it makes sense why it was completely broken.

[image]

xeinebiu, 2022-06-18:
I still wonder, why it was failing only on "fixing NTFS". Is there any specific part of the USB which could be failing for this process?