pbatard/rufus #1951: all the explanations by 'experts' are only step-wise operations with no insights or background for interpreting what you thought you were intending to convey to users

kimballjohnson, 2022-06-17:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
I always do a thorough web search to find examples and explanations of how to use the tools that appear to perform the tasks I want to accomplish. Normally, I find between 5 and 8 explanations regarding how to successfully achieve what i need to accomplish. Then I save that explanation to a Pdf and store it in a ring binder.
But... in the case of Rufus, I am unable to find any two sentences in any of the 5-8 examples I found this time around that agree in the meaning and utility of the features visible in the Rufus UI.
Ok, 'Device' has got to mean the USB...
Then we get into 'Boot Selection'... so which of the many meanings of Boot and Selection is this really about?
My expectation is that this item is a means to identify and remember what the OS Installation CD is called or where it is found on the system. But what is the circle with the checkmark inside it? I really can't make anything out of this icon and cannot imagine what feature of the UI it is referring to, represents, or where it the process does it become part of the context.
But immediately, I'm confounded again by the choice between 'select' and 'download'. What could this be about other than the ISO or the OS Installation CD? So what does 'select' mean in the Rufus context? And under what circumstances would I 'select' it, or 'download it'? Since the drop down doesn't allow a selection of one option or the other... Its just another suspicion.
And then there's 'Target System': now that's got to be about the boot process... except that the selection between 'Partition Scheme' and 'UEFI or MBR master boot record'? doesn't really represent a 'system' but might mean a 'boot technology'.
So all the explanations by 'experts' are only step-wise operations with no insights or background for interpreting what you thought you were intending to convey to users about how to keep all these details straight in their minds (I stopped doing that years ago.)
So, could you please straighten out the confusion by identifying the purpose and capability and result of these features in a UI that is to some extent more lucid?
Thanks,
Kimball Johnson
Kimball.Johnson@engagile.io

Log
---
```
<FULL LOG>
```

JonnyTech, 2022-06-17:
Please study the comprehensive [FAQ](https://github.com/pbatard/rufus/wiki/FAQ)

pbatard, 2022-06-17:
Rufus expects you to be familiar with both standard terminology (_device_ is standard computing terminology, and so is _Partition scheme_, _UEFI_, _Boot record_), standard Windows UI elements (a split button, such as the _Select/Download_ one, or a drop down are standard Windows UI elements, and if you see a _Boot selection_ dropdown, it should be rather explicit that the elements for that dropdown are _selections_ of _Boot_ methods you can enact), as well as the elements that can play a part in the boot process (such as a source ISO image, that one may already have available or need to download).

If that is not agreeable to you, then I'm afraid that you are looking at the wrong utility, and I will kindly ask you to orient yourself towards a utility that provides hand holding when it comes to creating boot media, which Rufus isn't, because, again, it considers that creating and using boot media is far from an entry level operation and, even if it might come as a shock to you, intentionally attempts to filter out users who are expected to have trouble with the whole process and are therefore better off not attempting it at all.

If you disagree with that approach, then I'm afraid that it is your duty to educate/familiarize yourself with the process you are planning to attempt until you can use the tools of the trade (just like, if you are planning to change the starter of your car, which is a nontrivial operation, you shouldn't expect the notice on the new starter, if there is even one, to be oriented towards anybody else but people who have a minimum of car mechanic know-how).

> So, could you please straighten out the confusion by identifying the purpose and capability and result of these features in a UI that is to some extent more lucid?

I could __attempt__ to. But I am not going to do that, because:
1. Considering that a lot of it boils down to a matter of what a specific set of individual __believes__ an intuitive UI _should_ look like, it's mostly a fools' errand, since you won't be able to satisfy everyone (unless you do design your UI for that one person who screams the loudest, but then you'll only gather the scorn of others like them which, despite what that one person may believe, happen not to share their allegedly "universal" opinion of how an intuitive UI should be designed). I have designed enough UIs to know that the best you can ever accomplish in terms of "intuitiveness" is a compromise that, and I can guarantee it, does not ever leave anyone properly satisfied.
2. Even if I was happy with trying a _new_ compromise (because, even as it may not look like it, the current UI already is one!) it would require a lot of time and effort, which I don't have, and which you seem to expect me to invest for free for a request that exceedingly few people besides you happen to have, as demonstrated by the fact that, even as it has existed for 10 years, this issue tracker is rather devoid of other requests like yours.
3. I don't want to provide the expectation that Rufus (and by extension handling a boot process or installing an OS) should be accessible to people who I estimate are likely to shoot themselves in the foot on account that, if they don't understand what Rufus is telling them in its UI, I have to consider that they are also very unlikely to understand what the boot process is going to tell them if, for instance, is starts talking about enabling/disabling Secure Boot, creating partitions and so on, or if it produces any kind of error. It may be counter intuitive, and you may even qualify it as an attempt against your freedom to use a product in whichever way you like but my take on this is that as long as a user has trouble with the Rufus UI or the elements that are required for the use of Rufus, they are better off not attempting to boot a system with media created by Rufus at all, as there is a good chance they will do something wrong and lose data they didn't plan to lose. So, and this is the part that I expect to be counter-intuitive, if not infuriating, to you, whereas the UI of Rufus is actually relatively simple (and tries to be "intuitive" where possible), it is not and will never be designed to be so simple as to make the boot process creation accessible to everyone and their grandmothers. By design, and through use of explicit boot terminology, I did bake out into the UI a minimum level of filtering out of end users who I estimate are likely to make matters worse if they come to Rufus with the idea that this is an application that should be technically as accessible as, say, a video or messaging application.

What I usually do then, for the small amount users who indicate that they have trouble understanding how they should use Rufus, is point them to perform a search for youtube videos of Rufus usage for a context similar to the one they plan to use Rufus for (Windows installation, Linux installation, DOS boot, and so on), and, once they have watched these videos and have had an example of the manner in which one uses the Rufus UI, complete whatever blanks there might be left for the process they are trying to accomplish by doing additional research.

But if you come to Rufus with the idea that it's the job of its developer to hide the "scary terminology" or make its UI so simple that a 4 year old should be able to use it, in order to pretend that creating and using boot media should be considered a trivial operation, you should understand that I am never going to follow you there, as a consider that booting is not and never will be trivial, and that trying to bastardise it as such is only going to make end-users lives worse, because then they are likely to be too ill-prepared to address the various issues that can be thrown at them during boot..

closed 2022-06-17 by pbatard