pbatard/rufus #1956: tpm

senolfndk, 2022-06-24:
Why can't I see the tpm bypass feature removed in the beta version?
[image]

pbatard, 2022-06-24:
Press START. It was just moved to a different place:
[image]

senolfndk, 2022-06-24:
When you press start, it switches to typing.
[image]

pbatard, 2022-06-24:
Ah I see. You need to make sure that partition scheme is set to GPT, then you'll see it. I'll try to fix MBR for the release.

senolfndk, 2022-06-25:
ok

pbatard, 2022-06-25:
Fix has been applied for the release, but I will keep this issue opened for visibility.

leonsk29, 2022-06-25:
> Ah I see. You need to make sure that partition scheme is set to GPT, then you'll see it. I'll try to fix MBR for the release.

I have selected GPT as the partitioning scheme in Rufus 3.19 BETA with Windows 11 install media and still doesn't show the Windows User Experience Dialog after clicking START, it just starts formatting and writing the image to the drive without asking anything else except confirmation.

[image]

[image]

leonsk29, 2022-06-25:
> 

Oh, sorry, I posted seconds before you published the fix. I'll try it now. Sorry.

pbatard, 2022-06-25:
@leonsk29, you should have seen it with GPT.
It seems you are using a custom ISO though.
Can you please post your log? The customization options are dependent on Rufus being able to read the Windows version an build number from the ISO (which it should do fine with __official__ ISOs, but which I cannot guarantee with unofficial ones).
You can access the log with <kbd>Ctrl</kbd>-<kbd>L</kbd> or the small log button left of `START`.
Please paste your log between triple backtick marks like this:
````
```
<Your log>
```
````

pbatard, 2022-06-25:
Also where exactly did you get your Windows 11 ISO?

senolfndk, 2022-06-25:
👍
[image]

leonsk29, 2022-06-25:
> Also where exactly did you get your Windows 11 ISO?

You're right, I didn't remember that was a custom ISO involving Windows 10 setup media with a Windows 11 image. Now I tested it with a proper Windows 11 22H2 ISO and it shows up. So sorry for bothering you, my mistake.

pbatard, 2022-06-25:
Thanks for the update, no worries.

A Windows 10 setup will definitely screw our version detection since we look into `sources/boot.wim` to find if we're dealing with Windows 11.

For the record, you want to see something like `Detected: Windows 11 ISO (Build 22621.1)` in the log when Rufus scans the image, because, if you don't get the report that Rufus detected Windows 11, you won't get the options.

TP555, 2022-06-25:
Hi

Where i can get the 3,19.1907 Fixed version  ?

pbatard, 2022-06-25:
> Where i can get the 3,19.1907 Fixed version ?

When it is released on July 1st (**or later**).

If you really can't wait, you can always pick the ALPHA build from the [MinGW Artifcats](https://github.com/pbatard/rufus/actions/runs/2560985467).

pbatard, 2022-06-28:
Please don't close this issue. It is fixed, but I want to leave it open for visibility until the release.

pbatard, 2022-06-29:
Locked, as I already explained why I want to keep that issue opened, and appear to be ignored.

pbatard, 2022-07-01:
Closing this issue with the release of Rufus 3.19.

commit d457a0ba (2022-06-25): [ui] fix Windows 11 customization options not appearing with MBR

* Fixes #1956
closed 2022-07-01 by pbatard