pbatard/rufus #1957: Windows 11 ISO 21H2 (Build 22000.318) won't work for the MS Account Fix ?

TP555, 2022-06-25:
Hi

i can't bypass the Account Check , this only Works with 22H2  ?
I had tried with Unplug or Full Disable LAN  !

Rufus/Settings etc.

Windows 11 ISO 21H2 (Build 22000.318) German
Partition: MBR
Rufus 3.19.1907

pbatard, 2022-06-25:
Yes, it only applies to 22H2.

If you are using Windows 11 21H2, the bypass should not be needed unless you are running Windows Home.
Are you running Windows Home edition?

TP555, 2022-06-25:
@pbatard

Yes sry , Win 11 Home  !

pbatard, 2022-06-25:
Then I'm afraid you're on your own. I am not planning to do anything for Windows 21H2 Home edition at this stage. Maybe in the future, but definitely not for this release.

closed 2022-06-25 by pbatard

TP555, 2022-06-25:
@pbatard

But not forgot , 22H2 [Sun Valley 2 isn't official or Final Released now  !
Release Date : may Early/Late Autumn  !

And in the 22H2 Pro, MS also wants to introduce compulsory accounts for the Pro version !
May re think about a small Fix for 21H2 Users

pbatard, 2022-06-25:
> May re think about a small Fix for 21H2 Users

Not gonna happen for 3.19. I already have plans for how I am going to handle 21H2, but those plans are not part of the 3.19 release, and I am not going to start to add piecemeal features that late in a release cycle.

Outside of a libcdio bug that needs fixing, what will go into the 3.19 release is pretty much frozen at this stage and only bugfixes will be accepted... which this isn't. since I did design the Microsoft account bypass for 22H2 __only__.