pbatard/rufus #1963: Missing "Remove requirement for an online Microsoft account"

Dantasstic, 2022-07-02:
[image]

Remove account requirement for Windows 11 is missing

WereCatf, 2022-07-02:
I just tried it myself with a freshly-downloaded ISO and yeah, I'm missing the option as well. There's a check in the source-code for build-version of 22500 or higher, but it looks like it's not detecting the build-version correctly and thus the option doesn't show up.

Dantasstic, 2022-07-02:
The ISO I used was freshly downloaded today through Rufus itself. I grabbed the newest Win 11 build.

WereCatf, 2022-07-02:
Rufus's logs say my freshly-downloaded image is build-version 22000.194, so obviously the check fails. Removing the build-version check from rufus.c and recompiling Rufus does enable the option to disable Microsoft-account requirement.

I'm not a Windows-nerd so I have no idea how those build-versions are supposed to work, though. Is 22500 some Insider-release or something?

JtMotoX, 2022-07-02:
I went through with the USB creation and installed Windows 11 just to see if maybe Rufus removes this requirement by default.  It does not.  The Windows 11 setup is still requiring a Microsoft Account.

WereCatf, 2022-07-02:
I just said that there's a build-version check and I even told which file it is in. I also mentioned that you'd have to remove the check and recompile Rufus. I thought I was pretty clear, but apparently not.

JtMotoX, 2022-07-02:
> I just said that there's a build-version check and I even told which file it is in. I also mentioned that you'd have to remove the check and recompile Rufus. I thought I was pretty clear, but apparently not.

Did somebody have a bad day today?

> but it looks like it's not detecting the build-version correctly

I had just downloaded a new copy of the Windows 11 Media Creation Tool and it downloaded and created a Windows 11 installation ISO.  DISM is reporting that it is version 22000.653. So Rufus is detecting the build number just fine. Not sure why you believe that Rufus is detecting it incorrectly.

WereCatf, 2022-07-02:
> I had just downloaded a new copy of the Windows 11 Media Creation Tool and it downloaded and created a Windows 11 installation ISO. DISM is reporting that it is version 22000.653. So Rufus is detecting the build number just fine. Not sure why you believe that Rufus is detecting it incorrectly.

Rufus checks for build-version 22500 or greater, that's why. 22000 is less than 22500, which is why the check fails.

JtMotoX, 2022-07-02:
According to the changelog:

`Also allows to bypass the mandatory requirement for a Microsoft account with Windows 11 22H2`

It looks like this is for the upcoming 22H2 release which is currently only available on the Preview channel and is not expected to be officially released to the public until October.

For the current release (22000), you simply disconnect the network and it will allow you to create a local (non-microsoft) account.  For 22H2 release, you will need this bypass (along with disconnecting from internet) in order to create a local account.

pbatard, 2022-07-02:
Yes, the bypass only applies to __22H2__, as mentioned in the ChangeLog notes.

It really needs to be stressed out that all the account bypass does, for 22H2, is __restore__ the behaviour that exists in 21H2 where one can unplug the network to be able to create a local account.

So, __obviously__, since the local account bypass is to make 22H2 behave like 21H2, it is __only__ proposed if you use a 22H2 ISO.

Again, from the ChangeLog notes (emphasis mine);

> * Also allows to bypass the mandatory requirement for a Microsoft account with Windows 11 __22H2__  
(NB: __Network MUST be temporarily disabled or unplugged for the local account creation to be proposed__)

And yeah, I'll probably add a new FAQ section about this, because it sure seems to confuse a lot of people.

closed 2022-07-02 by pbatard