pbatard/rufus #1966: Alternative method of skipping all collection questions

msoxzw, 2022-07-02:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Currently, the Rufus uses [unattend files](https://docs.microsoft.com/windows-hardware/manufacture/desktop/update-windows-settings-and-scripts-create-your-own-answer-file-sxs) to customize Windows. However, there are some meaningless numbers and processor architecture in unattend files, and hence, unattend files are not very friendly to cross operating system, e.g., Rufus 3.19 can skip all collection questions only for Windows 11.

Rufus 3.19 uses `ProtectYourPC` to skip these questions:
https://github.com/pbatard/rufus/blob/3cf41abf2e3dfe8cdb52a7ebd734eeb49731cc10/src/rufus.c#L1344

The `ProtectYourPC` is used for turning on automatic update for Windows 7, Windows 8, and Windows 8.1. Actually, setting `ProtectYourPC` as 1 could also skip and deny these questions, and many options in unattend files are written to registry.

Some [OOBE settings](https://docs.microsoft.com/windows-hardware/customize/desktop/unattend/microsoft-windows-shell-setup-oobe) including `ProtectYourPC` are reflected in [OOBE.reg](https://github.com/msoxzw/quiet-windows-10/blob/master/sources/%24OEM%24/%241/Software/Configuration/Registry/System/OOBE.reg), though I have not tested all settings. In addition, [`Setupcomplete.cmd`](https://docs.microsoft.com/windows-hardware/manufacture/desktop/add-a-custom-script-to-windows-setup) could be used to import these registries regardless of Windows version and processor architecture. [Here](https://github.com/msoxzw/quiet-windows-10/tree/master/sources/%24OEM%24/%24%24/Setup/Scripts) are my custom scripts, which, I think, are more flexible than unattend files, e.g. "OneDrive" could be prevented from being installed before a new account is created.

Log
---
```
<FULL LOG>
```

pbatard, 2022-07-03:
Thanks, but I don't think I'm going to follow your suggestion. There are quite a few points to cover here, so here goes:

1. Per the link you'll see in the source, ` <ProtectYourPC>3</ProtectYourPC>` is designed to answer no to all the personalization questions. My understanding is that using `3` will not disable automatic updates, which is what I want.
2. Rufus 3.19 __is designed__ to only skip collection questions for Windows 11. That's 100% intended. I may possibly extend it for Windows 10, but no further.
3. As far as I know, to cover Windows 10 and Windows 11, there's only the arch that needs to be set, which is really a no brainer. All the rest is common boilerplate. I have absolutely zero need to design the unattend from Rufus to be friendly to cross operating systems, since they are designed for the single operation system installation files that are on the boot media.
4. The use of an unattend allows me not to have to multiply methods of customizing Windows boot settings. Especially, I need to be able to set internal disks offline for Windows To Go, for which I can use the `offlineServicing` pass of the unattend __on the applied image__ (and not at boot time). I don't believe your script would be of any help for that, so if I were to use it, I'd have to break what's currently nicely factorized into a single location.
5. Unattend tasks are exceedingly well documented by Microsoft, and, I will assert, are the much better approach in an application that needs to demonstrate that it is beyond the speck of a doubt with regards to how it will alter Windows settings at boottime. While the script solution is elegant, it has the huge drawback of being a script running ill documented commands, that people, who want to assert that it is not doing anything unwarranted __will__ have a lot more trouble validating than the well documented `unattend.xml`.
6. I am only interested in addressing tasks that are a hurdle __during setup__ or are downright preventing the installation to proceed. Anything that can be accomplished by the user post install (such as removing OneDrive) is pretty much out of scope of what I am planning to touch with Rufus. In short, I am not designing Rufus as a utility that will provide full customization of your Windows installation settings &mdash; There are way too many of those, and it would be a complete nightmare to try to satisfy all the requests I would get then (_"Please add an option to turn on dark mode by default!"_, etc.). So let me go over the current options Rufus provides and why they are there:
    - Bypass of TPM and Secure Boot &rarr; _User may not be able to install at all if these aren't enabled, so these are obviously needed_
    - Bypass of RAM/Disk min size &rarr; _Same as above_
    - Bypass of MSA &rarr; _Well, technically, the user can proceed (if they have an internet connection), but it's a __major__ inconvenience and having to create accounts is downright unfriendly for users, so of course, we want that_
    - Set internal drives offline (Windows To Go) &rarr; _Required because if you don't do that and have any ReFS drives on a Windows 10 platform, booting Windows 11 To will upgrade all these drives to the latest version of ReFS __and prevent Windows 10 from ever being able to access them again__, so, because the upgrade happens silently __at boot time__ you sure want an option during the drive creation that prevents Windows To Go from accessing internal drives_
    - Skip all the personalization data collection questions and answer no to all &rarr; _While not technically a showstopper during installation, having to go through all the dialogs during an install that you want to get through as fast as possible, just to answer no, is still a major pain in the ass, so if we can easily add an option to skip these, we sure will_

At this stage, besides _potentially_ bringing some of the options above to Windows 10 (but no further down) the only other customization option I am considering adding is to allow the user to default the locale to the same as the one as their current system (i.e. skip the locale selection on installer boot, which is also a major time waster during install whereas most people will use the same locale for the system they want to install as the one from the system they created the media on), and that's about it. I'm actually not sure this latter customization is something that I would be able to accomplish with your script (whereas it's definitely something that unattend is designed to do), but, again, you'll noticed that the only customization Rufus will bother itself with is stuff that will __explicitly__ nag or hinder the user during Windows installation. Anything else, and especially anything that the user will have the capability to customized from their installed Windows system is out of scope.

For all these reasons, whereas I appreciate your suggestion, I don't think I want to move away from using an unattend file. And I'll also add that you may want to be careful not to look at what other people are trying to solve through your specific lens, because different applications and environments usually try to solve very different problems, and therefore a solution that might be most appropriate to one may not be that appropriate to the other.

&rarr; Rejected

closed 2022-07-03 by pbatard

msoxzw, 2022-07-08:
@pbatard  I fully appreciate your judgement and decision, but I still want to clarify some of my viewpoints.

> 1. Per the link you'll see in the source, <ProtectYourPC>3</ProtectYourPC> is designed to answer no to all the personalization questions. My understanding is that using 3 will not disable automatic updates, which is what I want.
> 2. Rufus 3.19 is designed to only skip collection questions for Windows 11. That's 100% intended. I may possibly extend it for Windows 10, but no further.
> 3. As far as I know, to cover Windows 10 and Windows 11, there's only the arch that needs to be set, which is really a no brainer. All the rest is common boilerplate. I have absolutely zero need to design the unattend from Rufus to be friendly to cross operating systems, since they are designed for the single operating system installation files that are on the boot media.

According to [the previous Microsoft document](https://technet.microsoft.com/en-us/library/cc749278(v=ws.10).aspx), the "ProtectYourPC" actually stems from Windows Vista, which is used for selection among important, optional, automatic, and manual updates. From Windows 10 onward, updates have become mandatory, and thus the "ProtectYourPC" also lost its original meaning, which causes the oddness of the same effect of value 1 and value 2 in [the current document](https://docs.microsoft.com/windows-hardware/customize/desktop/unattend/microsoft-windows-shell-setup-oobe-protectyourpc). More strangely, all of the values 1, 2, 3 of the "ProtectYourPC" have the consistent result. Therefore, for me, it is very attractive to use one setting from Windows Vista to Windows 11, and presumably Windows version in the future, during and after installation, without meaningless words and numbers, though undocumented. 

> 6. I am only interested in addressing tasks that are a hurdle during setup or are downright preventing the installation to proceed. Anything that can be accomplished by the user post install (such as removing OneDrive) is pretty much out of scope of what I am planning to touch with Rufus. In short, I am not designing Rufus as a utility that will provide full customization of your Windows installation settings — There are way too many of those, and it would be a complete nightmare to try to satisfy all the requests I would get then ("Please add an option to turn on dark mode by default!", etc.).

I mean to use custom script to skip personalization questions rather than to use my custom script or anything else. Moreover, the OneDrive is not installed instead of removed, and users can install it anytime.
[image]
The OneDrive installer is located in the "C:\Windows\SysWOW64\OneDriveSetup.exe", which is installed at the first sign-in of new account.

pbatard, 2022-07-08:
> and thus the "ProtectYourPC" also lost its original meaning

[Tell me about it](https://github.com/pbatard/rufus/blob/f17e287e397c52392d608422aa25baf92a16f9be/src/rufus.c#L1339-L1346)...

> Therefore, for me, it is very attractive to use one setting from Windows Vista to Windows 11, and presumably Windows version in the future, during and after installation, without meaningless words and numbers, though undocumented.

Well, https://docs.microsoft.com/en-us/windows-hardware/customize/desktop/unattend/microsoft-windows-shell-setup-oobe-protectyourpc kind of documents it, and, again, in an Open Source application that needs to demonstrate that it's not doing anything dodgy, I'll always try to go with what's documented where possible.

> Moreover, the OneDrive is not installed instead of removed, and users can install it anytime.

That's a good point, but I'm still not sure I want to have to test and support OneDrive non-installation with Rufus, because, from reading around, it looks like different people advertise different ways of accomplishing this, and since it's undocumented, I don't want to end up in a situation where the latest version of Windows breaks that method and I end up with people complaining that the OneDrive skip doesn't work. As much as I do prefer systems without OneDrive, there;s just too much of a chance adding the feature will eventually become a hassle to support. Plus, considering the amount of elements one could add to trying to declutter Windows (some of which you apply in your script), I'm also pretty much guaranteed requests for _"You added an option to disable OneDrive installation, so why won't you add an option to disable X?"_ and I'll never see the end of it.

So, even if it'd make sense to apply customization during media creation rather than post install, I still think I'm going to stick with limiting what Rufus deals with to actual showstoppers or annoyances that are explicitly thrown at the users during the setup phase only.

msoxzw, 2022-07-08:
I appreciate the significance and the value of documentation that gives a reliable guarantee to usability. It also makes the long-term maintenance possible to Do One Thing and Do It Well.