pbatard/rufus #1967: Windows automount stays disabled for written drives even after reboot

s2do, 2022-07-04:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
After writing, the drive does not get a letter (D:, E:, ...) from Windows automatically. It does not "auto-mount" any more. Giving a letter per hand solved the problem. This is something impossible without knowledge and admin rights.

It can be repeated every time, no matter what image or hardware. If one dongle used in one USB port, that port get disabled, others work properly.

Log
---
Rufus x86 v3.18.1877 (Portable)
Windows version: Windows 10 Education, 64-bit (Build 19043.1766)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'Generic Mass-Storage USB Device' (1908:0226)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3697, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0EE3E8A8
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 256 MB (268435456 bytes)
  Start Sector: 8192, Boot: No
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: ext4
  Size: 1.6 GB (1728053248 bytes)
  Start Sector: 532480, Boot: No
Scanning image...
ISO analysis:
  'C:\Users\ssolerpe\ownCloud\lehre\iCPS\2022-04-04-raspios-bullseye-arm64-lite.img.xz' doesn't look like an ISO image
Disk image analysis:
Error: unsupported check; not verifying file integrity
  Image is a compressed bootable disk image
Using image: 2022-04-04-raspios-bullseye-arm64-lite.img.xz (270.4 MB)

Computing checksum for 'C:\Users\ssolerpe\ownCloud\lehre\iCPS\2022-04-04-raspios-bullseye-arm64-lite.img.xz'...
  MD5:    36373e2eceb141075b739944f2991030
  SHA1:   73b83f329220f808d174a9c30a2f092951da1fbd
  SHA256: 35f1d2f4105e01f4ca888ab4ced6912411e82a2539c53c9e4e6b795f25275a1f

Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Opened \\.\PhysicalDrive1 for exclusive write access
Requesting logical volume handle...
Writing compressed image:
+Error: unsupported check; not verifying file integrity
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

Found USB 2.0 device 'Generic Mass-Storage USB Device' (1908:0226)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3697, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0EE3E8A8
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 256 MB (268435456 bytes)
  Start Sector: 8192, Boot: No
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: ext4
  Size: 1.6 GB (1728053248 bytes)
  Start Sector: 532480, Boot: No

pbatard, 2022-07-04:
This is intentional, per [this comment from a previous issue](https://github.com/pbatard/rufus/issues/1842#issuecomment-998758284).

If your image contains an ESP (which it does since it has a 256 MB FAT partition), then Rufus deliberately does not assign a letter to prevent media check, like the one conducted by Ubuntu, from failing.

I'm afraid that, unless I start to waste a lot of time trying to maintain a list of DD images that should or should not be reassigned a drive letter, I have no choice but to dissatisfy a minority of users (such as RaspiOS ones) in order to satisfy a majority of other users (Ubuntu ones).

Closed as "by design".

closed 2022-07-04 by pbatard

s2do, 2022-07-04:
I think, that keep it disabled by design will generate more trouble than otherwise... at least with SD Card Readers. Maybe that way would be easier. But OK if it is added to FAQs ;)