pbatard/rufus #1969: Error when try to make Windows 11

daeeros, 2022-07-05:
I try to make windows 11 in my flash drive, i using GPT, and enable bypass TMP, SECURE BOOT, DRIVE SPACE, RAM.
But when it in step NTFS check i olvays get error, why?
This is log
[rufus.log](https://github.com/pbatard/rufus/files/9046943/rufus.log)

pbatard, 2022-07-05:
1. Please reboot your computer and try again
2. If you still see the error, please close Rufus, then from an elevated command prompt type:  
    `dism /cleanup-wim`

All in all, it looks like you had an error in Rufus, before the session you provided a log for, that prevented the `boot.wim` from being unmounted, and the issues you are seeing in this session are a direct result of that. I would have liked to see that original error because something happened before you got that `[0xC1420127] The specified image in the specified wim is already mounted for read and write access` error from your log, and knowing that would help me try to fix the issue (instead of trying to add workarounds for mounted images down the line).

daeeros, 2022-07-05:
> 1. Please reboot your computer and try again
> 2. If you still see the error, please close Rufus, then from an elevated command prompt type:
>    `dism /cleanup-wim`
>
> All in all, it looks like you had [...]

Reboot my pc is fixed, thank you <3

closed 2022-07-05 by daeeros