pbatard/rufus #1970: No image selected

sardiani, 2022-07-05:
Hi, i cant browse for locate ISO files

Log file show:

Could not show FileOpenDialog: Windows error code 0x80004005 (FormatMessage error code 0x00003AFC)
Could not select file for open. Error FFFF
Could not show FileOpenDialog: Windows error code 0x80004005
Could not select file for open. Error FFFF

Rufus x86 v3.19.1911 (Portable)
SO: Win10 
can help me?

pbatard, 2022-07-05:
I still need a full log. If you provide a partial log, I cannot help you.

sardiani, 2022-07-05:
ok,...
Rufus x86 v3.19.1911 (Portable)
Windows version: Windows 10 Pro, 64-bit (Build 19043.1766)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0C0A (es-ES)
NOTE: No Windows Language Pack is installed for English (English) on this system.
This means that some controls may still be displayed using the system locale.
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Kingston DT 101 G2 USB Device' (0930:6545)
Found USB 2.0 device 'SAMSUNG HM500JI USB Device' (18A5:0225)
Device eliminated because it was detected as a Hard Drive (score 6 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 3.0 device 'WD Elements 25A2 USB Device' (1058:25A2)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
1 device found
Disk type: Removable, Disk size: 7.7GB, Sector size: 512 bytes
Cylinders: 941, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 7.2 GB (7739867136 bytes)
  Start Sector: 8064, Boot: Yes
Could not show FileOpenDialog: Windows error code 0x80004005 (FormatMessage error code 0x00003AFC)
Could not select file for open. Error FFFF

pbatard, 2022-07-05:
The `NOTE: No Windows Language Pack is installed for English (English) on this system.` warning along with getting a `FormatMessage error code` tells me that your installation of Windows differs from vanilla, in that you either removed or altered components that are essential for Rufus to be able to run.

The fact that you then receive a `Windows error code 0x80004005` (`E_FAIL`) afterwards is therefore not surprising.

For the record, Rufus actually tries not one but __two__ different methods to display common dialogs like the one for File Selection. So there is really nothing more I can do here: For reason that __you__ will have to determine, your system deviates too much from a vanilla installation of Windows to allow Rufus to do perform its job, and, considering that [you are the only person to ever report error code `0x80004005`](https://github.com/pbatard/rufus/issues?q=0x80004005) (and that I have not changed anything relating to the file selection dialog in recent versions of Rufus), I'm afraid that I am not going to invest any time to help you investigate in what manner you have altered your Windows environment to make Rufus fail to select an image.

From where I stand, your problem is 100% due to your environment, and not Rufus. As such, I will close this issue.

closed 2022-07-05 by pbatard