pbatard/rufus #1973: Windows 11 options not shown

asrusch, 2022-07-07:
I am using Rufus 3.19.1911 and after selecting my Windows 11 ISO (Business editions, updated June 2022), I don't get any Windows 11 specific options like the Changelog says I should be seeing...

![image](https://user-images.githubusercontent.com/17188253/177880941-02da4840-5ebe-48c5-9296-197585badb60.png
[rufus-3.19.log](https://github.com/pbatard/rufus/files/9067643/rufus-3.19.log)
)

closed 2022-07-07 by pbatard

asrusch, 2022-07-07:
Thank you- and this feature:
"Also add an option to skip all collection questions (Sets all answers to "Don't allow")"

Is that also for 22H2 as well?

pbatard, 2022-07-07:
You've seen this exact dialog, right?
[image]

What do you think?