pbatard/rufus #1979: 3.19 have malware

natytgn, 2022-07-09:
''it has malware called [Win32:MalwareX-gen [Trj]

alessio-businaro, 2022-07-09:
same issue, after download W11.iso start to apply into USB key and at 99.x%  Avast find a malware called [Win32:MalwareX-gen [Trj]'

Janvier123, 2022-07-09:
Same here on AVG 
[image]

pbatard, 2022-07-09:
1. Make sure to download from https://rufus.ie. Because of Rufus' popularity, there are fake sites out there that publish malicious versions of Rufus (but these won't have a __valid__ _Akeo Consulting_ signature, which official versions of Rufus all do)
2. The [current VirusTotal page for the Rufus executable](https://www.virustotal.com/gui/file/75827940e5d821f30951e641772696bab6d97e37e3479f70e1ca4b8cb4d40132/detection) does not confirm your report (For the record, Antiy-AVL, which is currently reporting a "generic" Trojan, meaning that they don't actually have anything that was formally identified by human engineers, is a simple false positive, and the "Sigma" rules you see are more ill tuned automated bullshit, since I'd really like to know where the heck they got python initiated threats from something that can be demonstrated to not use any python). Oh, and since all of you guys are reporting 'gen' (i.e. "Generic") malware, that also applies to your reports. Again, a "Generic" malware is something that was NEVER validated by a human person to be malware (because otherwise it would have a non generic name), so, in 99% of the cases, it is a false positive.
3. If you are seeing a report of a thread when an application (that wasn't flagged when you opened it) is scanning the full content of an ISO to compute its checksum, the report is __CLEARLY__ for what is on the ISO, not for Rufus. For instance, if you scan an ISO that contains malware or elements that are often identified as malware (which will be the case for Kali Linux) you will get AntiVirus notifications and they have __nothing__ to do with Rufus, but come from the Kali ISO itself.
4. See [this](https://github.com/pbatard/rufus/wiki/FAQ#Antivirus_X_reports_that_Rufus_contains_malware). If you want to maintain that Rufus contains a Virus, I'm afraid it is _your job_ to __ FIND A HUMAN PERSON WORKING FOR A SECURITY COMPANY THAT WILL CONFIRM THAT YOUR SCARY REPORT IS NOT YET ANOTHER ONE IN A VERY LOOOOOOOOONG LIST OF FALSE POSITIVES__. Especially, they (or you) will have to be able to point in the public code where exactly that malicious behaviour is located, [since it is formally impossible for Rufus, and this can be FORMALLY DEMONSTRATED, to have hidden something that does not come from its very public source](https://github.com/pbatard/rufus/wiki/Security#Tamperproof_signed_executable). Therefore, if you or someone want to scream that _"OMG, Rufus contains a Virus!!!1!"_, you __MUST__ be able to point to malicious code in our public source, since there isn't a single part of the official Rufus executable that doesn't come from there.
5. In the 10 years since Rufus was first release, we have gotten these reports about every month (because if there is one thing AV vendors are actually known for, it's __WRONGLY__ misidentifying threats and reporting that something contains malware when it doesn't, i.e. _false positives_) and whereas we have had many people argue that _"But if mutliple AV engines report it, it must be true"_, over these 10 years, there still hasn't been any substantiated evidence that any single version of Rufus has ever contained malware. So, either I am incredibly smart at hiding malware in plain sight (since, again, it is impossible for me to hide it anywhere but in the public code, since I don't actually compile the executable on any of my machines) and an army of security researchers are doing a very lousy job at identifying malware (in which case, since I allegedly managed to make Rufus evade their detection for a decade there isn't much point of using any of their solution) or all these security solutions are __wrong__ when they report that Rufus contains malware, especially if that report was never validated by an actual human person from the company that makes the solution.
6. I'm growing EXCEEDINGLY TIRED of this bullshit dance with AV vendors. You may disagree with this but, it is really __YOUR__ job to get an actual human person from these security company to confirm their claim __before__ you contact me. A robot does not make someone guilty. A human person does. So get back to me after you contacted AVG, or AVAST to report a potential false positive and one of their employee replies that this isn't just another false positive (and, if they do, please make sure to have their engineers explain where the heck that malware is in our public code, since, again, [it is impossible for us not to have it in plain sight there](https://github.com/pbatard/rufus/wiki/Security#Tamperproof_signed_executable).

closed 2022-07-09 by pbatard

StoneCypher, 2022-08-13:
> You may disagree with this but, it is really YOUR job to get an actual human person from these security company to confirm their claim before you contact me. 

Please remember that each time you interact with these folks, it's going to be different folks, and they're unlikely to have seen the last time this happened to you

I know they're frustrating, but they're just trying to help

Given how frequently this comes your way, making a prefabricated page about the topic that you could defer to might help

pbatard, 2022-08-13:
> Given how frequently this comes your way, making a prefabricated page about the topic that you could defer to might help

https://github.com/pbatard/rufus/wiki/FAQ#Antivirus_X_reports_that_Rufus_contains_malware

StoneCypher, 2022-08-13:
I'm sympathetic.  I used to be a principal at a security company.

For what it's worth, if anything, I find your tool quite impressive, and part of the reason you have to deal with this is you made something a lot of people really rely on