pbatard/rufus #1982: A USB-stick with Ubuntu 22.04 only boots to FreeDOS

Tlaq-ART, 2022-07-12:
Can't see any comment on my issue, running Rufus 3.19p (portable)

Trying for the 7:th time to make a USB-stick boot of Ubuntu 22.04. Follow the instructions. On an older stick, it takes 4 h to install, and on a newer some 2.5 h. Following the instructions as shown on Rufus.ie to config, but areas like Partition Schema: MBR and Target sys: Bios or UEFI-CFM as well as all other choices except multiple USB disks are totally greyed out. 

As said, when booting I only get the FreeDOS sys, not the Ubuntu image I hope I installed.  Have searched the whole internet, but those comments I see are listed regarding windows isos' or target system, I am not allowed to change. I chose the portable since it was a one-time install

pbatard, 2022-07-12:
> Following the instructions as shown on Rufus.ie to config

There are no instructions on rufus.ie. There's a screenshot, but the fact that it mentions Ubuntu is irrelevant (and you should __not__ use it as a guide).

>  I only get the FreeDOS sys, not the Ubuntu image

Then you didn't create the media properly.

Please pick any video from [here](https://www.google.com/search?q=rufus+ubuntu+yt) and study it.
And remember, you first need to download the Ubuntu ISO by yourself from [here](https://ubuntu.com/download/desktop), as you won't be able to create an Ubuntu bootable drive in Rufus unless you have the ISO, just like you wouldn't be able to burn Ubuntu to an optical disc unless you download the Ubuntu ISO first.

As I expect that watching the youtube videos on how to create an Ubuntu media in Rufus will solve your issue, I will close it.

closed 2022-07-12 by pbatard

Tlaq-ART, 2022-07-13:
Problem was with the Ubuntu desktop 22.04 iso, changed to the server image and everything worked as described.  Exactly the same process of generation but the desktop image took 2.5 - 4 hours while the server image took 15 min.