pbatard/rufus #1989: Feedback: Used your UEFI stuff to make bootable USB on M1 Macs, FYI

sawyermade, 2022-07-20:
﻿

Checklist
---------
- [ x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x ] I also tried one or more of the following:
  - [ x] Using a different USB drive.
  - [x ] Plugging the USB into a different port.
  - [x ] Running Rufus on a different computer.
- [ x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
Not really an issue but wasn't sure how to get a hold of you guys for feedback but said on the readme to use the issue tracker but I used your UEFI boot files to create a bootable USB for Win10 on M1 Pro MBP and it seems to work like a charm. Its a little ghetto but idk if you guys are interested in getting this to work on macOS? Let me know if there's a better place to talk about this or whatever and thanks for all the hard work, everyone I know I tell to use Rufus for windows bootable but there is a real lack of support for doing it in linux and apple silicon macOS. Thanks for your time and here's the link to the repo, its basically mostly a readme with your EFI files and works like a charm on the couple of systems Ive tested it on: https://github.com/sawyermade/macos_win10_usb 
Log
---
```
<FULL LOG>
```

pbatard, 2022-07-20:
Thanks for the report, but I'm not sure there's much I want to do here since, if an ARM64 based machine is UEFI compliant, then one should indeed be able to use UEFI:NTFS to boot an UEFI bootloader residing on an NTFS partition, and there's really nothing Mac specific to it. At least it's good to know that, for once, Apple might have decided not to do something "their own way" and instead appear to be using the UEFI established standards.

sawyermade, 2022-07-21:
The point isnt to boot the USB on the Mac but to be able to create a bootable USB for non-Mac PCs since there isnt really a nice and easy method to do so. I updated it since original post and use ExFAT now and is much simpler. Here is the script and it just uses you guys efi files. Everything I've seen online is much more complicated and/or uses proprietary software, this just uses built in Mac CLI shit. https://github.com/sawyermade/macos_win10_usb/blob/master/win10_usb_create.sh

sawyermade, 2022-07-21:
I'd actually be interested in creating a macOS "Rufus" GUI where they can select the USB Drive, the ISO, and then creates the ExFAT Windows bootable USB for install on non-Mac PCs using your UEFI files if that would be something you guys would be interested in? It would be a good Xcode macOS GUI project for me to do on the side. I know a lot of ppl who wish there was a Rufus style program for creating windows bootable USBs on Mac and Linux.

sawyermade, 2022-07-21:
I feel you. I've never heard of WoeUSB but thanks for the info. You guys dont mind if I use your UEFI files and what is the major difference, that Rufus can make persistent Windows USBs?

pbatard, 2022-07-21:
> You guys dont mind if I use your UEFI files 

I don't mind at all. As a matter of fact, the WoeUSB ones pretty much started using them without asking. ;)

> I wasn't planning on doing a Linux one, just macOS one.

I think that you are misreading what I was trying to say. What I mean is that Rufus does much more than create Windows installation media. It also can create Linux bootable media (which you may think any `dd` like utility should be able to do, but Rufus makes sure to also do so in alternates modes besides `dd`), create DOS bootable media, compute MD5/SHA of ISOs, save a USB media to virtual disk, perform a bad blocks check and so on. So, if there was an application that uses the name "Rufus" for Linux, and it was only for creating Windows bootable media, I'd see it as deceptive software, since it would be missing all of the other features that Rufus does provide. So what I wanted to point out when I talked about Linux is that Rufus does much more than create Windows bootable media, and that, for me to be happy with having a macOS software use the "Rufus" name, I'd expect that software to also provide the capability to create Linux bootable media and much of the other features that the Windows version of Rufus does provide.

sawyermade, 2022-07-21:
Ah, I didn't know Rufus did all of that. Thanks for the info and I really appreciate your feedback. Maybe one day it will make it there lmfao but it's good to know I have your permission to use your EFI files. Thanks so much and I seriously appreciate all the work you guys put into this, its easily the best thing on Windows. On linux I used to use unebootin a lot back in the day but that shit sucks now. For linux USBs, I just tend to use BalenaEtcher, which doesn't do windows bootable, but I didn't even realize how much Rufus actually did. Thanks for the info and I really appreciate your feedback bud. You guys are fucking awesome!

sawyermade, 2022-07-21:
Oh and quick question, I'm just curious what made you want to work on Windows stuff? Like, I tend to hate gd Windows for the most part unless I want to play games lmfao. You guys do some awesome work and seems like your skills would be better used on OSs that are more open and whatnot. Either way, it's super cool you guys do this and I know everyone that uses your stuff greatly appreciates it. Thanks again.

pbatard, 2022-07-21:
> what made you want to work on Windows stuff?

I'll do my best to condense it as much as possible but here goes:

In 2009, I unexpectedly found myself with a relatively large amount of free time, which I spent reverse engineering an porting an old obscure computer game that I won't name to modern platforms (Windows, Linux, etc.). During the course of that project, I came to realise that trying to handle user input in a truly universal & cross-platform manner could really be improved on so, once I was done with the port, I decided to see what I could to do try to tackle that problem. One of the things I needed, to craft a Proof of Concept for what I had in mind, was generic USB access across all main platforms (Linux, Mac and Windows). Thus I naturally started to look at whether their existed a library that would help, which led me to libusb 1.0, which did provide what I was after for Mac and Linux, but that was sadly still missing an implementation for Windows. So I started working on producing a Windows backend for that library, and since I still had a relatively large amount of free time in front of me, and a lot of very competent libusb contributors were supporting that effort, I started to pour a lot of development time into it. Unfortunately, the existing project maintainer of that library had to move to different pastures before they could integrate my work. And even more unfortunately, the main project maintainer happened to designate the one person that seemed to be averse to the Windows-backend as the new sole project maintainer. Well, to say that the libusb project became exceedingly dysfunctional after the departure of the original project maintainer, with the new project maintainer purposefully stalling everything in the pursuit of their own utopian idea of what the library should look like, would be an understatement (which can be validated by the fact that, whereas there had then been regular releases of libusb every few months, once the new project maintainer took over, the next release would take __YEARS__ to happen and only did when the threat of a fork of the library materialized). And even as during these years (yes, plural), myself and many other main libusb contributors tried everything we could to get our work formally integrated into the library to finally have it released to the public, there does come a point where seeing your work literally being s(p)at on leads one to look for some alternate project, and preferably one where extensive development work isn't at the mercy of a single individual with delusions of (non-benevolent) dictatorship. And that is when, since I had had some use for such a utility in the past and since (despite all the unfortunate circumstances surrounding the libusb project) I had found that developing for Windows was challenging enough to actually be _fun_, I took it upon myself to create an [Open Source version of a closed source Windows HP utility aimed created DOS bootable USB media](https://pete.akeo.ie/2011/12/rufus-dos-bootable-usb-formatting.html). Unsurprisingly, as it turns out, when you do release work in a timely manner, you also happen to get constructive feedback, which creates a vicious circle of both the software developer, wanting to produce the best possible & useful software they can, and the software users, finding that the applications satisfies their need well enough to want to recommend it to other people.

So to summarize:
1. I started to concentrate on Windows stuff because it so happened that libusb had a Windows-shape hole at the time. Had they had a Mac and Windows backend without a Linux backend, or a Linux and Windows backend without a Mac backend, I may very well have ended up being mostly a Linux or MacOS developer respectively, even if I obviously do have a strong preference for Linux.
2. Because it's next to impossible to find anything that's well thought through and also because Microsoft changes the deal at every opportunity they get, I'm pretty sure that Windows presents a greater challenge to accomplish anything compared to other mainstream OSes, which makes it a lot more _fun_ to develop for. In short, I'd rather have something of a challenge in a project I choose than smooth sailing.
3. Whereas Rufus does indeed help folks install a proprietary OS like Windows, I am quite proud of the fact that it also does help the same people try or install Open Source OSes like GNU/Linux, __in the very same package__. Which means that, if people want to, they can very much use the same tool that lets them carry on with the prison that is proprietary software to actually free themselves from it altogether. One still has to give somebody the tools that'll let them escape from their prison _from inside that prison_, but whether that somebody will actually use them to that effect remains their choice.

pbatard, 2022-07-22:
I'm going to close this issue now.

closed 2022-07-22 by pbatard

sawyermade, 2022-07-22:
Thanks for the comment and I completely. understand what you are talking about. Thanks for taking the time, I appreciate it!