pbatard/rufus #1991: allow to flesh linux hybrid iso image as a dd image

ilyapashuk, 2022-07-21:
hello

hybrid linux iso images may be fleshed to usb drive simply by dd.

but Rufus suggests to flesh them with it's usual scheme with it's own formatting and partitioning.

it usually works, but may break some usecases.

for example debian installer expects the dd-written media and may refuse to find the image if it is fleshed by an other way.

if user selects linux hybrid iso image, rufus should allow to flesh it as a simple dd image.

pbatard, 2022-07-21:
https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_recommend_DD_mode_over_ISO_mode_for_ISOHybrid_images_Surely_DD_is_better

pbatard, 2022-07-21:
Oh and for the record, we test Debian regularly and, since Debian maintainers seem to be one of the few distro maintainers that appear to understand that not limiting themselves to DD mode is __very beneficial for end users__, we're not currently aware of issues related to creating Debian bootable media in ISO mode.

closed 2022-07-21 by pbatard