pbatard/rufus #1994: Unable to patch/setup files for boot

ma744334, 2022-07-23:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
Unable to patch/setup files for boot （win11 22h2 beta)
Log
Applying Windows customization:
Renamed 'I:\sources\appraiserres.dll' → 'I:\sources\appraiserres.bak'
Mounting 'I:\sources\boot.wim'...
Could not mount 'I:\sources\boot.wim [2]' on 'C:\Users\m\AppData\Local\Temp\RufADB7.tmp': [0x00000057] The parameter is incorrect.
Finalizing, please wait...
I:autorun.inf already exists - keeping it
NTFS Fixup (Checkdisk)...

Found USB 2.0 device 'Flash USB Disk USB Device' (6919:3A04)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1912, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {88CF51AD-26CF-4768-AE92-59882F8C1AEC}
Max parts: 128, Start Offset: 17408, Usable = 15728605696 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {99B941CB-56CC-4442-A6EB-DFF546FEA57C}
  Size: 14.6 GB (15726509568 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {4DBE6DB2-5B8B-45A6-9F1B-CF85562FFCF2}
  Size: 1 MB (1048576 bytes)
  Start Sector: 30717887, Attributes: 0x9000000000000000

<FULL LOG>
```

pbatard, 2022-07-23:
Truncated the log &rarr; Closing.

If you want your issue dealt with, please follow the rules. They aren't that hard to follow.

closed 2022-07-23 by pbatard

stephnradd, 2022-07-25:
> Truncated the log → Closing.
> 
> If you want your issue dealt with, please follow the rules. They aren't that hard to follow.

Hey there
I wanna submit a new issue but i couldn't submit it because my issues exceeded maximum character (probably because of long log)
what's the workaround then?

pbatard, 2022-07-25:
> what's the workaround then?

Upload the log as a file to GitHub or to a file sharing site and provide a link to it. Or split your log in 2 halves and copy them in 2 separate comments.

stephnradd, 2022-07-25:
> > what's the workaround then?
> 
> Upload the log as a file to GitHub or to a file sharing site and provide a link to it. Or split your log in 2 halves and copy them in 2 separate comments.

okay, thanks