pbatard/rufus #1995: Error 17 while booting Windows installation media on UEFI mode

stephnradd, 2022-07-25:
﻿Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
I couldn't boot to Windows installation media and got Error 17. Tried official and custom ISOs but none worked, neither FAT32 nor NTFS worked too. Before this, Windows installation media could boot properly. Note that I can boot any Linux images properly.

Issue on https://github.com/pbatard/rufus/issues/1384 hasn't answered my problem.

[image]
[image]

Log
---
<FULL LOG>
I couldn't submit new issues while putting whole log here due to maximum character exceeded, so I'll upload it as a file instead.

[rufus.log](https://github.com/pbatard/rufus/files/9179700/rufus.log)

pbatard, 2022-07-25:
1. The log and screenshots you provided are ones from using an __unofficial__ version of Windows. I'm afraid I cannot support that, because bootloaders may have been altered.
2. The fact that you are stating that both FAT32 and NTFS fail to boot appear to indicate that this is indeed an incompatibility with the bootloaders from your ISO and your Lenovo UEFI firmware. For once thing, since `tiny10 21H2 x64 beta 1.iso (2.4 GB)` is less than 4 GB, you should not have to use NTFS, and I'm pretty sure that, if you simply extract the ISO files (e.g. using 7-zip) to a FAT32 formatted drive, you will see the exact same issue, because that's basically all Rufus does when using FAT32, which means that this can hardly be considered a Rufus related issue.
3. You don't need to have Secure Boot disabled when installing an __official__ version of Windows.

All in all, since you indicate that when using FAT32 in Rufus, you still have a problem, and you are also only providing data when using an __unofficial__ Windows 10 image, I do not see anything here for me to try to support. Especially, as I pointed out in #1384, code `[17] No mapping` being returned by a computer's UEFI firmware `StartImage()` when trying to execute the Microsoft Windows UEFI bootloaders definitely hints at a UEFI firmware issue, as Rufus (and UEFI:NTFS) have basically loaded the Microsoft UEFI bootloader in memory at this stage, and are simply telling the UEFI Firmware to execute it, which it should have no trouble doing.

stephnradd, 2022-07-25:
Forgor to mention that I tried official ISO too (Windows 10 Enterprise LTSC 2021) and Windows 10 21H1 but it didn't work too. It WAS working few months ago. Uploading the log at the moment...

pbatard, 2022-07-28:
> Uploading the log at the moment...

Any update on this?

Also can you please answer what happens when you try to boot tiny ISO after extracting the files to a FAT32 partition (which doesn't need to be created by Rufus)? Do you also get an error then? If so, what is it?

stephnradd, 2022-07-28:
> > Uploading the log at the moment...
> 
> Any update on this?
> 
> Also can you please answer what happens when you try to boot tiny ISO after extracting the files to a FAT32 partition (which doesn't need to be created by Rufus)? Do you also get an error then? If so, what is it?

Sorry I was busy doing another things
Going to, I'll keep you informed
By the way, I tried again yesterday with ISOs from Microsoft, still no luck  (still error 17), here's the log :
[rufus.log](https://github.com/pbatard/rufus/files/9209947/rufus.log)

JonnyTech, 2022-07-28:
> Rufus x86 v3.19.1911 (Portable)
> Windows version: Windows 11 Pro, 64-bit (Build 25163.1010)

I think that build of Windows is a dev / preview release. Just flagging that up for @pbatard to review.

pbatard, 2022-07-28:
> I think that build of Windows is a dev / preview release.

Yup. I tend to avoid looking at Insider builds altogether as I have been stung in the past wasting time trying to fix issues that were only present with Insider and that Microsoft fixed for the release.

I will wait for your report with an ISO that doesn't have file larger than 4 GB, so that you can report what happens when you use FAT32 for the file system.

stephnradd, 2022-07-29:
> I will wait for your report with an ISO that doesn't have file larger than 4 GB, so that you can report what happens when you use FAT32 for the file system.

It won't boot...

pbatard, 2022-07-29:
> > I will wait for your report with an ISO that doesn't have file larger than 4 GB, so that you can report what happens when you use FAT32 for the file system.
> 
> It won't boot...

The fact that it won't boot when using FAT32 makes it pretty clear that the error has nothing to do with Rufus and that `Start Failure` error you see when booting the media through UEFI:NTFS is simply the result of an incompatibility between your system and the UEFI bootloaders (which don't come from Rufus but from the ISO you are using).

As such, since there isn't much I can help with, I will close this issue.

closed 2022-07-29 by pbatard