pbatard/rufus #1996: Impossible to use Secure Boot and TPM bypass for already downloaded image

ww898, 2022-07-25:
Could you please return Secure Boot and TPM bypass options as was made in 3.18? Impossible to use with custom donwload images.

pbatard, 2022-07-25:
https://github.com/pbatard/rufus/wiki/FAQ#Help_I_no_longer_see_the_option_to_bypass_TPMSecure_Boot_with_Windows_11

Next time, follow the check list and see if your question is already answered in the FAQ.

ww898, 2022-07-25:
Well, tested 3.19 with new UX, so Secure Boot and TPM bypass doesn't work.

pbatard, 2022-07-25:
Seems to work fine for other folks. And of course, since you also didn't provide a log and indicate that you use a __custom__ download, it's very hard to fault Rufus there...

pbatard, 2022-07-25:
By the way, the best way to find out if the bypass was properly applied is to press <kbd>Shift</kbd>-<kbd>F10</kbd> once you get to Windows Setup (provided you actually boot from the USB. Are you booting from USB or trying an in-place upgrade?) to open a command prompt, then type `regedit` and navigate to `HKEY_LOCAL_MACHINE\SYSTEM\Setup\LabConfig`. If you see `Bypass###` entries there, it means Rufus did properly apply the TPM and Secure Boot bypass, and your issue is elsewhere.

ww898, 2022-07-25:
[image]
```
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CPRA_A64FRE_EN-US_DV5'
  Detected: Windows 11 ISO (Build 22621.1)
  Size: 5.1 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (UEFI only)
  Uses: Install.wim (version 0.13.1)
Using image: 22621.160.220604-1803.NI_RELEASE_SVC_PROD2_CLIENTPRO_OEMRET_A64FRE_EN-US.ISO (5.1 GB)

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive3 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 28.7 GB)
● Creating UEFI:NTFS Partition (offset: 30764111872, size: 1 MB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Found volume \\?\Volume{a0b6d71a-0c1c-11ed-9eab-2cf05d3f35e8}\
Disabling file indexing...
Successfully remounted \\?\Volume{a0b6d71a-0c1c-11ed-9eab-2cf05d3f35e8}\ as F:
Extracting files...
Image is a UDF image
Extracting: F:\boot\boot.sdi (3 MB)
Extracting: F:\boot\bootsect.exe (83.5 KB)
Extracting: F:\boot\etfsboot.com (4 KB)
Extracting: F:\boot\fonts\chs_boot.ttf (3.5 MB)
Extracting: F:\boot\fonts\cht_boot.ttf (3.7 MB)
Extracting: F:\boot\fonts\jpn_boot.ttf (1.9 MB)
Extracting: F:\boot\fonts\kor_boot.ttf (2.3 MB)
Extracting: F:\boot\fonts\malgun_boot.ttf (191.7 KB)
Extracting: F:\boot\fonts\malgun_console.ttf (81.5 KB)
Extracting: F:\boot\fonts\malgunn_boot.ttf (189.4 KB)
Extracting: F:\boot\fonts\meiryo_boot.ttf (162.0 KB)
Extracting: F:\boot\fonts\meiryo_console.ttf (90.4 KB)
Extracting: F:\boot\fonts\meiryon_boot.ttf (160.4 KB)
Extracting: F:\boot\fonts\msjh_boot.ttf (179.1 KB)
Extracting: F:\boot\fonts\msjh_console.ttf (200 KB)
Extracting: F:\boot\fonts\msjhn_boot.ttf (177.2 KB)
Extracting: F:\boot\fonts\msyh_boot.ttf (171.2 KB)
Extracting: F:\boot\fonts\msyh_console.ttf (96.2 KB)
Extracting: F:\boot\fonts\msyhn_boot.ttf (169.4 KB)
Extracting: F:\boot\fonts\segmono_boot.ttf (44.8 KB)
Extracting: F:\boot\fonts\segoe_slboot.ttf (100.5 KB)
Extracting: F:\boot\fonts\segoen_slboot.ttf (100.3 KB)
Extracting: F:\boot\fonts\wgl4_boot.ttf (48.9 KB)
Extracting: F:\boot\resources\bootres.dll (156.9 KB)
Extracting: F:\boot\resources\bootres_hci.dll (99.9 KB)
Extracting: F:\bootmgr.efi (1.7 MB)
Extracting: F:\efi\boot\bootaa64.efi (1.7 MB)
[... 939 lines cut ...]

Found USB 3.0 device ' USB  SanDisk 3.2Gen1 USB Device' (0781:55A9)
Using 'autorun.inf' label for drive F: 'CPRA_A64FRE_EN-US_DV5'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3740, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {04E407A7-DA1F-4E10-8571-AE1269DB364C}
Max parts: 128, Start Offset: 17408, Usable = 30765185536 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {17CD3FE8-7EEC-4110-AED9-C3FBE10F9E41}
  Size: 28.7 GB (30763063296 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {849C3F9E-91E1-4AD1-B041-0B29C0F2F612}
  Size: 1 MB (1048576 bytes)
  Start Sector: 60086156, Attributes: 0x9000000000000000
```
[image]

ww898, 2022-07-25:
[image]

pbatard, 2022-07-25:
That's weird. The log definitely shows that the customization options weren't applied, even without checking the registry keys.

Typically, you should have gotten something like:
```
Applying Windows customization:
Renamed 'F:\sources\appraiserres.dll' → 'F:\sources\appraiserres.bak'
Mounting 'F:\sources\boot.wim'...
mounted 'F:\sources\boot.wim [2]' on 'C:\Users\nil\AppData\Local\Temp\Ruf60D4.tmp'
Added 'Autounattend.xml' to 'F:\sources\boot.wim'
Unmounting 'F:\sources\boot.wim'...
Unmounted 'C:\Users\nil\AppData\Local\Temp\Ruf60D4.tmp [2]'
```
Before the `Finalizing` phase, and that does not happen for your image.

The only thing I can think of (but that I still need to validate) is that your __custom__ image is UEFI only (`Uses: Bootmgr (UEFI only)`) whereas official images from Microsoft are BIOS + UEFI (`Uses: Bootmgr (BIOS and UEFI)`).

Obviously, I test with images that come either from Microsoft or (like the ones from https://uupdump.net/) are as close as possible to an official Microsoft image, so I can't guarantee that something nonofficial will work. Can you provide some details on how you obtained that `22621.160.220604-1803.NI_RELEASE_SVC_PROD2_CLIENTPRO_OEMRET_A64FRE_EN-US.ISO` image?

ww898, 2022-07-25:
You are absolutely right the image was created from https://uupdump.net/ for Windows ARM64.

ww898, 2022-07-25:
v3.18 works properly

pbatard, 2022-07-25:
Ah, `A64` should have told me it was ARM64 indeed. And I can replicate the issue with a 21H2 ARM64 ISO I have (but not with a UEFI-only x86_64 ISO). I think the new Windows 11 options have altered how the bypass was were handled for non x86, so I'll fix that for 3.20.

pbatard, 2022-07-25:
And the cause of this issue is a typo. [This line](https://github.com/pbatard/rufus/blob/3cf41abf2e3dfe8cdb52a7ebd734eeb49731cc10/src/rufus.c#L1272) should have had `arch > ARCH_ARM_64` and not `arch >= ARCH_ARM_64`, since it means that the bypass and other options will not be applied indeed.

Thanks a lot for reporting this!

pbatard, 2022-07-25:
If you want to test that this is fixed in the upcoming release, feel free to download the Artifacts from the current MinGW build (once it's completed) at https://github.com/pbatard/rufus/actions/runs/2733562383.

commit 01dbb3d9 (2022-07-25): [wue] fix WUE options not being applied for ARM64 images

* Closes #1996
* Also set Coverity GitHub Actions workflow to use windows-latest