pbatard/rufus #1998: 9GB ISO Ubuntu error

Boshen97, 2022-07-26:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: 3.18.1877` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I made an Ubuntu system image iso file (Ubuntu system version is 18.04), this ISO file is about 9 G, I found that I couldn’t write the system disk normally, and the burned system disk was only 150MB, and I can't use it either. I don't think it's written normally, but the Ubuntu and Windows system image files can be written normally, so I want to know if my image file can be written normally? , if possible, what should I do to burn it in, please guide me.

Log
---
```
Rufus x86 v3.18.1877
Windows version: Windows 10 Pro, 64-bit (Build 19044.1826)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0804 (zh-CN)
Will use default UI locale 0x0804
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'zh-CN'
Found USB 3.0 device 'SanDisk Ultra USB 3.0 USB Device' (0781:5591)
Using 'autorun.inf' label for drive F: 'Prometheus'
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 14961, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001ECED7
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 114.6 GB (123059830784 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'sblive'
  Size: 8.6 GB (Projected)
  Has a >4GB file
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
Using image: sblive.iso (8.6 GB)

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 114.6 GB)
[... 320 lines cut ...]
NTFS Fixup completed.

Found USB 3.0 device 'SanDisk Ultra USB 3.0 USB Device' (0781:5591)
Using 'autorun.inf' label for drive F: 'Prometheus'
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 14961, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {2AA427EC-CCCE-4678-8253-41D351C7B792}
Max parts: 128, Start Offset: 17408, Usable = 123060845056 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {97A56AB1-01AE-4D5B-AE59-825A999E9483}
  Size: 114.6 GB (123058736640 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {6CE5C7B2-D52A-4AEC-9802-B90F3910DACD}
  Size: 1 MB (1048576 bytes)
  Start Sector: 240351143, Attributes: 0x9000000000000000
```

JonnyTech, 2022-07-26:
> libcdio: Non consecutive multiextent file parts for ''

This looks like a `libcdio` bug linked to Joliet extensions (see https://www.mail-archive.com/libcdio-devel@gnu.org/msg01927.html). @pbatard will be able to give a more comprehensive answer (and eventually a solution!), but for now can you describe the process that you used to create the image?

pbatard, 2022-07-26:
Yes, this looks like the issue that was fixed in Rufus 3.19 with ae377ae8cafea7a5454d1b1b0b0966bbf1f4b6f4.

You are running an __old__ version of Rufus, so please upgrade to Rufus 3.19 and you should find that your issue is fixed.
For the record, this is why you should enable the check for updates, because bugs like this get fixed all the time in newer releases, and by not having updates enabled, you are not being alerted to new versions and are running into issues that have been fixed.

As I believe upgrading to Rufus 3.19 should fix your problem, I will close this issue.

closed 2022-07-26 by pbatard